zephyrproject-rtos/zephyr #14156: Mac OSX Documentation Update Needed:

sgoggins, 2019-03-07:
**Describe the bug**
Documentation update needed : https://docs.zephyrproject.org/1.9.0/getting_started/installation_mac.html
```markdown
Install tools needed for building the toolchain (if needed):

works:

$ brew install gettext help2man mpfr gmp coreutils wget

does not work: 

$ brew tap homebrew/dupes

(zephyr) ➜  ~ brew tap homebrew/dupes               

Error: homebrew/dupes was deprecated. This tap is now empty as all its formulae were migrated.
(zephyr) ➜  ~ 

$ brew install grep --with-default-names

Error: invalid option: --with-default-names

(zephyr) ➜  ~ 

```
**To Reproduce**
Steps to reproduce the behavior:
1. Follow instructions noted above. 

**Expected behavior**
Successful configuration

**Impact**
Appears to be an optional configuration, so possibly low. 

**Screenshots or console output**

**Environment (please complete the following information):**
 - OS: OSX

**Additional context**
First time Zephyr installation.

sgoggins, 2019-03-07:
Adding comments from subsequent steps that do not work because I do not know if the failure noted above is the reason.  In short, the make command in `samples/hello_world` returns the error 

```bash
(zephyr) ➜  hello_world git:(master) make
make: *** No targets specified and no makefile found.  Stop.
```

nashif, 2019-03-09:
looks like you are looking at 1.9.0 documentation, which is very old, can you please consult the latest documentation?

https://docs.zephyrproject.org/latest/getting_started/installation_mac.html

carlescufi, 2019-03-09:
Closing this since this is clearly not the case anymore

closed 2019-03-09 by carlescufi