zephyrproject-rtos/zephyr #14173: Configure QEMU to run independent of the host clock for all arches

andrewboie, 2019-03-07:
We have been struggling for years with issues related to how QEMU attempts to synchronize guest timer interrupts with the host clock, for example #12553 

QEMU does nominally support some options to decouple the guest from host clock:

```
-rtc [base=utc|localtime|date][,clock=host|rt|vm][,driftfix=none|slew]
                set the RTC base and clock, enable drift fix for clock ticks (x86 only)
-icount [shift=N|auto][,align=on|off][,sleep=on|off,rr=record|replay,rrfile=<filename>,rrsnapshot=<snapshot>]
                enable virtual instruction counter with 2^N clock ticks per
                instruction, enable aligning the host and virtual clocks
                or disable real time cpu sleeping
```

Unfortunately to date nobody has ever gotten this working.

The objective is as follows: for all architectures that use QEMU for sanitycheck runs (currently x86, x86_64, arm, riscv32, xtensa, nios2) configure QEMU to use icount.

Beware that there may be bugs in QEMU which prevent this from working, and whoever works on this should be prepared to dive into the QEMU source and/or work with QEMU upstream.

Also it's worth noting that moving ANY arch to use icount is progress, so even if we can get only some but not all arches working with this, that is worth doing. At this time I recommend x86 and ARM (the qemu_x86 and mps2_an385 targets) as priority. But eventually we need them all to finally get rid of these spurious sanitycheck failures.

wentongwu, 2019-06-04:
I did some sanitychecks with "-icount shift=4,align=off,sleep=off" on qemu_x86 and mps2_an385, only tests/kernel/critical/kernel_critical failed on mps2_an385. And run that case(tests/kernel/critical/kernel_critical) on mps2_an385 specially, I can get this case run successfully, but it takes long time to execute, so it's why get the timeout result on sanitychecks. And it means icount mode produces deterministic output, but does not provide cycle accurate. 
I will try to review the qemu code to try to understand icount mode well, and then do next plan.

andrewboie, 2019-06-06:
@wentongwu thanks for the update :)

andrewboie, 2019-06-06:
I remember tests/kernel/critical takes a while even without icount, so perhaps we need to change the test to not take so much time.

wentongwu, 2019-06-06:
I run several rounds stress tests, founding more failed/timeout, and most of them are related to timer accurateness, I need do more debug and code review.

wentongwu, 2019-06-07:
With below patch, I did some sanitychecks again with "-icount shift=4,align=off,sleep=off **-rtc clock=vm**" on qemu_x86 and mps2_an385, so far no error occured. Still reviewing qemu code...

diff --git a/tests/kernel/critical/src/main.c b/tests/kernel/critical/src/main.c
index 372f931fa1..38d26a987c 100644
--- a/tests/kernel/critical/src/main.c
+++ b/tests/kernel/critical/src/main.c
@@ -32,8 +32,8 @@
 #include <linker/sections.h>
 #include <ztest.h>

-#define NUM_MILLISECONDS        5000
-#define TEST_TIMEOUT            20000
+#define NUM_MILLISECONDS        500
+#define TEST_TIMEOUT            2000

 static u32_t critical_var;
 static u32_t alt_thread_iterations;

wentongwu, 2019-06-07:
With below patch in every code path,  four processes run " for i in {1..10000}; do ./scripts/sanitycheck
-p qemu_x86 -p mps2_an385 -T tests/kernel ; done " individually in four different code path on the same one machine (ubuntu 18.04), two processes with "-icount shift=4,align=off,sleep=off -rtc clock=vm" configuration, one process with "-icount shift=4,align=off,sleep=off" configuration, the last one with the default configuration (the original configuration, no changes on qemu configuration). So far the default one easily fail, others run correctly. 

diff --git a/scripts/sanitycheck b/scripts/sanitycheck
index 662172cda2..12788cf555 100755
--- a/scripts/sanitycheck
+++ b/scripts/sanitycheck
@@ -1453,7 +1453,7 @@ testcase_valid_keys = {"tags": {"type": "set", "required": False},
                        "build_on_all": {"type": "bool", "default": False},
                        "skip": {"type": "bool", "default": False},
                        "slow": {"type": "bool", "default": False},
-                       "timeout": {"type": "int", "default": 60},
+                       "timeout": {"type": "int", "default": 600},
                        "min_ram": {"type": "int", "default": 8},
                        "depends_on": {"type": "set"},
                        "min_flash": {"type": "int", "default": 32},

andrewboie, 2019-06-07:
five minutes is a lot for a default. how many / which tests start taking longer than 60 seconds with the icount change?

andrewboie, 2019-06-07:
> "-icount shift=4,align=off,sleep=off -rtc clock=vm" 

Can you give some detail on how these parameters were selected? What do you observe for different shift values, and why did you settle on 4?

wentongwu, 2019-07-22:
@andrewboie if 10 PRs submitted at the same time, what's the CI's behavior? run shippable for each PR one by one ? or run shippable for all the PRs at the same time ? Thanks

andrewboie, 2019-07-22:
@wentongwu it runs it on  the last patch in the series.
Having said that, your PRs need to be bisectable without breaking the build.

wentongwu, 2019-07-23:
> @wentongwu it runs it on the last patch in the series.
> Having said that, your PRs need to be bisectable without breaking the build.

I mean if ten different developers submit their own PRs at the same time, will CI schedule it one by one or run it concurrently? If concurrently, icount mode can't guarantee the complete time of the test case, because first it will take longer time than normal mode, second more qemu processes in one machine will cause little time for each qemu process.

andrewboie, 2019-07-23:
> If concurrently, icount mode can't guarantee the complete time of the test case

I see what you mean. The build slaves are supposed to be load balanced, with the number of CPUs selected for the job matching the hardware it is running on.

Having said that, perhaps we need a more sophisticated accounting on how long tests are taking to complete to determine whether they have gotten stuck somewhere.

Are you at the point where icount is working but we just need longer test timeouts?

wentongwu, 2019-07-23:
1) from the point of qemu end user, it seems icount mode is working for mps2_an385 and qemu_x86.
2) and for mps2_an385, the shift value located in [1:4] is working except there are some timeouts. And for qemu_x86, the shift value located in [4, 9] also is working except there is a always failed case( kernel.memory_protection.stack_random) which need more time debug. 
3) how to run icount mode. Run this command first (export QEMU_EXTRA_FLAGS="-icount shift=N, align=off,sleep=off -rtc clock=vm") which is to set icount mode and it's parameters( N is the shift value),  and then run Zephyr as usual. 
4) for why other shift values not working, I'm GDBing qemu which runs zephyr.elf to try to figure out the reason, and it related to Zephyr's timer driver and also Qemu timer's implementation(timer hardware).

wentongwu, 2019-07-24:
problems to solve for platform mps2_an385 (will update here if found more failed case):

1. shift=9, tests/kernel/lifo/lifo_usage/kernel.lifo.usage case always fail. 
2. so far there are some timeout cases, we can adjust timeout value as #17107 do. 
3. find the most stable shift value by test results and qemu icount code analysis.

wentongwu, 2019-08-12:
for the fail with shift=9 or shift=10 on case tests/kernel/lifo/lifo_usage/kernel.lifo.usage, I think the reason is that the frequency of the emulated CPU is too low, 2Mhz(2^9ns per instruction) or 1Mhz(2^10ns per instruction), and actually the frequency of mps2_an385 should be 25Mhz. 
```
struct timeout_order_data timeout_order_data[] = {
        {0, &lifo_timeout[0], 20, 2, 0},
        {0, &lifo_timeout[0], 40, 4, 1},
        {0, &lifo_timeout[0], 0, 0, 2},
        {0, &lifo_timeout[0], 10, 1, 3},
        {0, &lifo_timeout[0], 30, 3, 4},
};
```
And Qemu in icount mode can give deterministic execution for guest code timeout if the frequency of the emulated CPU is similar to or great than the real frequency. But if the frequency of the emulated CPU is too high, that means the shift value is too low, the test code in Zephyr like spin_for_ms will take very long wall time to complete the test case, so it's very likely timeout happened for that cases spinning for some time. 
So I think for the guest code timeout,  similar to the real frequency of the machine, at least for mps2_an385, is the best choice (And before this conclusion, I'm trying to make every shift work for our test cases, now it seems we can't achieve it).

wentongwu, 2019-08-12:
another timing issue in Zephyr test case is that we use the delta between two point-in-time, for example case tests/kernel/timer/timer_api 
```
void test_timer_k_define(void)
{
...
        k_timer_start(&ktimer, DURATION, PERIOD);

        /* Call the k_timer_start() again to make sure that
         * the initial timeout request gets cancelled and new
         * one will get added.
         */
        busy_wait_ms(DURATION / 2);
        k_timer_start(&ktimer, DURATION, PERIOD);
        tdata.timestamp = k_uptime_get();
        busy_wait_ms(DURATION + PERIOD * EXPIRE_TIMES + PERIOD / 2);

...
}
```
and case tests/kernel/sched/schedule_api
```
void test_slice_reset(void)
{
...
                /*set reference time*/
                cycles_delta(&elapsed_slice);

                /* current thread (ztest native) consumed a half timeslice*/
                t32 = k_cycle_get_32();
                while (k_cycle_get_32() - t32 < HALF_SLICE_SIZE_CYCLES) {
#if defined(CONFIG_ARCH_POSIX)
                        k_busy_wait(50);
#endif
                }

                /* relinquish CPU and wait for each thread to complete*/
                for (int i = 0; i < NUM_THREAD; i++) {
                        k_sem_take(&sema, K_FOREVER);
                }

...
}
```
And I don't think the time point in Qemu icount mode is accurate, even though the cases using delta between two time point are PASSED on most of the shift value, but I believe there should be deviation. So my recommendation is to not use time point in test case instead we should use timeout in guest code. I post this point in Qemu mail list https://lists.gnu.org/archive/html/qemu-devel/2019-08/msg01497.html, so far nobody answer me, but I will post more to try to discuss it with Qemu developer, hope I'm right.

andrewboie, 2019-10-31:
What is the current status of this?

wentongwu, 2019-11-01:
> What is the current status of this?

all my changes are here https://github.com/wentongwu/zephyr/commits/icount
after you visit Shanghai, I have no time to handle this issue. If this issue is urgent, I will try to find the root cause for the only failed case next week.

carlocaione, 2019-11-15:
Currently fighting with this on the ARM64 port when using the QEMU release by Xilinx. The only way to have all the tests passing is basically using `-icount shift=0,align=off,sleep=off` and disabling timeouts in the `sanitycheck` script.

wentongwu, 2019-11-16:
> Currently fighting with this on the ARM64 port when using the QEMU release by Xilinx. The only way to have all the tests passing is basically using `-icount shift=0,align=off,sleep=off` and disabling timeouts in the `sanitycheck` script.

@carlocaione 
that's great, as my above comments the method about how to choose value of shift is that the the clock-frequency decided by shift should be similar to the real frequency of that machine(at least that's correct for mps2_an385), so I guess the clock-frequency of your system is most close to 1/(2^0 ns), could you please confirm that? Thanks

yes, the little of shift value the more host time(wall time) needed for some cases which spin to reach some time period like busy_wait_ms, I think the timeout on your side should also happen on these cases.

carlocaione, 2019-11-16:
> that's great, as my above comments the method about how to choose value of shift is that the the clock-frequency decided by shift should be similar to the real frequency of that machine(at least [...]

I'm seriously confused by this statement. Can you give a technical explanation of this? `1/(2^0 ns)` is 1MHz that is definitely a low frequency for my platform.

I did some tests today. Let focus for example on the `tests/kernel/sleep/kernel.timing`. This keeps failing when not using `-icount`.

In particular what fails is:
```
start_tick = k_uptime_get_32();
k_sleep(ONE_SECOND);
end_tick = k_uptime_get_32();

if (!sleep_time_valid(start_tick, end_tick, ONE_SECOND_ALIGNED)) {
```
- When not using `-icount` I have something like `start_tick==51` and `end_tick==1054`.
- When using `-icount shift=0,align=off,sleep=off` I have `start_tick==1` and `end_tick==1001`

So basically when not using `-icount` we are off by a few ticks. `ONE_SECOND_ALIGNED` is strictly `1000` so I was thinking if here one workaround could just relax a bit this alignment when `-icount` is not used.

Please note that this solves one of the two problems I have. The other issue is that when using `-icount` some tests just take too much time for the `60s` timeout we have in the `sanitycheck` script.

carlocaione, 2019-11-16:
Just to add one more thing. This behavior is strictly dependent on QEMU.

When I test my ARM64 branch with a regular QEMU emulating a Cortex-A53 virt platform I don't need any `-icount` trick and I'm able to pass 100% of the test. The `-icount` trick is only useful when emulating a Xilinx board using the QEMU released by Xilinx.

wentongwu, 2019-11-16:
> I'm seriously confused by this statement. Can you give a technical explanation of this? `1/(2^0 ns)` is 1MHz that is definitely a low frequency for my platform.

should be 1GHz

carlocaione, 2019-11-16:
> should be 1GHz

Ah, I cannot do math apparently. You are right. Still I don't get the reasoning behind.

carlocaione, 2019-11-16:
So it turns out that at least in my case I just forgot to select `QEMU_TARGET`. With that I pass all the tests again.

andrewboie, 2020-02-19:
I've moved this to 2.3 but we've made some good progress lately.

wentongwu, 2020-02-20:
quick testing, seems most of other QEMU platforms also can work, will pick-up little timeout or failed cases.

maksimmasalski, 2020-03-23:
@andrewboie Are you going to develop test case for that QEMU configuration?

carlescufi, 2020-04-28:
The current manifestation of the lack for this enhancement is the following:
`tests/counter/counter_basic_api` failing on `mps2_an385` in CI occasionally.

andrewboie, 2020-05-01:
I found and fixed two issues that were exposed by enabling icount on x86 targets:

https://github.com/zephyrproject-rtos/zephyr/pull/24879
https://github.com/zephyrproject-rtos/zephyr/pull/24880

After https://github.com/zephyrproject-rtos/zephyr/pull/24879 is merged, I have found icount works well on 32-bit x86, and I think we should enable it. I tested with `-icount shift=7,align=off,sleep=off -rtc clock=vm`

x86_64 unfortunately has other issues. With icount turned on and the above fix applied, I still see:
```
Deleting output directory /dev/shm/sanity-out
INFO    - JOBS: 48
INFO    - Building initial testcase list...
INFO    - 176 test configurations selected, 631 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Total complete:   39/ 176  22%  skipped:   12, failed:    0
ERROR   - qemu_x86_64               tests/kernel/mbox/mbox_usage/kernel.mailbox.usage  FAILED: Failed
ERROR   - see: /dev/shm/sanity-out/qemu_x86_64/tests/kernel/mbox/mbox_usage/kernel.mailbox.usage/handler.log
INFO    - Total complete:   48/ 176  27%  skipped:   12, failed:    1
ERROR   - qemu_x86_64               tests/kernel/lifo/lifo_usage/kernel.lifo.usage     FAILED: Failed
ERROR   - see: /dev/shm/sanity-out/qemu_x86_64/tests/kernel/lifo/lifo_usage/kernel.lifo.usage/handler.log
INFO    - Total complete:   97/ 176  55%  skipped:   17, failed:    2
ERROR   - qemu_x86_64               tests/kernel/smp/kernel.multiprocessing.smp        FAILED: Failed
ERROR   - see: /dev/shm/sanity-out/qemu_x86_64/tests/kernel/smp/kernel.multiprocessing.smp/handler.log
INFO    - Total complete:  106/ 176  60%  skipped:   19, failed:    3
ERROR   - qemu_x86_64               tests/kernel/common/kernel.common.misra            FAILED: Timeout
ERROR   - see: /dev/shm/sanity-out/qemu_x86_64/tests/kernel/common/kernel.common.misra/handler.log
INFO    - Total complete:  107/ 176  60%  skipped:   19, failed:    4
ERROR   - qemu_x86_64               tests/kernel/common/kernel.common                  FAILED: Timeout
ERROR   - see: /dev/shm/sanity-out/qemu_x86_64/tests/kernel/common/kernel.common/handler.log
INFO    - Total complete:  108/ 176  61%  skipped:   19, failed:    5
ERROR   - qemu_x86_64               tests/kernel/spinlock/kernel.multiprocessing.spinlock FAILED: Timeout
ERROR   - see: /dev/shm/sanity-out/qemu_x86_64/tests/kernel/spinlock/kernel.multiprocessing.spinlock/handler.log
INFO    - Total complete:  109/ 176  61%  skipped:   19, failed:    6
ERROR   - qemu_x86_64               samples/portability/cmsis_rtos_v2/timer_synchronization/sample.portability.cmsis_rtos_v2.timer_synchronization FAILED: Timeout
ERROR   - see: /dev/shm/sanity-out/qemu_x86_64/samples/portability/cmsis_rtos_v2/timer_synchronization/sample.portability.cmsis_rtos_v2.timer_synchronization/handler.log
INFO    - Total complete:  121/ 176  68%  skipped:   19, failed:    7
ERROR   - qemu_x86_64               tests/kernel/mem_protect/userspace/kernel.memory_protection.userspace FAILED: Failed
ERROR   - see: /dev/shm/sanity-out/qemu_x86_64/tests/kernel/mem_protect/userspace/kernel.memory_protection.userspace/handler.log
INFO    - Total complete:  176/ 176  100%  skipped:   20, failed:    8
INFO    - 148 of 156 tests passed (94.87%), 8 failed, 20 skipped with 0 warnings in 116.42 seconds
INFO    - In total 818 test cases were executed on 1 out of total 243 platforms (0.41%)
```

wentongwu, 2020-05-01:
```
INFO    - 20/86 qemu_x86_64               tests/kernel/smp/kernel.multiprocessing.smp        FAILED Failed (qemu 16.006s)
ERROR   - see: /home/wentongw/zephyr/zephyrproject/zephyr/sanity-out/qemu_x86_64/tests/kernel/smp/kernel.multiprocessing.smp/handler.log
```

this problem should be "with icount mode enable, although configure Qemu multi cpu, there is only one vCPU thread to handle guest code", for case `test_coop_resched_threads` current cooperative thread doesn't give up the only vCPU and other high priority cooperative thread doesn't get vCPU time to run, so there is assert.

wentongwu, 2020-05-01:
```
INFO    - 61/86 qemu_x86_64               tests/kernel/mbox/mbox_usage/kernel.mailbox.usage  FAILED Failed (qemu 2.324s)
ERROR   - see: /home/wentongw/zephyr/zephyrproject/zephyr/sanity-out/qemu_x86_64/tests/kernel/mbox/mbox_usage/kernel.mailbox.usage/handler.log
```
if increase the timeout value as below, this case can pass almost because the same reason, only one vCPU run for QEMU icount even configured multi cores, sender thread can't complete within 2 milliseconds virtual time. 
```
diff --git a/tests/kernel/mbox/mbox_usage/src/main.c b/tests/kernel/mbox/mbox_usage/src/main.c
index d97ae50565..12bdffefec 100644
--- a/tests/kernel/mbox/mbox_usage/src/main.c
+++ b/tests/kernel/mbox/mbox_usage/src/main.c
@@ -103,7 +103,7 @@ void test_msg_receiver(void)
                              test_send, &mbox, NULL, NULL,
                              K_PRIO_PREEMPT(0), 0, K_NO_WAIT);

-       msg_receiver(&mbox, K_ANY, K_MSEC(2));
+       msg_receiver(&mbox, K_ANY, K_MSEC(200));
        k_thread_abort(tid);
 }
```

wentongwu, 2020-05-01:
```
INFO    - 74/86 qemu_x86_64               tests/kernel/lifo/lifo_usage/kernel.lifo.usage     FAILED Failed (qemu 7.419s)
ERROR   - see: /home/wentongw/zephyr/zephyrproject/zephyr/sanity-out/qemu_x86_64/tests/kernel/lifo/lifo_usage/kernel.lifo.usage/handler.log
```

```
INFO    - 61/86 qemu_x86_64               tests/kernel/mbox/mbox_usage/kernel.mailbox.usage  FAILED Failed (qemu 2.324s)
ERROR   - see: /home/wentongw/zephyr/zephyrproject/zephyr/sanity-out/qemu_x86_64/tests/kernel/mbox/mbox_usage/kernel.mailbox.usage/handler.log
```

```
INFO    - 71/86 qemu_x86_64               tests/kernel/mem_protect/userspace/kernel.memory_protection.userspace FAILED Exited with 2 (qemu)
ERROR   - see: /home/wentongw/zephyr/zephyrproject/zephyr/sanity-out/qemu_x86_64/tests/kernel/mem_protect/userspace/kernel.memory_protection.userspace/handler.log
```

these three issues are all related to only one vCPU existed when icount enabled, but actually run time CONFIG_MP_NUM_CPUS=2

wentongwu, 2020-05-01:
```
INFO    - 50/86 qemu_x86_64               tests/kernel/spinlock/kernel.multiprocessing.spinlock FAILED Timeout (qemu 61.099s)
ERROR   - see: /home/wentongw/zephyr/zephyrproject/zephyr/sanity-out/qemu_x86_64/tests/kernel/spinlock/kernel.multiprocessing.spinlock/handler.log
```
this case requires
```
tests:
  kernel.multiprocessing.spinlock:
    filter: CONFIG_SMP and CONFIG_MP_NUM_CPUS > 1
```
but we only have one core with icount enabled.

wentongwu, 2020-05-01:
I see only above 5 failed cases with icount enabled on qemu_x86_64 platform, if config CONFIG_MP_NUM_CPUS=1 for these 5 cases, qemu_x86_64 can work well with `-icount shift=5,align=off,sleep=off -rtc clock=vm`

andrewboie, 2020-05-01:
OK. I think we should open a separate issue specifically for x86_64 SMP. We may need a completely different emulator to do this right, or this may have to be a documented limitation.

closed 2020-05-14 by carlescufi

abrodkin, 2020-07-06:
Solution for ARC should be in https://github.com/zephyrproject-rtos/zephyr/pull/26646.