zephyrproject-rtos/zephyr #14184: zephyr.git/tests/benchmarks: Stuck at  delaying boot banner

cinlyooi-intel, 2019-03-08:
**Describe the bug**
Only seeing `***** delaying boot 1000ms (per build configuration *****`
Not seeing the follow up `***** Booting Zephyr OS v1.14.0-rc1-885-g32caf24cb4aa (delayed boot 1000ms) *****`

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=quark_se_c1000_ss_board -DARCH=arc
3. make
4. make run

**Screenshots or console output**
```
***** delaying boot 1000ms (per build configuration) *****
```

**Environment (please complete the following information):**
 - OS: fedora29
 - Toolchain: zephyr
 - Commit SHA or Version used:  1f74e0b1f75eff63568c1c807f003df7c3b7793b

**Additional context**
I added `printk()` statement to the start of `main()` and it did not get print out on the console. This strongly suggests the main() function was not entered.

AdithyaBaglody, 2019-03-14:
@cinlyooi-intel unable to reproduce this issue. tested with timing_info. works as expected.

cinlyooi-intel, 2019-03-14:
@AdithyaBaglody Still have the same problem on 4278bf5673480effdca3b1e048cecb7a9ab32976
My description of the problem is bad. Thanks for figuring out it is `timing_info`

rljordan, 2019-03-18:
@AdithyaBaglody  It appears this is still an issue.

AdithyaBaglody, 2019-03-19:
I am unable to reproduce this issue with the latest master f4c8e3ff3197d36773d68b4b4802daabc401f918

rljordan, 2019-03-25:
@cinlyooi-intel  Can this be closed?

cinlyooi-intel, 2019-04-02:
@rljordan : Sorry I cannot close it. I am still seeing it on latest master (193fb971c24f827464982307b5b3bb34de0fa98d). However, if someone can show that it cannot be reproduced on a real board then we can close it.

AdithyaBaglody, 2019-04-02:
>cannot be reproduced on a real board then we can close it.

@cinlyooi-intel how are you testing this? because I am not able to see this issue on the actual hardware. if you are testing on a test framework then please assign it to the respective person so that this can move ahead.

cinlyooi-intel, 2019-04-02:
Closed it since @AdithyaBaglody says it works on actual hardware.

closed 2019-04-02 by cinlyooi-intel