zephyrproject-rtos/zephyr #14186: tests/cmsis_rtos_v1 fails on nrf boards

arun1joshi, 2019-03-08:
The subtest test_kernel_systick in  tests/cmsis_rtos_v1 fails on nrf boards - nrf52_pca10040, nrf52840_pca10056, nrf51_pca10028 

The test is failing on this commit: [2384ebc]

Arch: arm
Platforms: nrf52_pca10040, nrf52840_pca10056, nrf51_pca10028 
```
Error on console log:
***** Booting Zephyr OS v1.14.0-rc1-899-g2384ebc2e325 (delayed boot 1000ms) *****
Running test suite test_cmsis_apis
===================================================================
starting test - test_kernel_start
PASS - test_kernel_start
===================================================================
starting test - test_kernel_systick

    Assertion failed at /home/joshiaru/zephyr.git/tests/cmsis_rtos_v1/src/kernel_apis.c:51: test_kernel_systick: diff >= WAIT_TIME_US is false

FAIL - test_kernel_systick
===================================================================
starting test - test_thread_apis
PASS - test_thread_apis
===================================================================
starting test - test_thread_prio
PASS - test_thread_prio
===================================================================
starting test - test_thread_instances
PASS - test_thread_instances
===================================================================
starting test - test_timer
oneshot_callback (Timer 1) = 1
periodic_callback (Timer 2) = 1
periodic_callback (Timer 2) = 2
periodic_callback (Timer 2) = 3
periodic_callback (Timer 2) = 4
periodic_callback (Timer 2) = 5
periodic_callback (Timer 2) = 6
PASS - test_timer
===================================================================
starting test - test_mutex
PASS - test_mutex
===================================================================
starting test - test_mutex_lock_timeout
PASS - test_mutex_lock_timeout
===================================================================
starting test - test_semaphore
PASS - test_semaphore
===================================================================
starting test - test_mempool
PASS - test_mempool
===================================================================
starting test - test_mailq
PASS - test_mailq
===================================================================
starting test - test_messageq
PASS - test_messageq
===================================================================
starting test - test_signal_events_no_wait
PASS - test_signal_events_no_wait
===================================================================
starting test - test_signal_events_timeout
PASS - test_signal_events_timeout
===================================================================
starting test - test_signal_events_signalled
PASS - test_signal_events_signalled
===================================================================
starting test - test_signal_events_isr
PASS - test_signal_events_isr
===================================================================
Test suite test_cmsis_apis failed.
===================================================================
RunID: :vmdf
PROJECT EXECUTION FAILED

andrewboie, 2019-03-08:
The cited commit seems orthogonal to this feature.

arun1joshi, 2019-03-09:
[corrected]
The test is failing on this commit: [b710177]

rgundi, 2019-03-15:
The cited commit doesn't concern this feature. The KernelSysTick may have gotten broken due to changes done to delay handling for these nrf boards (which are very slow).
I have tried running this on nrf boards but the results are inconsistent. I'll continue checking this but it would help if @arun1joshi can do a git bisect to find out the offending commit.

overheat, 2019-03-15:
@rgundi  It works pretty well with the latest zephyr on my nrf board. I cannot reproduce the failure.
[image]

rgundi, 2019-03-18:
hmm.. looks like 222fd8f1ab63d758bacfd31d36cb33ae117f771a did the trick.

rgundi, 2019-03-18:
@arun1joshi .. Can you please confirm this is working fine on the latest stack? If yes, please close this issue.

arun1joshi, 2019-03-19:
@rgundi, I still see the same failure on the latest commit [3a8767c]
===================================================================
starting test - test_kernel_systick

    Assertion failed at /home/joshiaru/zephyr.git/tests/cmsis_rtos_v1/src/kernel_apis.c:51: test_kernel_systick: diff >= WAIT_TIME_US is false

FAIL - test_kernel_systick
===================================================================

rgundi, 2019-03-19:
@arun1joshi Yes. Able to reproduce on the latest tip.  I could see that the commit d4b4b992725360a95ca1ffa357286be0b16615e5 may be the culprit. It introduced "ARCH_HAS_CUSTOM_BUSY_WAIT" and when i disabled it for nrf51 and nrf52 boards (via their respective Kconfig.defconfig.series), i could see that it was passing. Please verify this. Please note reverting that commit is not an option as git complains since things have moved on quite far now.

ManojSubbarao, 2019-03-21:
@arun1joshi Any updates from your verification as mentioned by @rgundi

nashif, 2019-03-27:
still an issue.

rgundi, 2019-03-27:
@nashif Agree it's still an issue but i have pointed out the commit where it's breaking. It apparently is a nrf specific change.
@pabigot and @carlescufi,
Can you please look into this issue?

pabigot, 2019-03-27:
That change was made because `k_busy_wait()` was terminating 30% early for delays less than 100 us (#11626).  I proposed that patch because the shared implementation was obscure while the basic functionality was straightforward.  After getting more experience in Zephyr's internals I can imagine there's some kernel-level entanglement that Zephyr expects that got lost with the custom implementation.  (I think I saw a comment somewhere recently stating that `k_busy_wait()` would only work if multithreading was enabled, which seems odd but if true definitely suggests there's some internal coupling.)

Reverting  d4b4b992725360a95ca1ffa357286be0b16615e5 should be easy regardless of subsequent changes, and should probably be tried, with the understanding that doing so might restore the original problematic behavior, which may or may not reveal new problems.

I'm not using Zephyr anymore and cannot contribute further to any workaround/solution at this time.

rgundi, 2019-03-28:
@carlescufi Can somebody from nordic take a look at d4b4b99? Perhaps modify or worst case, revert the patch?

pizi-nordic, 2019-03-28:
I have found root cause of such behaviour. All the details are below:

- The nRF5x chips have 2 distinct clocks (Low Frequency and High Frequency).
- Each clock could be generated by dedicated RC or crystal oscillator.
- The default configuration of the test selects XTAL oscillator for LF clock (but does not wait for oscillator startup) and RC for the HF clock.

Since the k_busy_wait() uses CPU clock driven from HF clock and k_cycle_get_32 uses RTC feed by LF clock we are not comparing apples to apples and depending on the chip, temperature and other things we could get this test either passing or failing.

I suggest reverting the https://github.com/zephyrproject-rtos/zephyr/commit/d4b4b992725360a95ca1ffa357286be0b16615e5as Zephyr assumes that k_busy_wait() will be in match with k_cycle_get_32(). Unfortunately such step will significantly reduce precision of k_busy_wait() and should be perceived as workaround.

closed 2019-03-28 by galak