zephyrproject-rtos/zephyr #14197: Missing implementation for hwinfo for some STM32 families

therealprof, 2019-03-08:
**Describe the bug**

hwinfo for STM32 is gated to be only usable for  STM32F0, STM32F3, STM32L4 and STM32F7 families. However according to ST documentation I don't see any family which doesn't support the 96 bit unique ID.

Especially for devices supporting any kind of networking (e.g. STM32F1, STM32F2 and STM32F4) it is an important feature to have, e.g. to derive Mac addresses from.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=nucleo_f103rb
3. make menuconfig
4. enable CONFIG_HWINFO
5. See error

**Expected behavior**

Hwinfo device id is available for all families supporting it.

**Impact**

Major annoyance.

therealprof, 2019-03-08:
cc @alexanderwachter

alexanderwachter, 2019-03-08:
The STMcube only has HAL_GetUIDwX(void) implementations for those devices.

therealprof, 2019-03-08:
Well, not quite. It does exist but with a different signature:
```
ext/hal/st/stm32cube/stm32f1xx/drivers/include/stm32f1xx_hal.h:void HAL_GetUID(uint32_t *UID);
ext/hal/st/stm32cube/stm32f1xx/drivers/src/stm32f1xx_hal.c:void HAL_GetUID(uint32_t *UID)
ext/hal/st/stm32cube/stm32f2xx/drivers/include/stm32f2xx_hal.h:void HAL_GetUID(uint32_t *UID);
ext/hal/st/stm32cube/stm32f2xx/drivers/src/stm32f2xx_hal.c:void HAL_GetUID(uint32_t *UID)
ext/hal/st/stm32cube/stm32f4xx/drivers/include/stm32f4xx_hal.h:void HAL_GetUID(uint32_t *UID);
ext/hal/st/stm32cube/stm32f4xx/drivers/src/stm32f4xx_hal.c:void HAL_GetUID(uint32_t *UID)
```

therealprof, 2019-03-08:
Maybe it would make more sense to use the `LL` API instead of the `HAL` API because there it is uniform across all families:
```
  __STATIC_INLINE uint32_t LL_GetUID_Word0(void);
  __STATIC_INLINE uint32_t LL_GetUID_Word1(void);
  __STATIC_INLINE uint32_t LL_GetUID_Word2(void);
```

alexanderwachter, 2019-03-08:
Good to know :-). Thanks for investigating this. I'll try to make an implementation as soon as possible.

alexanderwachter, 2019-03-08:
@therealprof  #14208

alexanderwachter, 2019-03-10:
Resolved by #14208

closed 2019-03-10 by alexanderwachter