zephyrproject-rtos/zephyr #14209: unable to flash sam_e70_xplained

andrewboie, 2019-03-08:
I cloned the Zephyr repo like I usually do.
I then build a test case in the normal way:

    mkdir out
    cd out
    export BOARD=sam_e70_xplained
    cmake ..
    make -j8

I then try to flash:

    make flash

I get:

```
West version found in path does not support '/usr/bin/gmake flash', ensure west is installed and not only the bootstrapper. run 'west init' to fetch west.
Built target flash
```

This used to work.

carlescufi, 2019-03-09:
@andrewboie can you run:

```
west update
west --version
```

from your zephyr folder?

andrewboie, 2019-03-11:
I didn't setup a parent folder with west initialized in it.
Closing since this is required for flashing to work.

closed 2019-03-11 by andrewboie