zephyrproject-rtos/zephyr #14240: Logging Kconfig options are confusing

ulfalizer, 2019-03-10:
There's these two options in `(top menu) -> Logging Options`:

[image]
[image]

The prompts don't tell you how they differ, and even after reading the help texts, it's still confusing.

For example, if "Enable Logging" is and "Enable Logger" is off, then that says that log calls will be executed even though they're not compiled in.

Not looking for an explanation. Just saying that it could probably be made clearer, e.g. with better prompts/help texts.

jukkar, 2019-03-11:
The "Enable Logging" is the legacy SYS_LOG logging. I thought we had already deprecated it.

nordic-krch, 2019-03-13:
@nashif should we remove sys_log at all? Do want that in LTS? It'sremoed from the documentation and not used in the tree.

jfischer-phytec-iot, 2019-03-13:
Please remove it or at last change the prompt.

ulfalizer, 2019-03-15:
@nordic-krch @jfischer-phytec-iot 
Would you be happy with something like "Enable logger" and "Enable legacy logger" in the meantime?

jfischer-phytec-iot, 2019-03-15:
>Would you be happy with something like "Enable logger" and "Enable legacy logger" in the meantime?

okay

nashif, 2019-03-23:
> @nashif should we remove sys_log at all? Do want that in LTS? It's removed from the documentation and not used in the tree.

yes, I would actually remove it altogether. Looks like we just forgot about it.

ulfalizer, 2019-03-28:
SYS_LOG was removed in https://github.com/zephyrproject-rtos/zephyr/pull/14883.

closed 2019-03-28 by ulfalizer