zephyrproject-rtos/zephyr #14244: zephyr.git/tests/crypto/rand32/testcase.yaml#crypto.rand32.random_hw_xoroshiro.rand32: Not Booting

cinlyooi-intel, 2019-03-11:
**Describe the bug**
Attempted on two separate boards, both failed to boot up.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=sam_e70_xplained
3. make
4. make run

**Environment (please complete the following information):**
 - OS:  fedora29
 - Toolchain: zephyr
 - Commit SHA or Version used: 7967829a261ef8ae21622710dd5699d30110af11

aurel32, 2019-03-11:
I am also not able to reproduce that bug :(.

> Attempted on two separate boards, both failed to boot up.

When you say two separate boards, do you mean two SAM E70 boards, or two different models?

cinlyooi-intel, 2019-03-13:
@aurel32 Two separate SAM E70 boards

Do you mean you managed to boot the board? and run the test correctly?

aurel32, 2019-03-13:
Yes, the board boots, but the test fails as you reported in #13935. In my initial testing I didn't modify the configuration file to select the xoroshiro generator instead of the default one.

cinlyooi-intel, 2019-03-13:
@aurel32 Practical consdierations over here means I am willing to accept that my boards here actually booted but I just did not see the full output to verify it boot. If you are fine with this, I think we can close the ticket. If you agree, can you close the ticket please? Many thanks for all your help.

wentongwu, 2019-03-21:
Cinly, it can boot, but it's the same issue as #13935.

wentongwu, 2019-03-26:
This is a duplicate issue of #13935. And it seems @aurel32 has already root caused it.

closed 2019-03-28 by rljordan