zephyrproject-rtos/zephyr #14246: ./sample/bluetooth/mesh/ always issue an "HARD FALUT"

kaiser-ren, 2019-03-11:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?
I compiled the code in ./sample/bluetooth/mesh/ for micro:bit board, after flashing the firmware into micro:bit, there is always a "HARD FAULT" when the board boot up, the fault is:
"***** HARD FAULT *****
***** Hardware exception *****
Current thread ID = 0x200011bc
Faulting instruction address = 0xaaaaaaaa
Fatal fault in ISR! Spinning…"

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
Please verify and fix it. 

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.
"***** HARD FAULT *****
***** Hardware exception *****
Current thread ID = 0x200011bc
Faulting instruction address = 0xaaaaaaaa
Fatal fault in ISR! Spinning…"

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used
Compiler: gcc-arm
Board: bbc_microbit
Commit: 7967829a261ef8ae21622710dd5699d30110af11 

**Additional context**
Add any other context about the problem here.

aescolar, 2019-03-11:
Hi @kaiser-ren , please do not copy the template/instructions on how to write the bug report into the bug report itself, but replace them with the actual details of the bug.
Specially clarify the steps one can follow to reproduce the issue.

kaiser-ren, 2019-03-11:
> Hi @kaiser-ren , please do not copy the template/instructions on how to write the bug report into the bug report itself, but replace them with the actual details of the bug.
> Specially clarify the steps one can follow to reproduce the issue.

Apologized, I haven't read the document clearly and careful, will avoid it happen again. 
Thanks.

jhedberg, 2019-03-11:
@kaiser-ren I can't reproduce this. I used `west` to build and flash:

```
west build -s samples/bluetooth/mesh -d mesh -b bbc_microbit
west flash -d mesh
```

After that the board boots up just fine:
```
***** Booting Zephyr OS v1.14.0-rc1-988-g7967829a261e *****
Initializing...
Bluetooth initialized
Mesh initialized
```

Do you have any provisioning information in flash? Is that a pre-requisite for reproducing this? Have you tried erasing all flash?

jhedberg, 2019-03-11:
@kaiser-ren btw, which version of Zephyr did you test this with? If it wasn't latest master I'd suggest trying with that as the first thing, since there's been some fine-tuning of various thread stack sizes in the past few weeks.

arn-soft, 2019-03-11:
@kaiser-ren I was having same issues and latest master resolves that. so, you can try by taking a git pull.

kaiser-ren, 2019-03-12:
> @kaiser-ren btw, which version of Zephyr did you test this with? If it wasn't latest master I'd suggest trying with that as the first thing, since there's been some fine-tuning of various thread stack sizes in the past few weeks.

kaiser-ren, 2019-03-12:
Sorry for chaos, clicked incorrect button.

kaiser-ren, 2019-03-12:
> @kaiser-ren btw, which version of Zephyr did you test this with? If it wasn't latest master I'd suggest trying with that as the first thing, since there's been some fine-tuning of various thread stack sizes in the past few weeks.

This is the commit I used: https://github.com/zephyrproject-rtos/zephyr/commit/7967829a261ef8ae21622710dd5699d30110af11 
It's the latest commit yesterday morning, I will try to git pull latest commit today.

kaiser-ren, 2019-03-12:
> @kaiser-ren I can't reproduce this. I used `west` to build and flash:
>
> ```
> west build -s samples/bluetooth/mesh -d mesh -b bbc_microbit
> west flash -d mesh
> ```
> After that the board boots up just [...]

I haven't used *west*, I still use this: "cmake -GNinja -DBOARD=bbc_microbit .."

kaiser-ren, 2019-03-12:
> @kaiser-ren I was having same issues and latest master resolves that. so, you can try by taking a git pull.

Thanks! I git pull the latest commit on master tree, 3aa8443ab41202f978258810961dbc5a74ad2727 , it's still the same.

kaiser-ren, 2019-03-12:
> > @kaiser-ren btw, which version of Zephyr did you test this with? If it wasn't latest master I'd suggest trying with that as the first thing, since there's been some fine-tuning of various thread stack sizes in the past few weeks.
> 
> This is the commit I used: [7967829](https://github.com/zephyrproject-rtos/zephyr/commit/7967829a261ef8ae21622710dd5699d30110af11)
> It's the latest commit yesterday morning, I will try to git pull latest commit today.

For this project, I think the source code and *west*/ should be okay, the only thing in my mind which can make it different is compiler, I'm using gcc-arm-none-eabi-5_3-2016q1, which compiler did you use for this project?

bfrog, 2019-03-12:
@kaiser-ren That does seem like a possibility. I had issues with several sample projects using newer versions of gcc which are confounding, using the same version in the Zephyr SDK fixed the sample projects for me previously when I first started working with zephyr.

kaiser-ren, 2019-03-12:
> For this project, I think the source code and _west_/ should be okay, the only thing in my mind which can make it different is compiler, I'm using gcc-arm-none-eabi-5_3-2016q1, which compiler did [...]

@jhedberg after upgrading my compile from *gcc-arm-none-eabi-5.3-2016q1* to *gcc-arm-none-eabi-8-2018-q4-major*, no hard fault there.

kaiser-ren, 2019-03-12:
> confounding

@bfrog 

> @kaiser-ren That does seem like a possibility. I had issues with several sample projects using newer versions of gcc which are confounding, using the same version in the Zephyr SDK fixed the sample projects for me previously when I first started working with zephyr.

@bfrog right, SDK may be a good choice for developers to have a consistent dev environment.

closed 2019-03-12 by kaiser-ren