zephyrproject-rtos/zephyr #14260: samples/drivers/flash_shell doesn't rely on drivers/flash/flash_shell

erwango, 2019-03-11:
**Is your enhancement proposal related to a problem? Please describe.**
samples/drivers/flash_shell implements an alternative to drivers/flash/flash_shell.
Hence, we're maintaining two files that do equivalent processing and the sample does not allow to validate the driver.

**Describe the solution you'd like**
Implement samples/drivers/flash_shell based on drivers/flash/flash_shell

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

erwango, 2021-02-05:
@nvlsianpu  I'm adding you as assignee. Feel free to close if you think this is not something we want.

erwango, 2022-01-03:
@nvlsianpu Feel free to close if you don't think this should be tackled.

nvlsianpu, 2022-01-03:
@erwango I think this ticket is still valid.

nashif, 2022-11-19:
This is really bug IMO

cfriedt, 2022-11-20:
Just some observations: 
- `tests/drivers/flash` does not seem very complete
  - only 1 test
  - gives false reassurance that drivers are properly implemented
- `drivers/flash/flash_shell.c` utility does not work on esp32-c3
  - not sure why, as it passes all 1 tests from `tests/drivers/flash`
- `samples/drivers/flash_shell` utility does work on esp32-c3
  - it's obviously doing something different than `drivers/flash/flash_shell.c`
  - solution is not as simple as implementing `samples/drivers/flash_shell` on top of `drivers/flash/flash_shell`
  - likely needs to be merged with some attention paid to detail
  - additional testsuite coverage would be nice!

erwango, 2023-12-19:
Done. Closing

closed 2023-12-19 by erwango