zephyrproject-rtos/zephyr #14265: Bluetooth GATT descriptor discovery returns all attributes

paulgavrikov, 2019-03-11:
**Describe the bug**
Running a GATT Descriptor discovery i.e. `bt_gatt_discover_params.type = BT_GATT_DISCOVER_DESCRIPTOR` results in the discovery callback being triggered for every attribute in the ATT table instead of only the descriptor.

**Expected behavior**
Instead of returning all attributes, the discovery shall only return descriptors (like service, characteristic discovery and similar)

**Impact**
Showstopper

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.9.5
 - Commit SHA: 1.13-branch@c478b5bb6e

Vudentz, 2019-03-11:
Setting type BT_GATT_DISCOVER_DESCRIPTOR uses Find Information which will find every attribute on the database since that doesn't contain a UUID like Read by Group Type to filter. Or you are suggesting we filter out the services and characteristic UUIDs? It is perhaps possible to do it but we still gonna need to discover so the only thing it will do is stopping the callbacks to be called, though applications can already do on the callback so it is more a convenient feature rather than a bug imo.

To properly implement a discovery it has to be done in 3 steps:

1. Find Services: Read By Group Type - Primary and Secondary
2. For each Service found in 1 Find its Characteristics: Read by Type - Characteristics
3. For each Characteristic found in 2 Find its Descriptors: Find Information

The reason why this is done that way is because Find Information can only find UUIDs of the same size, so as soon the server gets a vendor UUID using 128 bits format it will cause a lot more round trips in order to discover.

paulgavrikov, 2019-03-12:
@Vudentz PR looks good. While I generally agree with you it is confusing to users. Also, I'd suggest adding another discovery type that returns all attributes.

Vudentz, 2019-03-12:
@paulgavrikov if you please add you review to the PR that would probably make things faster. About the discovering of all attributes, I guess you are saying having a type which match the 3 steps I mentioned above? Unfortunatelly it is not that simple to implement that on Zephyr since that would require much more state tracking, and at least up to know there wasn't a requirement to implement the so called discover all procedure as zephyr don't support caching remote services. Another option would be to implement this in the shell so we don't have to incorporate code directly to the stack, but applications such as nrf connect, lightblue and BlueZ's bluetoothctl are imo much better alternatives in this respect.

commit 76deb657 (2019-03-13): Bluetooth: GATT: Skip certain attributes when discovering descriptors

When discovering descriptor Find Information procedure is used which
does not allow any filtering by the server so it will return all
attributes in the given range including services and characteristics.

Fixes #14265