zephyrproject-rtos/zephyr #14267: bt_le_adv_start() with sd_len=0

paulgavrikov, 2019-03-11:
**Describe the bug**
Calling `bt_le_adv_start` with `sd_len=0` and non-NULL sd will act as if scan response exists. 

**Expected behavior**
The function shall check if `sd != null || sd_len == 0`

**Additional context**
Add any other context about the problem here.

jhedberg, 2019-03-11:
@paulgavrikov I assume you're filing this against 1.13? The code has substantially changed since then, so the description doesn't really make sense when looking at the current code.

jhedberg, 2019-03-11:
Anyway, this seems like nonsense behavior from the application. All the current (master branch) does in this case is that it clears the scan response data. There are no buffer overflows or anything like that. So I'm not even sure whether this qualifies as a bug.

paulgavrikov, 2019-03-12:
@jhedberg yes, sorry I still wasn't able to check out 1.14. Problem is if you handle your advertising data dynamically and switch from scan response to no scan response you can do that easily by setting` sd_len=0` but zephyr will then still use `BT_LE_SCAN_IND` .

jhedberg, 2019-03-12:
@paulgavrikov ok, so that's a use case we haven't considered. I'm going to downgrade this to a feature request, but I do agree that we should try to find a solution to this.

jhedberg, 2019-03-19:
@paulgavrikov I'm looking again at existing master, and it seems to me like it should already behave like you want. Also note that we have new `bt_le_adv_update_data()` API which might suite your purpose better.

paulgavrikov, 2019-04-03:
@jhedberg Almost perfect there except: [](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/host/hci_core.c#L5217)

I think changing the `!sd `checks to `!sd_len` is a much safer way.

jhedberg, 2019-04-03:
> I think changing the !sd checks to !sd_len is a much safer way.

@paulgavrikov perhaps, though all the branch does is calls `sd_has_name()` and that function will not do anything if sd_len is 0. Anyway, if you feel strongly about this feel free to submit a PR :)

paulgavrikov, 2019-04-03:
@jhedberg Not that strongly :P

closed 2019-04-03 by paulgavrikov