zephyrproject-rtos/zephyr #14275: [ci.foundries.io] regression 4344e27 all: Update reserved function names

doanac, 2019-03-11:
**Describe the bug**
Since the merge of 4344e27 all ARM based builds in sanitycheck have been failing the "tests/benchmarks/app_kernel/benchmark.application" test:

https://ci.foundries.io/projects/zephyr/builds/5287

The build break as can be sane for [blenano2](https://ci.foundries.io/projects/zephyr/builds/5287/sanity-compile-nrf52/artifacts/console.log) is:
~~~
[ 96%] Linking C executable zephyr_prebuilt.elf
cd /tmp/outdir/nrf52_blenano2/tests/benchmarks/app_kernel/benchmark.application/zephyr && /usr/local/bin/cmake -E cmake_link_script CMakeFiles/zephyr_prebuilt.dir/link.txt --verbose=1
ccache /opt/toolchains/zephyr-sdk-0.9.5/sysroots/x86_64-pokysdk-linux/usr/bin/arm-zephyr-eabi/arm-zephyr-eabi-gcc    CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj  -o zephyr_prebuilt.elf -T linker.cmd -Wl,-Map=/tmp/outdir/nrf52_blenano2/tests/benchmarks/app_kernel/benchmark.application/zephyr/zephyr.map -u_OffsetAbsSyms -u_ConfigAbsSyms -Wl,--whole-archive ../app/libapp.a libzephyr.a arch/arch/arm/core/libarch__arm__core.a arch/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a arch/arch/arm/core/cortex_m/mpu/libarch__arm__core__cortex_m__mpu.a lib/libc/minimal/liblib__libc__minimal.a drivers/serial/libdrivers__serial.a -Wl,--no-whole-archive kernel/libkernel.a CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj -L"/opt/toolchains/zephyr-sdk-0.9.5/sysroots/armv5-zephyr-eabi/usr/lib/arm-zephyr-eabi/6.2.0/armv7e-m" -L/tmp/outdir/nrf52_blenano2/tests/benchmarks/app_kernel/benchmark.application/zephyr -lgcc -Wl,--print-memory-usage -mthumb -nostdlib -static -no-pie -Wl,--fatal-warnings -Wl,-X -Wl,-N -Wl,--gc-sections -Wl,--build-id=none -Wl,--orphan-handling=warn -mabi=aapcs 
Memory region         Used Size  Region Size  %age Used
           FLASH:       26496 B       512 KB      5.05%
            SRAM:       22732 B        64 KB     34.69%
        IDT_LIST:          40 B         2 KB      1.95lib/libc/minimal/liblib__libc__minimal.a(stdout_console.c.obj): In function `z_zephyr_fwrite':
/tmp/outdir/nrf52_blenano2/tests/benchmarks/app_kernel/benchmark.application/zephyr/include/generated/syscalls/libc-hooks.h:19: undefined reference to `z_impl_z_zephyr_fwrite'
~~~

galak, 2019-03-11:
This builds fine for me on ARM platforms:

`./scripts/sanitycheck -T tests/benchmarks/app_kernel -S -l -b`

```
Cleaning output directory /home/galak/git/zephyr/sanity-out
Selecting all possible platforms per test case
Building testcase defconfigs...
106 tests selected, 214 tests discarded due to filters
total complete:  106/ 106  100%  failed:    0
106 of 106 tests passed with 0 warnings in 102 seconds
```

pfl, 2019-03-12:
Interesting, shippable is missing this part from its tests as shippable was fine with the original renaming patch.

PR #14277 (2019-03-12): libc: Fix fwrite function name
Commit 4344e27c267e1139c26dbd9cafcca4b3069ea821 changed the reserved
function names, but got the naming wrong for fwrite.  Change fwrite to
match fputc naming and fixes build issues.

Fixes #14275
commit c82f23ca (2019-03-13): libc: Fix fwrite function name

Commit 4344e27c267e1139c26dbd9cafcca4b3069ea821 changed the reserved
function names, but got the naming wrong for fwrite.  Just use the
name zephyr_fwrite everywhere.

Fixes #14275
closed 2019-03-12 by galak