zephyrproject-rtos/zephyr #14283: tests/drivers/watchdog/wdt_basic_api fails on test_wdt_callback_1() for Ardruino

cinlyooi-intel, 2019-03-12:
**Describe the bug**
`test_wdt: test_wdt_callback_1() == TC_PASS is false` for Arduino_Due
[Note: Same as #13347 but that was for QuarkSE/ARC. Not changing that bug because a PR was already created for that]

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -GNinja  -DBOARD=arduino_due -DARCH=arm 
3. ninja
4. ninja run

**Screenshots or console output**
```
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
 Testcase: test_wdt_callback_1
 Assertion failed at /home/cinlyooi/tests/zephyr.git/tests/drivers/watchdog/wdt_basic_api/src/test_wdt.c:308: test_wdt: test_wdt_callback_1() == TC_PASS is false

 FAIL - test_wdt
===================================================================

```
**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: zephyr
 - Commit SHA or Version used: 3aa8443ab41202f978258810961dbc5a74ad2727

cinlyooi-intel, 2019-03-13:
Additional observation on quark_se_c1000_ss_devboard:arc logged

SavinayDharmappa, 2019-03-13:
for  quark_se_c1000_ss_devboard this testcase bypassed please refer the #14067 .

cinlyooi-intel, 2019-03-15:
Reproduced for arduino_due using commit 96235c5027461534b75059a65f8833231d293312

SavinayDharmappa, 2019-03-15:
@cinlyooi-intel  https://github.com/zephyrproject-rtos/zephyr/pull/14466  PR should fix issue on arduino_due as well .

overheat, 2019-03-28:
@cinlyooi-intel Hi, I think #14968 will resolve those issues on ATSAMxxxx based boards, like arduino_due(which I have tested)

ManojSubbarao, 2019-04-08:
@cinlyooi-intel  can this be closed if issue is fixed?

closed 2019-08-19 by nashif