zephyrproject-rtos/zephyr #14288: USB CDC ACM has low throughput

masz-nordic, 2019-03-12:
**Is your enhancement proposal related to a problem? Please describe.**
To reproduce:
- Disable loopback in `cdc_acm` sample by commenting out `write_data(dev, data_buf, *bytes_read);`.
- Build and flash `cdc_acm` sample.
- `pv --force -B 256 sample.txt 2>&1 > /dev/ttyACM1` where sample.txt is a big file.
- Observe the throughput.

Results:
- `cdc_acm` without loopback: 51 KB/s
- as above, with disabled logs: 96 KB/s

**Describe the solution you'd like**
CDC ACM should reach throughput of at least 400 kB/s.

**Additional context**
Board: nrf52840_pca10056
OS: Ubuntu 16.04, 4.13.0
Zephyr: c78c5e693679cc75f6e72e55c70f179ae9f6bea8

masz-nordic, 2019-03-12:
The bottleneck seems to be located [in this read](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/usb/class/cdc_acm.c#L339).

finikorg, 2019-03-12:
> The bottleneck seems to be located [in this read](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/usb/class/cdc_acm.c#L339).

Since we do not have it for other classes, we can remove it. This should be handled in the dc code.

carlescufi, 2019-03-13:
@finikorg can you send a PR to remove that read?

finikorg, 2019-03-19:
> @finikorg can you send a PR to remove that read?

PR #14677

PR #14677 (2019-03-25): USB: cdc_acm: use ringbuf library and fixes
Remove old artefacts and use ringbuf for RX path.

Fixes #14697
Fixes #14698
Might Fixes #14288
commit cd95560c (2019-03-25): usb: cdc_acm: Use ringbuf for RX path

Use ringbuf library for handling RX CDC ACM path. Remove old code
artifacts handling specific hardware.

Fixes #14288
closed 2019-03-25 by carlescufi