zephyrproject-rtos/zephyr #14289: Cannot build GRUB2 boot loader image in Clear Linux

gvancuts, 2019-03-12:
**Describe the bug**
It is not possible to build the GRUB2 boot loader image as described in the [here](https://docs.zephyrproject.org/latest/boards/x86/up_squared/doc/index.html#creating-a-grub2-boot-loader-image-from-a-linux-host) on a Clear Linux (ver. 28230) system.

**To Reproduce**
Steps to reproduce the behavior:
1. Make sure you have [this patch](https://github.com/zephyrproject-rtos/zephyr/pull/14281) (still being reviewed) in your tree
2. Run: `$ ./boards/x86/common/scripts/build_grub.sh x86_64`
3. This results in the following error (snipped):
```
    -e "/OUTPUT_TERMINAL_LIST_MARKER *( *\"/{s/.*( *\"\([^\"]*\)\".*/o\1: $b/;p;}" $pp; \
done) | sort -u > terminal.lst
In file included from ../include/grub/dl.h:29,
                 from loader/i386/bsd.c:26:
loader/i386/bsd.c: In function ‘grub_cmd_openbsd’:
../include/grub/misc.h:71:10: error: ‘ptr’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
   return grub_memmove (dest, src, n);
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
loader/i386/bsd.c:264:9: note: ‘ptr’ was declared here
   void *ptr;
         ^~~
In file included from ../include/grub/dl.h:29,
```

**Expected behavior**
`grub_x86_64.efi` is correctly generated.

**Impact**
Clear Linux cannot be used as the development machine. The other environment that I have tested was Fedora 29 (with [this patch](https://github.com/zephyrproject-rtos/zephyr/pull/14281)), that works correctly.

**Environment (please complete the following information):**
 - OS: Clear Linux 28230
 - Toolchain: native tools

andrewboie, 2019-03-15:
Your referenced patch got merged, may we close this?

gvancuts, 2019-03-15:
> Your referenced patch got merged, may we close this?

My bad, I forgot to add the `Fixes:` tag... sorry about that.

gvancuts, 2019-03-27:
This seems to have come back (sigh), I cannot identify what changed in the meantime and the failure has been reported by different sources. I've tried the latest Clear Linux version (as of today: 28510) and the problem is reproducible in that environment too... reopening.

gvancuts, 2019-03-27:
One of the `CFLAGS` setting in Clear Linux is causing this, if you unset it before calling the script, it will successfully generate the `grub_x86_64.efi` file.

A (pretty unrefined) patch would be to unset this from the script directly, i.e.:
```diff --git a/boards/x86/common/scripts/build_grub.sh b/boards/x86/common/scripts/build_grub.sh
index 11b2067fe5..9b5b0801e4 100755
--- a/boards/x86/common/scripts/build_grub.sh
+++ b/boards/x86/common/scripts/build_grub.sh
@@ -4,6 +4,7 @@
 # https://github.com/otcshare/contiki-x86
 
 set -e
+unset CFLAGS
 
 JOBS=5
 SCRIPT_DIR=$( cd "$( dirname "${BASH_SOURCE[0]}" )" && pwd )
```

**Note:** this only unsets `CFLAGS` in the context of the script. Clear Linux environment still has it after you run the script.

PR #15016 (2019-03-31): boards/x86: scripts: unset CFLAGS when running build_grub.sh
On Clear Linux, the CFLAGS is set and defines a number of aggressive checks and
optimizations. This causes a build failure when generating a GRUB2 boot loader
image using the 'build_grub.sh' script.

Unsetting it within the script allows it to proceed and successfully build a
functional GRUB2 boot loader image to be used with Zephyr.

Fixes: #14289
commit 5900a6fe (2019-03-31): boards/x86: scripts: unset CFLAGS when running build_grub.sh

On Clear Linux, the CFLAGS is set and defines a number of aggressive
checks and optimizations. This causes a build failure when generating a
GRUB2 boot loader image using the 'build_grub.sh' script.

Unsetting it within the script allows it to proceed and successfully
build a functional GRUB2 boot loader image to be used with Zephyr.

Fixes: #14289
closed 2019-03-31 by nashif