zephyrproject-rtos/zephyr #14293: mimxrt1060_evk tests/benchmarks/latency_measure failed

hakehuang, 2019-03-12:
**Describe the bug**
Error: tick occurred  during context switch benchmark testing

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1060_evk ..
3. make
4. See error

**Expected behavior**
all pass

**Impact**
benchmark

**Screenshots or console output**
```
***** Booting Zephyr OS 1.14.0-rc1 *****

|-----------------------------------------------------------------------------|

|                            Latency Benchmark                                |

|-----------------------------------------------------------------------------|

|  tcs = timer clock cycles: 1 tcs is 1 nsec                                  |

|-----------------------------------------------------------------------------|

| 1 - Measure time to switch from ISR back to interrupted thread              |

| switching time is 5569 tcs = 9281 nsec                                      |

|-----------------------------------------------------------------------------|

| 2 - Measure time from ISR to executing a different thread (rescheduled)     |

| switch time is 7266 tcs = 12110 nsec                                        |

|-----------------------------------------------------------------------------|

| 3 - Measure average time to signal a sema then test that sema               |

| Average semaphore signal time 2432 tcs = 4053 nsec                          |

| Average semaphore test time 175 tcs = 291 nsec                              |

|-----------------------------------------------------------------------------|

| 4- Measure average time to lock a mutex then unlock that mutex              |

| Average time to lock the mutex 590559 tcs = 984265 nsec                     |

| Average time to unlock the mutex 2916 tcs = 4860 nsec                       |

|-----------------------------------------------------------------------------|

| 5 - Measure average context switch time between threads using (k_yield)     |

| Average thread context switch using yield 4423 tcs = 7372 nsec              |

|-----------------------------------------------------------------------------|

| 6 - Measure average context switch time between threads (coop)              |

| Error: tick occurred                                                        |

|-----------------------------------------------------------------------------|
```

**Environment (please complete the following information):**
 - OS: ( Linux)
 - Toolchain ( Zephyr SDK)
 - Commit SHA or Version used: v1.14.0_rc1

hakehuang, 2019-03-12:
@MaureenHelm RT1050 does not have such problem
'''
***** Booting Zephyr OS 1.14.0-rc1 *****

|-----------------------------------------------------------------------------|

|                            Latency Benchmark                                |

|-----------------------------------------------------------------------------|

|  tcs = timer clock cycles: 1 tcs is 1 nsec                                  |

|-----------------------------------------------------------------------------|

| 1 - Measure time to switch from ISR back to interrupted thread              |

| switching time is 3221 tcs = 5368 nsec                                      |

|-----------------------------------------------------------------------------|

| 2 - Measure time from ISR to executing a different thread (rescheduled)     |

| switch time is 3874 tcs = 6456 nsec                                         |

|-----------------------------------------------------------------------------|

| 3 - Measure average time to signal a sema then test that sema               |

| Average semaphore signal time 324 tcs = 540 nsec                            |

| Average semaphore test time 34 tcs = 57 nsec                                |

|-----------------------------------------------------------------------------|

| 4- Measure average time to lock a mutex then unlock that mutex              |

| Average time to lock the mutex 588021 tcs = 980035 nsec                     |

| Average time to unlock the mutex 340 tcs = 566 nsec                         |

|-----------------------------------------------------------------------------|

| 5 - Measure average context switch time between threads using (k_yield)     |

| Average thread context switch using yield 404 tcs = 674 nsec                |

|-----------------------------------------------------------------------------|

| 6 - Measure average context switch time between threads (coop)              |

| Average context switch time is 418 tcs = 697 nsec                           |

|-----------------------------------------------------------------------------|

===================================================================

PROJECT EXECUTION SUCCESSFUL

'''

MaureenHelm, 2019-03-25:
I'm not able to reproduce. @dleach02 will you please give it a try?

```
***** Booting Zephyr OS v1.14.0-rc1-1408-g2d43d181bf85 *****
|-----------------------------------------------------------------------------|
|                            Latency Benchmark                                |
|-----------------------------------------------------------------------------|
|  tcs = timer clock cycles: 1 tcs is 1 nsec                                  |
|-----------------------------------------------------------------------------|
| 1 - Measure time to switch from ISR back to interrupted thread              |
| switching time is 4053 tcs = 6755 nsec                                      |
|-----------------------------------------------------------------------------|
| 2 - Measure time from ISR to executing a different thread (rescheduled)     |
| switch time is 4218 tcs = 7030 nsec                                         |
|-----------------------------------------------------------------------------|
| 3 - Measure average time to signal a sema then test that sema               |
| Average semaphore signal time 428 tcs = 713 nsec                            |
| Average semaphore test time 713 tcs = 1188 nsec                             |
|-----------------------------------------------------------------------------|
| 4- Measure average time to lock a mutex then unlock that mutex              |
| Average time to lock the mutex 444 tcs = 740 nsec                           |
| Average time to unlock the mutex 439 tcs = 732 nsec                         |
|-----------------------------------------------------------------------------|
| 5 - Measure average context switch time between threads using (k_yield)     |
| Average thread context switch using yield 1718 tcs = 2864 nsec              |
|-----------------------------------------------------------------------------|
| 6 - Measure average context switch time between threads (coop)              |
| Average context switch time is 781 tcs = 1302 nsec                          |
|-----------------------------------------------------------------------------|
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

hakehuang, 2019-03-28:
test setting issue, close it. @MaureenHelm @galak

closed 2019-03-28 by hakehuang

hakehuang, 2019-03-29:
@MaureenHelm  I find the root cause of this problem is:
I set the SDRAM clock to 166M the context switch will be very slow, exceed the 1s limitation. and after raise the SDRAM clock them this issue will be resolved.