zephyrproject-rtos/zephyr #14295: USB MSC sample crashes

masz-nordic, 2019-03-12:
**Describe the bug**
USB mass storage sample crashes during initialization. Device is not detected by the host.

**To Reproduce**
- `cd samples/subsys/usb/mass`
- `west build --board nrf52840_pca10056`
- `west flash`

**Expected behavior**
Proper initialization, a disk should be detected.

**Impact**
Showstopper.

**Screenshots or console output**
```
***** Booting Zephyr OS v1.14.0-rc1-1025-gc78c5e693679 *****                    
[00:00:00.004,272] <err> usb_msc: Storage init ERROR !!!! - Aborting USB init   
[00:00:00.004,272] <inf> main: The device is put in USB mass storage mode.   
```
**Environment (please complete the following information):**
 - OS: Ubuntu 16.04, 4.13.0
 - Toolchain: Zephyr SDK 0.9.5
 - Zephyr: c78c5e693679cc75f6e72e55c70f179ae9f6bea8

aurel32, 2019-03-12:
This is expected, you need to either add the `overlay-ram-disk.conf` or the `overlay-flash-disk.conf` as explained in the README.

masz-nordic, 2019-03-12:
Right. In case of samples, I think it would be beneficial to have a default `.conf` (or at least have a board-specific configuration for most used ones).

finikorg, 2019-03-12:
I think that the build is also described in `samples/subsys/usb/mass/sample.yaml` But we could also add `#error ....` code somewhere in mass storage code to warn user about right usage

PR #14299 (2019-03-13): usb: samples: mass: add warning for no conf and nrf52840 conf
Added a warning that will abort build without disks.
Added a default configuration for nrf52840_pca10056.

Fixes #14295
closed 2019-03-13 by galak