zephyrproject-rtos/zephyr #14297: mimxrt1020_evk tests/kernel/gen_isr_table test failure

hakehuang, 2019-03-12:
**Describe the bug**
interrupt 159 didn't run once, ran 0 times

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1030_evk
3. make
4. See error

**Expected behavior**
Pass

**Impact**
irq services

**Screenshots or console output**
```
***** Booting Zephyr OS 1.14.0-rc1 *****

starting test - Test gen_isr_tables

IRQ configuration (total lines 160):

isr1 isr=0x60002679 irq=159

isr2 isr=0x600026a5 irq=158

Checking _irq_vector_table entry 159 for irq 159

triggering irq 159

interrupt 159 didn't run once, ran 0 times

FAIL - main

===================================================================

===================================================================

PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: (Linux)
 - Toolchain (Zephyr SDK)
 - Commit SHA or Version used: v1.14.0_rc1

hakehuang, 2019-03-12:
@MaureenHelm  rt1060 and rt1050 all pass,but Rt1020 does not.

MaureenHelm, 2019-03-25:
I can reproduce this, but I can't explain yet why rt1020 behaves differently from rt1050 and rt1060. According to their respective hardware reference manuals, all three parts have 160 vectors, with the last few vectors reserved. The number of reserved vectors at the end of the table varies by part, with the rt1020 having the most reserved. I suspect that rt1020 may actually have less than 160 vectors, but I need to check with the hardware team.

PR #14921 (2019-03-27): soc: nxp_imx: Configure NUM_IRQS independently for each rt soc
Not all i.mx rt socs have the same number of irqs, so move the default
configuration from the soc series level to the individual socs. The
rt1020 hardware reference manual (IMXRT1020RM Rev.1 12/2018) incorrectly
documents 160 irqs (#142-159 reserved), but the soc actually has 142
irqs.

Fixes tests/kernel/gen_isr_table for the mimxrt1020_evk board.

Fixes #14297
closed 2019-03-27 by nashif