zephyrproject-rtos/zephyr #14300: Bluetooth connection using central and peripheral samples in nrf52840

ejpark0618, 2019-03-12:
Hi, I'm doing my research on BLE using nrf52840 board and Zephyr

However, whenever I tried to make connection using Central and Peripheral sample, 
it is repeatedly connected and disconnected with error code 22 and 19

Can I get some help on this? Thank you for your attention

Vudentz, 2019-03-13:
Are there any traces available for this issue?

carlescufi, 2019-03-13:
@ejpark0618 those are normal termination codes, someone has disconnected on the local or remote side. This is not a bug, it's either side disconnecting normally.

ejpark0618, 2019-03-14:
> Are there any traces available for this issue?
[image]
[image]
 Here it is

ejpark0618, 2019-03-14:
> @ejpark0618 those are normal termination codes, someone has disconnected on the local or remote side. This is not a bug, it's either side disconnecting normally.

Yes, but we do nothing on central and peripheral codes..... we did not disconnected by hand

closed 2020-10-04 by github-actions[bot]