zephyrproject-rtos/zephyr #14302: USB MSC fails USB3CV tests

masz-nordic, 2019-03-12:
**Describe the bug**
USB mass storage sample with RAM disk fails USB3CV (#11803) MSC specific tests:
- [ ] Error Recovery
- [ ] Case 3
- [ ] Case 12
- [ ] CB Length
- [ ] Command Set

**To Reproduce**
Flash `mass` sample with RAM overlay to nrf52840_pca10056.

**Expected behavior**
USB3CV MSC tests pass.

**Impact**
No immediate impact, but this has to be solved before MSC class is considered ready.

**Screenshots or console output**
Three unique errors are thrown in tests:

1. Error Recovery
```
Issuing required command (Test Unit Ready) to verify device has recovered
Issuing CBW (attempt #1):
|----- CBW LUN                  = 0x0
|----- CBW Flags                = 0x0
|----- CBW Data Transfer Length = 0x0
|----- CBW CDB Length           = 0x6
|----- CBW CDB-00 = 0x0
|----- CBW CDB-01 = 0x0
|----- CBW CDB-02 = 0x0
|----- CBW CDB-03 = 0x0
|----- CBW CDB-04 = 0x0
|----- CBW CDB-05 = 0x0
Issuing CSW : try 1
CSW Bulk Request timed out!

ERROR
Failed CSW phase : should have been success or stall
FAIL
(5.3.4) The CSW status value must be 0x00, 0x01, or 0x02.

Re-enumerating device
Detected hard error, issuing BOT MSC Reset

ERROR
Failed to execute required command (Test Unit Ready) after BOT MSC Reset
FAIL
(5.7.2) All devices must either support or properly fail the commands marked 
 as optional in 4.2 based on PDT. Devices that support or properly fail all the commands 
 marked as optional for Bootable devices will be reported as Bootable.
```

2. Case 3, Case 12, CB Length, Command Set:
```
Issuing INQUIRY
Issuing CBW (attempt #1):
|----- CBW LUN                  = 0x0
|----- CBW Flags                = 0x80
|----- CBW Data Transfer Length = 0x24
|----- CBW CDB Length           = 0x6
|----- CBW CDB-00 = 0x12
|----- CBW CDB-01 = 0x0
|----- CBW CDB-02 = 0x0
|----- CBW CDB-03 = 0x0
|----- CBW CDB-04 = 0x24
|----- CBW CDB-05 = 0x0
Issuing DATA IN
DATA Bulk Request timed out!

ERROR
Failed DATA phase : should have been success or stall!
FAIL
(5.2.3) Devices must consider the CBW meaningful if no reserved bits 
 are set, the LUN number indicates a LUN supported by the device, 
 bCBWCBLength is in the range of 1 through 16, and the length and 
 content of the CBWCB field are appropriate to the SubClass.

Re-enumerating device
Detected hard error, issuing BOT MSC Reset

ERROR
Failed INQUIRY request
FAIL
(5.7.1) All devices must support the commands marked as required in 4.2 based 
 on PDT. Devices that support all the commands marked as required for Bootable devices will 
 be reported as Bootable.
```

3. Command Set:
```
Issuing Command Set Test for Op Code 0x12, Test Variation #3
Issuing CBW (attempt #1):
|----- CBW LUN                  = 0x0
|----- CBW Flags                = 0x80
|----- CBW Data Transfer Length = 0x2
|----- CBW CDB Length           = 0x6
|----- CBW CDB-00 = 0x12
|----- CBW CDB-01 = 0x0
|----- CBW CDB-02 = 0x0
|----- CBW CDB-03 = 0x0
|----- CBW CDB-04 = 0x2
|----- CBW CDB-05 = 0x0
Issuing DATA IN
DATA Bulk Request timed out!

ERROR
Failed DATA phase : should have been success or stall!

FAIL
(5.2.3) Devices must consider the CBW meaningful if no reserved bits 
 are set, the LUN number indicates a LUN supported by the device, 
 bCBWCBLength is in the range of 1 through 16, and the length and 
 content of the CBWCB field are appropriate to the SubClass.

ERROR
READ CAPACITY(10) not supported.
```

**Environment (please complete the following information):**
 - OS: Windows x64 7 Professional SP1
 - PC: Dell Optiplex 9020
 - DK: nrf52840_pca10056
 - Toolchain: Zephyr SDK 0.9.5
 - Zephyr b87920bf3c789ad4eeebb749d9a06ae72f15d589

jfischer-phytec-iot, 2019-09-16:
@emob-nordic Can you rerun USB3CV MSC specific test, please?

emob-nordic, 2019-09-17:
@jfischer-phytec-iot Same test have failed:

- Error Recovery
- Case 3
- Case 12
- CB Length
- Command Set

emob-nordic, 2020-06-18:
USB MSC fails USB3CV test with nRF52840dk (nRF52840dk_nrf52840).
1) Command Set
```
Issuing DATA OUT
Issuing CSW : try 1
CSW residue returned = 0x0
CSW status returned = 0x0
Issuing Command Set Test for Op Code 0xaa, Test Variation #5
Issuing CBW (attempt #1):
|----- CBW LUN                  = 0x0
|----- CBW Flags                = 0x0
|----- CBW Data Transfer Length = 0x10000
|----- CBW CDB Length           = 0xc
|----- CBW CDB-00 = 0xaa
|----- CBW CDB-01 = 0x0
|----- CBW CDB-02 = 0x0
|----- CBW CDB-03 = 0x0
|----- CBW CDB-04 = 0x0
|----- CBW CDB-05 = 0x0
|----- CBW CDB-06 = 0x0
|----- CBW CDB-07 = 0x0
|----- CBW CDB-08 = 0x0
|----- CBW CDB-09 = 0x80
|----- CBW CDB-10 = 0x0
|----- CBW CDB-11 = 0x0
Issuing DATA OUT
DATA Bulk Request timed out!
Failed DATA phase : should have been success or stall!
(MSC: 5.2.3) Devices must consider the CBW meaningful if no reserved bits are set, the LUN number indicates a LUN supported by the device, bCBWCBLength is in the range of 1 through 16, and the length and content of the CBWCB field are appropriate to the SubClass.
Re-enumerating device
Detected hard error, issuing BOT MSC Reset

+ + + + + 

Getting Device Type
Issuing INQUIRY
Issuing CBW (attempt #1):
|----- CBW LUN                  = 0x0
|----- CBW Flags                = 0x80
|----- CBW Data Transfer Length = 0x24
|----- CBW CDB Length           = 0x6
|----- CBW CDB-00 = 0x12
|----- CBW CDB-01 = 0x0
|----- CBW CDB-02 = 0x0
|----- CBW CDB-03 = 0x0
|----- CBW CDB-04 = 0x24
|----- CBW CDB-05 = 0x0
CBW Bulk Request timed out!
Failed CBW phase, should have been successful!
(MSC: 5.2.3) Devices must consider the CBW meaningful if no reserved bits are set, the LUN number indicates a LUN supported by the device, bCBWCBLength is in the range of 1 through 16, and the length and content of the CBWCB field are appropriate to the SubClass.
Re-enumerating device
Detected hard error, issuing BOT MSC Reset
Failed INQUIRY request
(MSC: 5.7.1) All devices must support the commands marked as required in 4.2 based on PDT.  Devices that support all the commands marked as required for Bootable devices will be reported as Bootable.

READ CAPACITY(16) not supported.

```
The issue was introduced by #24696. cc @pabigot.

pabigot, 2020-06-18:
@emob-nordic Please open a new issue for this as the original issue was fixed.  I'll take a look to see what might have caused the problem.

closed 2020-06-18 by pabigot

pabigot, 2020-06-18:
@emob-nordic never mind, I've done it: #26275

PR #23307 (2020-03-10): usb: mass_storage: cleanup, fixes to pass USBCV MSC Test
 usb: mass_storage: cleanup, fixes to pass USBCV MSC Test

drivers: usb_dc_nrfx: always reset data toggle in usb_dc_ep_enable 
  Data toggle should be reset after SetConfiguration or SetInterface request.
  Always reset data toggle in usb_dc_ep_enable.
  (fixes Case x from #14302 issue)

usb: mass_storage:
  Use sys_get_be32, sys_put_be32 where possible.
  Check dCBWDataTransferLength for every read command.
   (fixes Command Set Test #14302 issue)

cherry-picked from #23188 and #23191 and tested on nRF52840
resolves #14302 for nRF drive [... cut]