zephyrproject-rtos/zephyr #14306: Automatic wake-up from system-level sleep modes

pizi-nordic, 2019-03-12:
At the moment, the system-level power management assumes that system timer is always able trigger exit from sleep modes. Unfortunately this is not true for most of the SoCs. Only few dedicated peripherals could wake-up the system from most of the sleep modes and options are even more limited for deep-sleep modes where many SoCs need external device for wake-up.

Things needed:
- [ ] API allowing to assign a counter device (RTC/Timer) which will be used to wake-up source from given power mode(s).
- [ ] Ability to switch system clock device to the selected one for the sleep duration (https://github.com/zephyrproject-rtos/zephyr/issues/12068 might help here)

wentongwu, 2019-03-13:
Not only counter device can do the wakeup,  it highly depends on soc design

pizi-nordic, 2019-03-19:
Yes, but the kernel could automatically handle only timers.

gmarull, 2023-05-30:
Closing as obsolete, open new issue with more details if needed.

closed 2023-05-30 by gmarull