zephyrproject-rtos/zephyr #14308: Better integration between system and device power modes.

pizi-nordic, 2019-03-12:
Currently the integration between system and device power management is not portable and very limited.
We need an API which allow:
- Select which devices should be suspended and which ones should remain active in given power mode(s).
- Select which devices should be configured as wake-up source in given power mode(s).

nashif, 2021-03-12:
This is similar to how Linux does things using `device_set_wakeup_capable()` and `device_set_wakeup_enable()`, defined in [include/linux/pm_wakeup.h](https://elixir.bootlin.com/linux/latest/source/include/linux/pm_wakeup.h).

ceolin, 2021-09-22:
This was implemented in 8eceeee798743d25ac882cf0374cb062a435cefe

closed 2021-09-22 by ceolin