zephyrproject-rtos/zephyr #14309: Automatic device dependency tracking

pizi-nordic, 2019-03-12:
We need abstraction of dependencies between devices in order to perform power management operations in correct order and on all devices affected by given operation. For example when suspending a SPI device, we might also suspend SPI Controller (driving SCLK, MISO and MOSI) and GPIO controller (driving /CS) if they are no longer used.

All data we need is in the DTS. We have to extract this and convert to a form which will be easy to use by the device power management code.

nashif, 2021-02-22:
should we close this?

pabigot, 2021-02-22:
Seems like this is done, yes.

closed 2021-02-22 by pabigot

ball-hayden, 2021-03-11:
@nashif / @pabigot - do you have a link to the PR that fixed this issue, please?
(I'm trying to trace whether this has landed for our fork yet)

pabigot, 2021-03-11:
#32127 supplies the necessary information.  There's very little infrastructure to actually use this data yet, and the API is still evolving (#32547).  But the data from DTS is now available, so this is closed.

ball-hayden, 2021-03-11:
Thank you. To clarify, then, the example you give ("when suspending a SPI device, we might also suspend SPI Controller") is _not_ yet implemented?

pabigot, 2021-03-11:
> To clarify, then, the example you give ("when suspending a SPI device, we might also suspend SPI Controller") is _not_ yet implemented?

Correct, no drivers have been updated to do this.  I expect it will ultimately be done using the device power management API, which is currently being discussed elsewhere (I don't know exactly where).