zephyrproject-rtos/zephyr #14397: [Coverity CID :195871]Integer handling issues in /drivers/counter/counter_ll_stm32_rtc.c

aasthagr, 2019-03-12:
Static code scan issues seen in File: /drivers/counter/counter_ll_stm32_rtc.c
 Category: Integer handling issues
 Function: rtc_stm32_read
 Component: Drivers
 CID: 195871
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

erwango, 2019-03-15:
Cf analysis in #14415.
Will be fixed when upgrading SDK to 0.10.0.

ceolin, 2019-03-21:
@erwango actually coverity spot this problem even with latest sdk. Question, why don't you cast this way:
```
ticks = counter_us_to_ticks(dev, (u64_t)ts * USEC_PER_SEC);
```
Here the we ensure that the multiplication won't overflow

ceolin, 2019-03-21:
actually you don't need the cast at all, `ts` is `time_t`

ceolin, 2019-03-21:
https://github.com/zephyrproject-rtos/zephyr/pull/14799

erwango, 2019-03-22:
@ceolin , cf my comment in #14799, none of the proposal works actually.

galak, 2019-03-26:
Can we close this now?

erwango, 2019-03-27:
Fixed by #14799

closed 2019-03-27 by erwango