zephyrproject-rtos/zephyr #14415: [Coverity CID :195793]Insecure data handling in /drivers/counter/counter_ll_stm32_rtc.c

aasthagr, 2019-03-12:
Static code scan issues seen in File: /drivers/counter/counter_ll_stm32_rtc.c
 Category: Insecure data handling
 Function: rtc_stm32_read
 Component: Drivers
 CID: 195793
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

erwango, 2019-03-14:
@galak, coverity complains on :
```
	time_t ts;
...
	ticks = counter_us_to_ticks(dev, (u64_t)(ts * USEC_PER_SEC));
```
With following comment:
```
CID 195871 (#1 of 1): Unintentional integer overflow 
(OVERFLOW_BEFORE_WIDEN)overflow_before_widen: Potentially overflowing expression ts * 
1000000UL with type unsigned long (32 bits, unsigned) is evaluated using 32-bit arithmetic,
and  then used in a context that expects an expression of type u64_t (64 bits, unsigned).
   To avoid overflow, cast either ts or 1000000UL to type u64_t rather than casting
 the result of the potential overflow.
```
But, sizeof(time_t) returns 8...

So actually this piece of code doesn't work anymore (it used to), and I need to cast ts on u32_t to get it back working. Is that possible time_t has changes from 4 to 8 bytes recently?

GeorgeCGV, 2019-03-14:
For me with `newlibc` (_required by stm32 rtc driver_) with arm none eabi 7-2017-q4-major toolchain `sizeof ` types `u32_t`, `s32_t`, and `time_t` are all `4` bytes.

erwango, 2019-03-15:
> with arm none eabi 7-2017-q4-major toolchain sizeof types u32_t, s32_t, and time_t are all 4 bytes

On my side, I'm using SDK0.10.0

erwango, 2019-03-15:
Coverity analysis assumes time_t is 4 bytes while it has been updated to 8 bytes with SDK0.10.0.
Report does not apply anymore with 8 byte time_t, so issue will be fixed by upgrading SDK to 0.10.0 at next run.
I'm not able to update coverity report with this anlaysis (rights issue?), @aasthagr can you do it?

GeorgeCGV, 2019-03-15:
newlib time_t has been changed to 64 bits in commit [4de8754bac676fc7f29082ccf8281b1c73ce629e ](https://www.sourceware.org/git/gitweb.cgi?p=newlib-cygwin.git;a=commit;h=4de8754bac676fc7f29082ccf8281b1c73ce629e)back in September 2017.

SDK 0.10.0 uses newlib 3.1.0.
And, as it seems from the release notes, amr none eabi 7-2017-q4-major shall already include the newlib with the 64 bits time_t (_but not in versions prior 7-2017-q4-major_).

closed 2019-03-21 by erwango