zephyrproject-rtos/zephyr #14454: tests/kernel/threads/no-multithreading/:  Repeated delay boot banner

cinlyooi-intel, 2019-03-13:
**Describe the bug**
Instead of seeing "It works", we see multiple '***** delaying boot 1000ms (per build configuration) *****'

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=same_e70_xplained
3. make
4. make run

**Screenshots or console output**
```
***** delaying boot 1000ms (per build configuration) *****
***** delaying boot 1000ms (per build configuration) *****
***** delaying boot 1000ms (per build configuration) *****
***** delaying boot 1000ms (per build configuration) *****
***** delaying boot 1000ms (per build configuration) *****
***** delaying boot 1000ms (per build configuration) *****
***** delaying boot 1000ms (per build configuration) *****
```

**Environment (please complete the following information):**
 - OS: fedora
 - Toolchain: zephyr
 - Commit SHA or Version used: 6368ed4edf5c5ffbccf9f904efe836ef5ca5d84a

**Additional context**
Add any other context about the problem here.

aurel32, 2019-03-13:
The loop on the SAM E70 board is due to the watchdog. The interrupts are not served anymore, so the watchdog fires.

Note that the test passes when not using a delayed boot.

andrewboie, 2019-03-15:
@aurel32 the delay takes place in init's bg_thread_main(), before it calls application main(). interrupts are not locked when this is happening. why would busy-looping here cause the watchdog to fail? what kicks the dog? is there some high-priority watchdog thread that needs to be started?

aurel32, 2019-03-15:
@andrewboie The watchdog is not the cause of the issue, it triggers after ~15 seconds. I was just trying to explain the difference between the two boards. Adding this 1 second delay causes other parts of the code to enter in an endless loop and prevents the watchdog ISR to be serviced. Disabling the watchdog on the SAM E70 boards doesn't fix the issue it just makes both boards to hang with a similar behavior.

andrewboie, 2019-03-15:
> prevents the watchdog ISR to be serviced.

What is supposed to service it?

What I'm seeing is that k_busy_wait() has a problem on sam e70 where instead of waiting 1000ms, it seems to loop forever.

Eventually the watchdog resets the system. But this always happens, if I don't enable the boot delay I just see the test succeed, then reset, and run again, succeed, reset, ad nauseum. Interrupts aren't locked. My feeling is that with watchdog enabled, unless there is a thread created to service it, the test will reset after 15 seconds no matter what is happening. Since this is a test with no threads we should disable the watchdog for this test.

Having said all that, watchdog aside, I think the real bug here is that k_busy_wait() is supposed to return after 1000ms and it just sits there forever instead.

andrewboie, 2019-03-15:
Yeah k_busy_wait() is broken, with this debug printout:

```
diff --git a/kernel/thread.c b/kernel/thread.c
index 5f8cab4883..90472b6dc5 100644
--- a/kernel/thread.c
+++ b/kernel/thread.c
@@ -112,6 +112,7 @@ void z_impl_k_busy_wait(u32_t usec_to_wait)
 
        for (;;) {
                u32_t current_cycles = k_cycle_get_32();
+               printk("%u %u\n", current_cycles - start_cycles, cycles_to_wait);
 
                /* this handles the rollover on an unsigned 32-bit value */
                if ((current_cycles - start_cycles) >= cycles_to_wait) {
```

I get output:

```
3346014 300000000
3838500 300000000
4330978 300000000
1823465 300000000
2315935 300000000
2808425 300000000
3300897 300000000
3793383 300000000
4285861 300000000
1778338 300000000
2270812 300000000
2763298 300000000
3255776 300000000
3748250 300000000
4240742 300000000
1733211 300000000
2225695 300000000
2718183 300000000
3210653 300000000
3703139 300000000
4195609 300000000
1688094 300000000
2180580 300000000
2673050 300000000
3165540 300000000
3658014 300000000
4150500 300000000
1642969 300000000
2135459 300000000
2627933 300000000
3120407 300000000
3612887 300000000
4105377 300000000
1597846 300000000
2090336 300000000
2582820 300000000
3075294 300000000
3567768 300000000
4060246 300000000
1552727 300000000
2045213 300000000
2537693 300000000
3030171 300000000
3522649 300000000
4015139 300000000
1507604 300000000
2000086 300000000
2492576 300000000
2985054 300000000
3477528 300000000
3970018 300000000
4462484 300000000
1954977 300000000
2447451 300000000
2939929 300000000
3432403 300000000
3924889 300000000
4417363 300000000
1909852 300000000
2402324 300000000
2894810 300000000
3387284 300000000
3879762 300000000
4372256 300000000
1864733 300000000
2357207 300000000
2849693 300000000
3342171 300000000
3834645 300000000
```

k_cycle_get_32() seems to wrap around at some upper bound far less than what is needed for the loop to ever break.

andrewboie, 2019-03-15:
I noticed that when CONFIG_MULTITHREADING=n, I get no timer interrupts. This doesn't seem right. No wonder things aren't working.

What's happening is the main thread is running with interrupts disabled. @andyross fixed this: https://github.com/zephyrproject-rtos/zephyr/pull/9668

And then later it was reverted along with a bunch of other stuff by @nashif :
https://github.com/zephyrproject-rtos/zephyr/issues/9808

The tree is now in a bad state. We still support disabling multithreading, but a lot of work towards fixing it was taken out, and the feature is essentially broken. We need to restore them.

andrewboie, 2019-03-15:
After much discussion on slack, it seems clear that several people are strongly opposed to disabling this for 1.14, even temporarily.

@cinlyooi-intel  @inakypg  @nashif Is CONFIG_BOOT_BANNER a hard requirement for hardware testing?

inakypg, 2019-03-22:
> After much discussion on slack, it seems clear that several people are strongly opposed to disabling this for 1.14, even temporarily.
> 
> @cinlyooi-intel @inakypg @nashif Is CONFIG_BOOT_BANNER a hard requirement for hardware testing?

It is highly desirable, because it allows us to detect if we saw all the output; the test runner will verify if it got all of the expected messages. If the first one is missing, something might be stinking (early missing console because who knows).

This, for example, catches a lot in ESP32, we loose a lot of early console and it is not clear yet why.

[what the hell -- sorry, by mistake I edited your comment instead of replying to it -- fixed, blame to -ENOCOFFEE ]

PR #14917 (2019-03-27): kernel: do not use  k_busy_wait when on single thread
k_busy_wait() does not work when multithreading is disabled, so do not try
to wait during boot.

Fixes #14454
commit 42f4538e (2019-03-27): kernel: do not use k_busy_wait when on single thread

k_busy_wait() does not work when multithreading is disabled, so do not
try to wait during boot.

Fixes #14454
closed 2019-03-27 by nashif