zephyrproject-rtos/zephyr #14459: usb: samples: mass: doesn't build with FLASH overlay

masz-nordic, 2019-03-13:
**Describe the bug**
USB MSC sample doesn't build with `overlay-flash-disk.conf` for nrf52840_pca10056.

**To Reproduce**
- `cd samples/subsys/usb/mass`
-  `west build --board nrf52840_pca10056 -- -DOVERLAY_CONFIG=overlay-flash-disk.conf`

**Expected behavior**
Sample builds.

**Impact**
Low priority for now. Should be fixed before MSC is considered ready.

**Screenshots or console output**
```
zephyr/misc/generated/configs.c: In function ‘_ConfigAbsSyms’:
zephyr/misc/generated/configs.c:275:154: error: invalid suffix "x" on integer constant
 GEN_ABSOLUTE_SYM(CONFIG_DISK_FLASH_START, 0x);
                                                                                                                                                          ^ 
zephyr/misc/generated/configs.c:276:172: error: expected expression before ‘)’ token
 GEN_ABSOLUTE_SYM(CONFIG_DISK_FLASH_MAX_RW_SIZE, );
                                                                                                                                                                            ^
zephyr/misc/generated/configs.c:277:184: error: invalid suffix "x" on integer constant
 GEN_ABSOLUTE_SYM(CONFIG_DISK_FLASH_ERASE_ALIGNMENT, 0x);
                                                                                                                                                                                        ^ 
zephyr/misc/generated/configs.c:278:169: error: invalid suffix "x" on integer constant
 GEN_ABSOLUTE_SYM(CONFIG_DISK_ERASE_BLOCK_SIZE, 0x);
                                                                                                                                                                         ^ 
zephyr/misc/generated/configs.c:279:154: error: invalid suffix "x" on integer constant
 GEN_ABSOLUTE_SYM(CONFIG_DISK_VOLUME_SIZE, 0x);
                                                                                                                                                          ^ 
[25/131] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/printk.c.obj
ninja: build stopped: subcommand failed.
```

Seems like following Kconfig symbols are defined empty:
`DISK_FLASH_DEV_NAME`, `DISK_FLASH_START`, `DISK_FLASH_MAX_RW_SIZE`, `DISK_FLASH_ERASE_ALIGNMENT`, `DISK_ERASE_BLOCK_SIZE`, `DISK_VOLUME_SIZE`

**Environment (please complete the following information):**
 - OS: Linux Mint 18.3 x64
 - Toolchain: Zephyr SDK 0.10.0
 - Zephyr 6368ed4edf5c5ffbccf9f904efe836ef5ca5d84a

carlescufi, 2019-03-15:
Fixed via #14299

masz-nordic, 2019-03-15:
This is different from #14299 - that fix was adding RAM to default configuration. Here, the FLASH configuration is not working at all.

carlescufi, 2019-03-18:
@nvlsianpu this looks like an easy fix, can you take a look?

nvlsianpu, 2019-03-18:
@masz-nordic As you can check in autoconf.h:
```
#define CONFIG_DISK_FLASH_START 0x
#define CONFIG_DISK_FLASH_MAX_RW_SIZE 
#define CONFIG_DISK_FLASH_ERASE_ALIGNMENT 0x
#define CONFIG_DISK_ERASE_BLOCK_SIZE 0x
#define CONFIG_DISK_VOLUME_SIZE 0x
```

Problem is that there is no default configuration for disk susbsystem.
I believe that you can use DT properties of storage-partition for default volume boundaries. erase-block-size is also avielable there.  I recommend to use specific *.conf for the board you used.

Any patch with default kconfig welcome.

galak, 2019-03-26:
We need to either add proper support for the  prj_nrf52840_pca10056.conf that configures all the params we need or remove this file so its clear that nrf52840_pca10056 isn't a supported target of the sample.

galak, 2019-03-26:
> We need to either add proper support for the prj_nrf52840_pca10056.conf that configures all the params we need or remove this file so its clear that nrf52840_pca10056 isn't a supported target of the sample.

To be clear, if we fix prj_nrf52840_pca10056.conf we also need to add it to sample.yaml

PR #14980 (2019-05-06): usb: samples: mass: fix FLASH overlay for nrf52840_pca10056
Adds required Kconfig options for flash driver.
Adds FatFS filesystem for flash disk.
Fixes #14459.
commit 323756a2 (2019-05-06): usb: samples: mass: add configuration and fs for FLASH

Adds required Kconfig options for flash driver.
Adds FatFS filesystem for flash disk.
Fixes #14459.

Note: As USB MSC does not support multiple disks (see #14937),
only one (flash by default, modify CONFIG_MASS_STORAGE_DISK_NAME
to "RAM" for RAM disk) will appear to host.
commit 7ebed1b3 (2019-05-06): usb: samples: mass: add configuration and fs for FLASH

Adds required Kconfig options for flash driver.
Adds FatFS filesystem for flash disk.
Fixes #14459.

Note: As USB MSC does not support multiple disks (see #14937),
only one (flash by default, modify CONFIG_MASS_STORAGE_DISK_NAME
to "RAM" for RAM disk) will appear to host.
commit be1ec782 (2019-05-06): usb: samples: mass: add configuration and fs for FLASH

Adds required Kconfig options for flash driver.
Adds FatFS filesystem for flash disk.
Fixes #14459.

Note: As USB MSC does not support multiple disks (see #14937),
only one (flash by default, modify CONFIG_MASS_STORAGE_DISK_NAME
to "RAM" for RAM disk) will appear to host.
closed 2019-05-06 by carlescufi