zephyrproject-rtos/zephyr #14462: tests/kernel/threads/no-multithreading/testcase.yam: Not Booting

chiachuanwu, 2019-03-13:
**Describe the bug**
Failed to boot up on altera_max10 board

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=altera_max10
3. make
4. make run

**Expected behavior**
Should able to boot zephyr and show log on console output.

**Screenshots or console output**
no console output.

OS: fedora29
Toolchain: zephyr
Commit SHA or Version used: 6368ed4

rljordan, 2019-03-15:
@ramakrishnapallala  Do you have the hardware to test this?

rljordan, 2019-03-15:
@ramakrishnapallala  You can use the internal pool for this.

ramakrishnapallala, 2019-03-19:
I see that test works:

***** Booting Zephyr OS v1.7.99-16394-g3a8767c1f269 *****
It works
===================================================================
PROJECT EXECUTION SUCCESSFUL
===================================================================
PROJECT EXECUTION SUCCESSFUL

Here are the steps I followed:

mkdir build; cd build
cmake -DBOARD=altera_max10 ..
make flash

Check the logs on the serial console.

ManojSubbarao, 2019-03-21:
@chiachuanwu can you close if this is not an issue anymore?

chiachuanwu, 2019-03-21:
hi @ManojSubbarao  , this issued might be similar with #14454.

ManojSubbarao, 2019-03-21:
Thanks @chiachuanwu . Closing this issue since it's duplicate of #14454

closed 2019-03-21 by ManojSubbarao