zephyrproject-rtos/zephyr #14471: Mpu fault during application startup

nordic-krch, 2019-03-13:
**Describe the bug**
Application can hit mpu fault before starting main thread. This is caused by stack misuse. In function `z_arch_switch_to_main_thread` thread entry is provided as forth argument (`_main`). Then within the function stack pointer is changed:
https://github.com/zephyrproject-rtos/zephyr/blob/269935dca9e139d365631979fbecf6d080cba758/arch/arm/include/kernel_arch_func.h#L106

After stack is changed, call to thread entry occurs which is using `_main` as the argument. If variable is stored on stack, it will be fetched from stack but at that point this is already different stack and value is corrupted:
https://github.com/zephyrproject-rtos/zephyr/blob/269935dca9e139d365631979fbecf6d080cba758/arch/arm/include/kernel_arch_func.h#L115
So `_main` variable will be corrupted.

**To Reproduce**
Modify hello_world example by using following prj.conf:
```
CONFIG_MPU_STACK_GUARD=y
CONFIG_NO_OPTIMIZATIONS=y
CONFIG_LOG=y
```

Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -GNinja -DBOARD=nrf52840_pca10056 ..
3. ninja flash
4. See error
```
Fatal fault in essential thread! Spinning...
***** MPU FAULT *****
  Instruction Access Violation
***** Hardware exception *****
Current thread ID = 0x200005d8
Faulting instruction address = 0x20001b48
Fatal fault in essential thread! Spinning...
```

**Expected behavior**
Sample should start.

**Impact**
Suddenly system crashes during initialization.

ioannisg, 2019-03-13:
Going to look into this.

ioannisg, 2019-03-13:
Yes, this looks like a bug. It was, probably, inserted when we refactored this file to look like a C file, instead of assembly code.

PR #14505 (2019-03-14): Arch arm fix sp corrupt in jump to main
Fixes #14471 

Reverts, partially, the changes introduced in (bbe1a19), concerning moving the implementation into C functions. Apparently, we did not catch issues when compiling without any optimization flags, where local variables are stored, anyways, in the stack (and the stack pointer changes inside the function). We revert using assembly, now. An ISB is also added, before branching to main thread. Finally, the patch is also adding some inline documentation to ease reading.

A second patch is also provided, that moves the MPU re-programming a bit high [... cut]
commit 78618fbc (2019-03-14): arch: arm: revert back to use assembly to jump to main thread

This commit partially reverts the changes introduced in
(bbe1a19786b06b3669be8fe9f90d160335dce3ea), where the PSP
modification, the interrupt enabling, and the branch to main
thread were refactored in C functions. This introduced an issue
with stack usage when compiling without any optimization.
Reverting to use assembly functions to jump to main thread
ensures, now, that the pointer to main() is preserved intact
while changing the PSP and passed correctly to z_thread_entry().

Fixes #14471.
closed 2019-03-14 by galak