zephyrproject-rtos/zephyr #14476: quark_d2000_crb/samples/sensor/bmg160 runs out of ROM (CI failure)

ulfalizer, 2019-03-13:
Seen in https://github.com/zephyrproject-rtos/zephyr/pull/14452. Note

```
[103/110] Linking C executable zephyr/zephyr_prebuilt.elf
FAILED: : && ccache /opt/sdk/zephyr-sdk-0.10.0/i586-zephyr-elfiamcu/bin/i586-zephyr-elfiamcu-gcc    zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj  -o zephyr/zephyr_prebuilt.elf  -T zephyr/linker.cmd -Wl,-Map=/home/buildslave/src/github.com/zephyrproject-rtos/zephyr/out-3nd-pass/quark_d2000_crb/samples/sensor/bmg160/test/zephyr/zephyr.map -u_OffsetAbsSyms -u_ConfigAbsSyms -Wl,--whole-archive app/libapp.a zephyr/libzephyr.a zephyr/arch/arch/x86/core/libarch__x86__core.a zephyr/lib/libc/minimal/liblib__libc__minimal.a zephyr/boards/x86/quark_d2000_crb/libboards__x86__quark_d2000_crb.a zephyr/ext/hal/qmsi/libext__hal__qmsi.a zephyr/drivers/gpio/libdrivers__gpio.a zephyr/drivers/i2c/libdrivers__i2c.a zephyr/drivers/sensor/bmg160/libdrivers__sensor__bmg160.a zephyr/drivers/serial/libdrivers__serial.a -Wl,--no-whole-archive zephyr/kernel/libkernel.a zephyr/CMakeFiles/offsets.dir/arch/x86/core/offsets/offsets.c.obj -L"/opt/sdk/zephyr-sdk-0.10.0/i586-zephyr-elfiamcu/bin/../lib/gcc/i586-zephyr-elfiamcu/8.3.0" -L/home/buildslave/src/github.com/zephyrproject-rtos/zephyr/out-3nd-pass/quark_d2000_crb/samples/sensor/bmg160/test/zephyr -lgcc -Wl,--print-memory-usage -nostdlib -static -no-pie -Wl,--fatal-warnings -Wl,-X -Wl,-N -Wl,--gc-sections -Wl,--build-id=none -Wl,--orphan-handling=warn && :
Memory region         Used Size  Region Size  %age Used
             ROM:       32772 B        32 KB    100.01%
             RAM:        5860 B       8160 B     71.81%
  SHARED_GDT_RAM:          0 GB         32 B      0.00%
        IDT_LIST:         464 B         2 KB     22.66/opt/sdk/zephyr-sdk-0.10.0/i586-zephyr-elfiamcu/bin/../lib/gcc/i586-zephyr-elfiamcu/8.3.0/../../../../i586-zephyr-elfiamcu/bin/ld: zephyr/zephyr_prebuilt.elf section `_k_mutex_area' will not fit in region `ROM'
/opt/sdk/zephyr-sdk-0.10.0/i586-zephyr-elfiamcu/bin/../lib/gcc/i586-zephyr-elfiamcu/8.3.0/../../../../i586-zephyr-elfiamcu/bin/ld: region `ROM' overflowed by 4 bytes
collect2: error: ld returned 1 exit status
%
ninja: build stopped: subcommand failed.
-----------------out-3nd-pass/quark_d2000_crb/samples/sensor/bmg160/test/build.log------------------
0 of 1 tests passed with 0 warnings in 13 seconds
```

ulfalizer, 2019-03-13:
CC @galak

andrewboie, 2019-03-15:
@ulfalizer is this a sporadic failure? Do you have any special build flags set?

When I built locally, it succeeds, although it's right at the knife edge:

```
[103/110] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
             ROM:       32756 B        32 KB     99.96%
             RAM:        5860 B       8160 B     71.81%
  SHARED_GDT_RAM:          0 GB         32 B      0.00%
        IDT_LIST:         464 B         2 KB     22.66%
[110/110] Linking C executable zephyr/zephyr.elf
```

ulfalizer, 2019-03-15:
No special build flags. I can reproduce it at 54ae600:

```
~/z/samples/sensor/bmg160/b $ ninja
...
[103/110] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
             ROM:       33212 B        32 KB    101.35%
             RAM:        5860 B       8160 B     71.81%
  SHARED_GDT_RAM:          0 GB         32 B      0.00%
        IDT_LIST:         464 B         2 KB     22.66%
/home/ulf/zephyr/sdk/sysroots/x86_64-pokysdk-linux/usr/libexec/i586-zephyr-elfiamcu/gcc/i586-zephyr-elfiamcu/6.2.0/real-ld: zephyr/zephyr_prebuilt.elf section `rodata' will not fit in region `ROM'
/home/ulf/zephyr/sdk/sysroots/x86_64-pokysdk-linux/usr/libexec/i586-zephyr-elfiamcu/gcc/i586-zephyr-elfiamcu/6.2.0/real-ld: region `ROM' overflowed by 444 bytes
```

PR #14583 (2019-03-18): samples: bmg169: don't enable CONFIG_LOG
It's not necessary for this sample to output to the user
(it uses printf) and saves over 5K of ROM size, eliminating
some build failures on Quark D2000.

Fixes: #14476
commit 435c1a3f (2019-03-18): samples: bmg169: don't enable CONFIG_LOG

It's not necessary for this sample to output to the user
(it uses printf) and saves over 5K of ROM size, eliminating
some build failures on Quark D2000.

Fixes: #14476
closed 2019-03-18 by andrewboie