zephyrproject-rtos/zephyr #14483: NXP cortexm7 platform set flexspi_nor in device type by MPU need avoid unaligned data access.

hakehuang, 2019-03-13:
**Describe the situation**
according to the DDI0403E_d_armv7m_arm.pdf A3.5.7 paragraph 3
```
An instruction that generates an unaligned memory access to Device or Strongly-ordered memory is
UNPREDICTABLE.
```
so if set the flexspi_nor memory area to device type in MPU, then we shall be careful not to generate unaligned memory access to those area.

hakehuang, 2019-03-13:
@MaureenHelm

aurel32, 2019-03-13:
I think this also applies to any Cortex-M4 or Cortex-M7 supported by Zephyr. I don't think it's possible to do configure the flash region as device or strongly ordered in practice.

dleach02, 2024-06-25:
@hakehuang this is an old item and I'm trying to understand what is being requested for us to do?

hakehuang, 2024-06-25:
> @hakehuang this is an old item and I'm trying to understand what is being requested for us to do?
> I don't think it's possible to do configure the flash region as device or strongly ordered in practice.

@dleach02  the thing is in two ways:
1. in normal application, we need ensure the flash region is not configure as device or strongly ordered.
2. in boot loader, we need ensure the flash access is aligned.

dleach02, 2024-06-25:
@aurel32 comment seems to suggest that this would be a generic thing across all platforms.