zephyrproject-rtos/zephyr #14493: implement Zephyr futexes

andrewboie, 2019-03-13:
**Is your enhancement proposal related to a problem? Please describe.**
It is a very common programming idiom to embed a synchronization primitive inside a data structure, to protect access to that data structure. For example:

```
struct client_context {
    int foo;
    int bar;
    int baz;
    k_mutex_t mutex;
}
```

This unfortunately does not work well with user mode, as the memory for the k_mutex_t is a kernel object and needs to live in the kernel's memory area without any direct access by user mode. Since user mode would need direct access to the foo, bar, and baz members, we would have to create the mutex separately and store a pointer to it:

```
struct client_context {
    int foo;
    int bar;
    int baz;
    k_mutex_t *mutex;
}
...
K_MUTEX_DEFINE(some_mutex);
...
```

Making things more cumbersome is that any thread using this mutex would have to be granted permission to perform operations on it otherwise access will fail. For synchronization primitives like this, buried deep within some library or middleware code, this rapidly becomes very painful to deal with.

**Describe the solution you'd like**
Implementing futexes makes this much easier.

- The user memory representing the futex is just an atomic_t and doesn't need to be treated as a kernel object; it can exist as a member of the containing struct in application memory.
- No separate permission management is needed; the memory domain access policy for whatever memory the futex's atomic_t lives in is quite sufficient.
- As a bonus, no syscalls at all for non-contended futexes.

@lpereira spent some time to put together a proof-of-concept which could be fleshed out further:
https://github.com/lpereira/zephyr/commit/11cee8db961571585dfdad4cc7af865f89168d5f

His implementation defines an array of futex 'channels', which maps futex memory addresses to a wait-queue of threads pending on that futex. There is an O(n) lookup to find the wait-queue for any given futex address that takes place with IRQs locked.

We need to know what the max number of channels should be for any given application; I think it is simply a function of the number of threads in the system.

We could probably make this into an array-based hash table for quicker lookups.

Futexes are a primitive API type that friendlier APIs are built on top of. Consider if we want to re-implement k_sem/k_mutex in terms of futexes, or introduce some new APIs (sys_sem? sys_mutex?) that use these.

lpereira, 2019-03-13:
For the userspace portion of futexes, the pthread implementation in the [bionic libc](https://android.googlesource.com/platform/bionic/+/10ce969/libc/bionic/pthread.c) can serve as an inspiration.

andrewboie, 2019-03-16:
For syscall side I am studying the Linux man page (but avoiding any glimpse of the code):
http://man7.org/linux/man-pages/man2/futex.2.html

Looking at what Bionic does, they just use a subset of futex operations. We need even less since everything is in the same address space. I think we want to implement the following operations:

FUTEX_WAIT
FUTEX_WAKE
FUTEX_LOCK_PI
FUTEX_UNLOCK_PI

lpereira, 2019-03-16:
Yes, FUTEX_WAIT and FUTEX_WAKE were implemented in my sketch because I had the same realization after reading Bionic. I had forgotten about the priority inheritance stuff, though. Good catch.

A good thing to keep in mind, however, is that futexes aren't the best solution for supervisor-mode code (or supervisor/user mode hybrid code). Instead of having specialized functions that takes a pointer to a structure containing everything they need to operate, a channel lookup would have to be performed to find the wait channel. No matter how efficient that lookup is, it's never going to be as efficient as dereferencing a pointer.

Maybe it would be a good idea to use this opportunity to rethink about the synchronization primitives in Zephyr? We've had our share of issues with early boot uses of semaphores, CONFIG_MULTITHREADING, and stuff like that.

andrewboie, 2019-03-17:
> A good thing to keep in mind, however, is that futexes aren't the best solution for supervisor-mode code (or supervisor/user mode hybrid code). 

I was thinking a decent starting point is to implement sys_sem_* and sys_mutex_* with the same semantics as their kernel counterparts (including some semaphore APIs being ISR safe), but futex-backed. On systems which don't turn on userspace these can just be linker aliases to the kernel versions.

I'm also considering using our gperf scanning to make a hashtable of all k_futex_t addresses found and map to wait queues.

> Maybe it would be a good idea to use this opportunity to rethink about the synchronization primitives in Zephyr? We've had our share of issues with early boot uses of semaphores, CONFIG_MULTITHREADING, and stuff like that.

I don;t immediately see how this could help that situation unfortuantely. What idea did you have?

lpereira, 2019-03-28:
One of the things that futexes could be used for is to move the network stack to user mode.

What I imagine that could be made here is to keep L2 drivers in kernel mode --  but their device type could expose ring buffers for transmission and reception that L3 upwards (running in user mode) could use and synchronize with futexes.  

If you want inspiration for this kind of thing, you may want to look at how AIO exposes its ring buffer in Linux (there are some articles written by ScyllaDB authors that are a great starting point).

andrewboie, 2019-05-07:
@wentongwu started work on this, adding him to the assignees list.
No specific due date yet.

commit 5611e923 (2019-06-24): kernel: add futex support

A k_futex is a lightweight mutual exclusion primitive designed
to minimize kernel involvement. Uncontended operation relies
only on atomic access to shared memory. k_futex structure lives
in application memory. And when using futexes, the majority of
the synchronization operations are performed in user mode. A
user-mode thread employs the futex wait system call only when
it is likely that the program has to block for a longer time
until the condition becomes true. When the condition comes true,
futex wake operation will be used to wake up one or more threads
waiting  [... cut]
commit a30162be (2019-06-24): tests: add futex test cases

add futex test cases.

Fixes: #14493.
closed 2019-06-24 by andrewboie