zephyrproject-rtos/zephyr #14499: sanitycheck --coverage on qemu_x86: Too many handler_crash

cinlyooi-intel, 2019-03-14:
**Describe the bug**
We have too many handler_crash failure when running code coverage using qemu_x86

**Impact**
Currently blocking code coverage improvement work.

**To Reproduce and console output**
```
$ ./scripts/sanitycheck -v --coverage --coverage-platform qemu_x86 -p qemu_x86  --gcov-tool /opt/zephyr-sdk-0.9.5/sysroots/x86_64-pokysdk-linux/usr/bin/i586-zephyr-elf/i586-zephyr-elf-gcov -T tests/kernel/ -T tests/subsys/
JOBS: 8
Cleaning output directory /path/to/zephyr/sanity-out
Reading test case configuration files under /path/to/zephyr/tests/kernel...
Reading test case configuration files under /path/to/zephyr/tests/subsys...
Reading platform configuration files under /path/to/zephyr.git/boards...
Reading platform configuration files under /path/to/zephyr.git/scripts/sanity_chk/boards...
Building testcase defconfigs...
86 tests selected, 17302 tests discarded due to filters
 1/85 qemu_x86                  common/kernel.common.misra                         FAILED: handler_crash
	see: sanity-out/qemu_x86/common/kernel.common.misra/build.log
 2/85 qemu_x86                  common/kernel.common                               FAILED: handler_crash
	see: sanity-out/qemu_x86/common/kernel.common/build.log
 3/85 qemu_x86                  fatal/kernel.common.stack_protection               FAILED: handler_crash
	see: sanity-out/qemu_x86/fatal/kernel.common.stack_protection/build.log
 4/85 qemu_x86                  context/kernel.common                              FAILED: handler_crash
	see: sanity-out/qemu_x86/context/kernel.common/run.log
 5/85 qemu_x86                  fatal/kernel.common.stack_sentinel                 FAILED: handler_crash
	see: sanity-out/qemu_x86/fatal/kernel.common.stack_sentinel/build.log
 6/85 qemu_x86                  device/kernel.device.pm                            PASSED
 7/85 qemu_x86                  fifo/fifo_api/kernel.fifo                          FAILED: handler_crash
	see: sanity-out/qemu_x86/fifo/fifo_api/kernel.fifo/run.log
 8/85 qemu_x86                  device/kernel.device                               PASSED
 9/85 qemu_x86                  boot_page_table/kernel.memory_protection           PASSED
10/85 qemu_x86                  can/frame/can.frame                                PASSED
11/85 qemu_x86                  fifo/fifo_api/kernel.fifo.poll                     FAILED: handler_crash
	see: sanity-out/qemu_x86/fifo/fifo_api/kernel.fifo.poll/run.log
12/85 qemu_x86                  fifo/fifo_timeout/kernel.fifo.timeout              FAILED: handler_crash
	see: sanity-out/qemu_x86/fifo/fifo_timeout/kernel.fifo.timeout/build.log
13/85 qemu_x86                  fifo/fifo_timeout/kernel.fifo.timeout.poll         FAILED: handler_crash
	see: sanity-out/qemu_x86/fifo/fifo_timeout/kernel.fifo.timeout.poll/build.log
14/85 qemu_x86                  early_sleep/kernel.common                          FAILED: handler_crash
	see: sanity-out/qemu_x86/early_sleep/kernel.common/run.log
15/85 qemu_x86                  critical/kernel.common                             PASSED
16/85 qemu_x86                  fs/multi-fs/filesystem.fs_shell                    FAILED: unexpected eof
	see: sanity-out/qemu_x86/fs/multi-fs/filesystem.fs_shell/handler.log
17/85 qemu_x86                  fifo/fifo_usage/kernel.fifo.usage                  PASSED
18/85 qemu_x86                  fs/fat_fs_dual_drive/filesystem.fat                PASSED
19/85 qemu_x86                  fifo/fifo_usage/kernel.fifo.usage.poll             PASSED
20/85 qemu_x86                  interrupt/arch.interrupt                           FAILED: handler_crash
	see: sanity-out/qemu_x86/interrupt/arch.interrupt/build.log
21/85 qemu_x86                  fs/multi-fs/filesystem.multifs                     PASSED
22/85 qemu_x86                  fs/nffs_fs_api/basic/filesystem.nffs.basic         PASSED
23/85 qemu_x86                  lifo/lifo_api/kernel.lifo                          FAILED: handler_crash
	see: sanity-out/qemu_x86/lifo/lifo_api/kernel.lifo/run.log
24/85 qemu_x86                  lifo/lifo_usage/kernel.lifo.usage                  FAILED: handler_crash
	see: sanity-out/qemu_x86/lifo/lifo_usage/kernel.lifo.usage/run.log
25/85 qemu_x86                  fs/nffs_fs_api/cache/filesystem.nffs.cache         PASSED
26/85 qemu_x86                  fs/nffs_fs_api/large/filesystem.nffs.large         PASSED
27/85 qemu_x86                  jwt/libraries.encoding                             PASSED
28/85 qemu_x86                  logging/log_core/logging.log_core                  PASSED
29/85 qemu_x86                  fs/nffs_fs_api/performance/filesystem.nffs.performance PASSED
30/85 qemu_x86                  mbox/mbox_api/kernel.mailbox                       FAILED: handler_crash
	see: sanity-out/qemu_x86/mbox/mbox_api/kernel.mailbox/run.log
31/85 qemu_x86                  logging/log_list/logging.log_list                  PASSED
32/85 qemu_x86                  logging/log_msg/logging.log_msg                    PASSED
33/85 qemu_x86                  logging/log_output/logging.log_output              PASSED
34/85 qemu_x86                  mem_pool/mem_pool/kernel.memory_pool               FAILED: handler_crash
	see: sanity-out/qemu_x86/mem_pool/mem_pool/kernel.memory_pool/run.log
35/85 qemu_x86                  mem_pool/mem_pool_api/kernel.memory_pool           FAILED: handler_crash
	see: sanity-out/qemu_x86/mem_pool/mem_pool_api/kernel.memory_pool/run.log
36/85 qemu_x86                  mem_pool/mem_pool_concept/kernel.memory_pool       FAILED: handler_crash
[... 204 lines cut ...]
Generating gcda files
Working on /home/cinlyooi/tmp/zephyr/sanity-out/qemu_x86/mem_protect/protection/kernel.memory_protection.protection/handler.log
Generating gcda files
Working on /home/cinlyooi/tmp/zephyr/sanity-out/qemu_x86/can/frame/can.frame/handler.log
Generating gcda files
Working on /home/cinlyooi/tmp/zephyr/sanity-out/qemu_x86/static_idt/arch.interrupt/handler.log
Generating gcda files
Working on /home/cinlyooi/tmp/zephyr/sanity-out/qemu_x86/context/kernel.common/handler.log
Generating gcda files
Working on /home/cinlyooi/tmp/zephyr/sanity-out/qemu_x86/boot_page_table/kernel.memory_protection/handler.log
Generating gcda files
Working on /home/cinlyooi/tmp/zephyr/sanity-out/qemu_x86/profiling/profiling_api/kernel.profiling/handler.log
Generating gcda files
Working on /home/cinlyooi/tmp/zephyr/sanity-out/qemu_x86/mutex/mutex_api/kernel.mutex/handler.log
Generating gcda files
Working on /home/cinlyooi/tmp/zephyr/sanity-out/qemu_x86/mutex/mutex/kernel.mutex/handler.log
Generating gcda files
HTML report generated: /home/cinlyooi/tmp/zephyr/sanity-out/coverage/index.html
29 of 85 tests passed with 0 warnings in 327 seconds
```

**Environment (please complete the following information):**
 - OS: fedora29
 - Toolchain (e.g Zephyr SDK, ...): zephyr
 - Commit SHA or Version used: a907e694e95d1a31208b4863966d93adcfa1a2a5

cinlyooi-intel, 2019-03-14:
On @inakypg  suggestion I started to examine the handler.log files

The few handler.log files that I check all  has "Double Fault" -- "Stack Check Fail!" like the example below
```
$/opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/qemu-system-i386 -m 8 -cpu qemu32,+nx,+pae -device isa-debug-exit,iobase=0xf4,iosize=0x04 -no-reboot -nographic -no-acpi -net none -pidfile qemu.pid -kernel /path/to/sanity-out/qemu_x86/tests/kernel/mem_pool/mem_pool/kernel.memory_pool/zephyr/zephyr.elf

SeaBIOS (version rel-1.11.2-0-gf9626ccb91-prebuilt.qemu-project.org)
Booting from ROM..***** Booting Zephyr OS v1.14.0-rc1-1138-g4278bf567348 *****
Running test suite mempool
===================================================================
starting test - test_pool_block_get
PASS - test_pool_block_get
===================================================================
starting test - test_pool_block_get_timeout
***** Double Fault *****
***** Stack Check Fail! *****
Current thread ID = 0x0040d560
eax: 0x0040c000, ebx: 0x80000000, ecx: 0x0040c007, edx: 0x00000000
esi: 0x0040c007, edi: 0x0040d5c0, ebp: 0x0042908c, esp: 0x00429000
eflags: 0x00000012 cs: 0x0008
call trace:
eip: 0x0000d1cb
     0x0000d7a9 (0x440000)
     0x0000d885 (0x40d3e0)
     0x0000d95d (0x40d3e0)
     0x0000c7a9 (0x40d380)
     0x000014bf (0x40d380)
     0x00011d50 (0x0)
     0x00011dbe (0x0)
     0x00006870 (0x0)
Fatal fault in essential thread! Spinning...
Terminate emulator due to fatal kernel error
```

wentongwu, 2019-03-19:
run the case(e.g. kernel.workqueue) manually, it works well.

wentongwu, 2019-03-27:
.noinit."/home/wentongw/zephyr/zephyrproject/zephyr/kernel/init.c".2
                0x0000000000424000     0x3000 zephyr/kernel/libkernel.a(init.c.obj)
                0x0000000000424000                _idle_stack

***** Stack Check Fail! *****
Current thread ID = 0x0040c3a0
eax: 0x0040b000, ebx: 0x80000000, ecx: 0x0040b067, edx: 0x00000000
esi: 0x0040b067, edi: 0x0040c440, ebp: 0x0042608c, esp: 0x00426000
eflags: 0x00000012 cs: 0x0008

@andrewboie  for above log, if increasing the idle thread stack, the case works well. it seems x86 also has stack overflow issue. But I'm not familiar with x86 arch, why either ebp or esp equal to _idle_stack? and what's x86's mpu(hope I named it correctly) mechanism? Thanks a lot!

Anyway I'm increasing all kinds of stack size to fix this issue.

cinlyooi-intel, 2019-03-29:
@wentongwu : Do you get this error `linker_pass_final.cmd:65 cannot move location counter backwards (from 000000000002e34e to 000000000002dfaa)` at all?

wentongwu, 2019-03-29:
> @wentongwu : Do you get this error `linker_pass_final.cmd:65 cannot move location counter backwards (from 000000000002e34e to 000000000002dfaa)` at all?

I didn't get this. which case are you running? Did you apply my PR?

cinlyooi-intel, 2019-03-29:
@wentongwu : 
Two scenarios 

(1)  Pure PR15002 no longer have handler_crash, but does not generate gcov data
```
$ git fetch origin pull/15002/head:pull/15002
$ git checkout pull/15002
$ ./scripts/sanitycheck --coverage --coverage-platform qemu_x86 -p qemu_x86 --gcov-tool /opt/zephyr-sdk-0.10.0/i586-zephyr-elf/bin/i586-zephyr-elf-gcov -T tests/kernel/common  | tee output.log
JOBS: 8
Cleaning output directory /home/cinlyooi/tests/zephyr.git/sanity-out
Building testcase defconfigs...
2 tests selected, 322 tests discarded due to filters
1/2 qemu_x86                  tests/kernel/common/kernel.common.misra            PASSED
2/2 qemu_x86                  tests/kernel/common/kernel.common                  PASSED
Generating coverage files...
geninfo: WARNING: no .gcda files found in /home/cinlyooi/tests/zephyr.git/sanity-out - skipping!
lcov: ERROR: no valid records found in tracefile /home/cinlyooi/tests/zephyr.git/sanity-out/coverage.info
lcov: ERROR: no valid records found in tracefile /home/cinlyooi/tests/zephyr.git/sanity-out/coverage.info
lcov: ERROR: no valid records found in tracefile /home/cinlyooi/tests/zephyr.git/sanity-out/coverage.info
lcov: ERROR: no valid records found in tracefile /home/cinlyooi/tests/zephyr.git/sanity-out/coverage.info
genhtml: ERROR: no valid records found in tracefile /home/cinlyooi/tests/zephyr.git/sanity-out/coverage.info
2 of 2 tests passed with 0 warnings in 19 seconds
```
Examining the output shows aht the GCDA data is not created:
```
$ tail sanity-out/qemu_x86/tests/kernel/pending/kernel.common.timing/handler.log 
Testing lifos delivered data correctly ...
Testing preemptible thread waiting on timer ...
PASS - test_pending
===================================================================
Test suite pend succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL

GCOV_COVERAGE_DUMP_START
GCOV_COVERAGE_DUMP_END
```

(2) If I merge PR15002 into master branch
```
$ git checkout master #9e81cbea7df8efad52cc23e0bbdbdde5db997e56
$ git merge pull/15002
$ ./scripts/sanitycheck --coverage --coverage-platform qemu_x86 -p qemu_x86 --gcov-tool /opt/zephyr-sdk-0.10.0/i586-zephyr-elf/bin/i586-zephyr-elf-gcov -T tests/kernel/common  | tee output.log
JOBS: 8
Cleaning output directory /home/cinlyooi/tests/zephyr.git/sanity-out
Building testcase defconfigs...
2 tests selected, 322 tests discarded due to filters
1/2 qemu_x86                  tests/kernel/common/kernel.common                  FAILED: handler_crash
	see: sanity-out/qemu_x86/tests/kernel/common/kernel.common/build.log
2/2 qemu_x86                  tests/kernel/common/kernel.common.misra            FAILED: handler_crash
	see: sanity-out/qemu_x86/tests/kernel/common/kernel.common.misra/build.log
Generating coverage files...
geninfo: WARNING: no .gcda files found in /home/cinlyooi/tests/zephyr.git/sanity-out - skipping!
lcov: ERROR: no valid records found in tracefile /home/cinlyooi/tests/zephyr.git/sanity-out/coverage.info
lcov: ERROR: no valid records found in tracefile /home/cinlyooi/tests/zephyr.git/sanity-out/coverage.info
lcov: ERROR: no valid records found in tracefile /home/cinlyooi/tests/zephyr.git/sanity-out/coverage.info
lcov: ERROR: no valid records found in tracefile /home/cinlyooi/tests/zephyr.git/sanity-out/coverage.info
genhtml: ERROR: no valid records found in tracefile /home/cinlyooi/tests/zephyr.git/sanity-out/coverage.info
0 of 2 tests passed with 0 warnings in 8 seconds
```
Handler crash again. if we look at `build.log` we have  the error shown in [previous comment]([https://github.com/zephyrproject-rtos/zephyr/issues/14499#issuecomment-477852185])
```
$ cat sanity-out/qemu_x86/tests/kernel/common/kernel.common/build.log
# <snip>
[ 98%] Building C object zephyr/CMakeFiles/kernel_elf.dir/misc/empty_file.c.obj
[100%] Linking C executable zephyr.elf
/opt/zephyr-sdk-0.10.0/i586-zephyr-elf/bin/../lib/gcc/i586-zephyr-elf/8.3.0/../../../../i586-zephyr-elf/bin/ld:linker_pass_final.cm
d:65 cannot move location counter backwards (from 0000000000047b1c to 0000000000047774)
collect2: error: ld returned 1 exit status
make[3]: *** [zephyr/CMakeFiles/kernel_elf.dir/build.make:102: zephyr/zephyr.elf] Error 1

#<snip>
```

wentongwu, 2019-03-29:
@cinlyooi-intel  without my PR, master branch also can't run sanitycheck. It's not my PR cause master branch broken, please open another github issue to track the issue. Thanks

cinlyooi-intel, 2019-03-29:
@wentongwu: Sent (2) to https://github.com/zephyrproject-rtos/zephyr/issues/15009. Not going to deal with issue (1) for now as it would had been blocked by issue (2) if I filed a GitHub ticket for it. It is also unclear whether issue (1) will persist once issue (2) is resolved.

wentongwu, 2019-03-29:
@cinlyooi-intel for issue (1), I also run it on mps2_an385, but there isn't gcda data. also my PR dosen't cause the problem. It seems we see another issue after fixed your original reporting. Thanks

andrewboie, 2019-03-29:
This is the same issue as #14500 
Closing this so this doesn't count as two bugs.

closed 2019-03-29 by andrewboie