zephyrproject-rtos/zephyr #14500: sanitycheck --coverage on mps2_an385 : Too many timeouts

cinlyooi-intel, 2019-03-14:
**Describe the bug**
We have too many timeouts when running sanitycheck --coverageon mps2_an385. The handler.logs generated by sanitycheck are empty.

**Impact**
Blocking code coverage improvement work.

**To Reproduce and screenshots**
```
$./scripts/sanitycheck -v  --coverage --coverage-platform mps2_an385 -p mps2_an385 --gcov-tool /opt/zephyr-sdk-0.9.5/sysroots/x86_64-pokysdk-linux/usr/bin/i586-zephyr-elf/i586-zephyr-elf-gcov -T tests/kernel/ -T tests/subsys/
JOBS: 8
Cleaning output directory /path/to/zephyr.git/sanity-out
Reading test case configuration files under /path/to/zephyr.git/tests/kernel...
Reading test case configuration files under /path/to/zephyr.git/tests/subsys...
Reading platform configuration files under /path/to/zephyr.git/boards...
Reading platform configuration files under /path/to/zephyr.git/scripts/sanity_chk/boards...
Building testcase defconfigs...
73 tests selected, 17315 tests discarded due to filters
 1/73 mps2_an385                tests/kernel/fatal/kernel.common.stack_sentinel    PASSED
 2/73 mps2_an385                tests/kernel/common/kernel.common                  PASSED
 3/73 mps2_an385                tests/kernel/fatal/kernel.common.stack_protection  FAILED: timeout
	see: sanity-out/mps2_an385/tests/kernel/fatal/kernel.common.stack_protection/handler.log
 4/73 mps2_an385                tests/kernel/arm_irq_vector_table/arch.interrupt   FAILED: timeout
	see: sanity-out/mps2_an385/tests/kernel/arm_irq_vector_table/arch.interrupt/handler.log
 5/73 mps2_an385                tests/kernel/critical/kernel.common                FAILED: timeout
	see: sanity-out/mps2_an385/tests/kernel/critical/kernel.common/handler.log
 6/73 mps2_an385                tests/kernel/early_sleep/kernel.common             FAILED: timeout
	see: sanity-out/mps2_an385/tests/kernel/early_sleep/kernel.common/handler.log
 7/73 mps2_an385                tests/kernel/context/kernel.common                 FAILED: timeout
	see: sanity-out/mps2_an385/tests/kernel/context/kernel.common/handler.log
 8/73 mps2_an385                tests/kernel/arm_runtime_nmi/arch.interrupt        FAILED: timeout
	see: sanity-out/mps2_an385/tests/kernel/arm_runtime_nmi/arch.interrupt/handler.log
 9/73 mps2_an385                tests/kernel/fifo/fifo_api/kernel.fifo             FAILED: timeout
	see: sanity-out/mps2_an385/tests/kernel/fifo/fifo_api/kernel.fifo/handler.log
10/73 mps2_an385                tests/kernel/fifo/fifo_api/kernel.fifo.poll        FAILED: timeout
	see: sanity-out/mps2_an385/tests/kernel/fifo/fifo_api/kernel.fifo.poll/handler.log
11/73 mps2_an385                tests/kernel/fifo/fifo_usage/kernel.fifo.usage     FAILED: timeout
	see: sanity-out/mps2_an385/tests/kernel/fifo/fifo_usage/kernel.fifo.usage/handler.log
12/73 mps2_an385                tests/kernel/fifo/fifo_timeout/kernel.fifo.timeout FAILED: timeout
	see: sanity-out/mps2_an385/tests/kernel/fifo/fifo_timeout/kernel.fifo.timeout/handler.log
13/73 mps2_an385                tests/kernel/fifo/fifo_timeout/kernel.fifo.timeout.poll FAILED: timeout
	see: sanity-out/mps2_an385/tests/kernel/fifo/fifo_timeout/kernel.fifo.timeout.poll/handler.log
14/73 mps2_an385                tests/kernel/gen_isr_table/arch.interrupt          FAILED: timeout
	see: sanity-out/mps2_an385/tests/kernel/gen_isr_table/arch.interrupt/handler.log
15/73 mps2_an385                tests/kernel/fifo/fifo_usage/kernel.fifo.usage.poll FAILED: timeout
	see: sanity-out/mps2_an385/tests/kernel/fifo/fifo_usage/kernel.fifo.usage.poll/handler.log
16/73 mps2_an385                tests/kernel/interrupt/arch.interrupt              FAILED: timeout
	see: sanity-out/mps2_an385/tests/kernel/interrupt/arch.interrupt/handler.log
17/73 mps2_an385                tests/kernel/lifo/lifo_api/kernel.lifo             FAILED: timeout
	see: sanity-out/mps2_an385/tests/kernel/lifo/lifo_api/kernel.lifo/handler.log
18/73 mps2_an385                tests/kernel/lifo/lifo_usage/kernel.lifo.usage     FAILED: timeout
	see: sanity-out/mps2_an385/tests/kernel/lifo/lifo_usage/kernel.lifo.usage/handler.log
19/73 mps2_an385                tests/kernel/mbox/mbox_api/kernel.mailbox          FAILED: timeout
	see: sanity-out/mps2_an385/tests/kernel/mbox/mbox_api/kernel.mailbox/handler.log
20/73 mps2_an385                tests/kernel/mbox/mbox_usage/kernel.mailbox        FAILED: timeout
	see: sanity-out/mps2_an385/tests/kernel/mbox/mbox_usage/kernel.mailbox/handler.log
21/73 mps2_an385                tests/kernel/mem_heap/mheap_api_concept/kernel.memory_heap FAILED: timeout
	see: sanity-out/mps2_an385/tests/kernel/mem_heap/mheap_api_concept/kernel.memory_heap/handler.log
22/73 mps2_an385                tests/kernel/mem_pool/mem_pool/kernel.memory_pool  FAILED: timeout
	see: sanity-out/mps2_an385/tests/kernel/mem_pool/mem_pool/kernel.memory_pool/handler.log
23/73 mps2_an385                tests/kernel/mem_pool/mem_pool_api/kernel.memory_pool FAILED: timeout
	see: sanity-out/mps2_an385/tests/kernel/mem_pool/mem_pool_api/kernel.memory_pool/handler.log
24/73 mps2_an385                tests/kernel/mem_pool/mem_pool_concept/kernel.memory_pool FAILED: timeout
	see: sanity-out/mps2_an385/tests/kernel/mem_pool/mem_pool_concept/kernel.memory_pool/handler.log
25/73 mps2_an385                tests/kernel/mem_pool/mem_pool_threadsafe/kernel.memory_pool FAILED: timeout
	see: sanity-out/mps2_an385/tests/kernel/mem_pool/mem_pool_threadsafe/kernel.memory_pool/handler.log
26/73 mps2_an385                tests/kernel/mem_pool/sys_mem_pool/kernel.memory_pool FAILED: timeout
	see: sanity-out/mps2_an385/tests/kernel/mem_pool/sys_mem_pool/kernel.memory_pool/handler.log
27/73 mps2_an385                tests/kernel/mem_protect/obj_validation/kernel.memory_protection.obj_validation FAILED: timeout
	see: sanity-out/mps2_an385/tests/kernel/mem_protect/obj_validation/kernel.memory_protection.obj_validation/handler.log
28/73 mps2_an385                tests/kernel/mem_protect/protection/kernel.memory_protection.protection FAILED: timeout
[... 213 lines cut ...]
Generating gcda files
Working on /path/to/zephyr.git/sanity-out/mps2_an385/tests/kernel/common/kernel.common/handler.log
Generating gcda files
Working on /path/to/zephyr.git/sanity-out/mps2_an385/tests/kernel/mutex/mutex_api/kernel.mutex/handler.log
Generating gcda files
Working on /path/to/zephyr.git/sanity-out/mps2_an385/tests/kernel/mutex/mutex/kernel.mutex/handler.log
Generating gcda files
Working on /path/to/zephyr.git/sanity-out/mps2_an385/tests/subsys/jwt/libraries.encoding/handler.log
Generating gcda files
Working on /path/to/zephyr.git/sanity-out/mps2_an385/tests/subsys/logging/log_msg/logging.log_msg/handler.log
Generating gcda files
Working on /path/to/zephyr.git/sanity-out/mps2_an385/tests/subsys/logging/log_list/logging.log_list/handler.log
Generating gcda files
Working on /path/to/zephyr.git/sanity-out/mps2_an385/tests/subsys/logging/log_output/logging.log_output/handler.log
Generating gcda files
Working on /path/to/zephyr.git/sanity-out/mps2_an385/tests/subsys/logging/log_core/logging.log_core/handler.log
Generating gcda files
HTML report generated: /path/to/zephyr.git/sanity-out/coverage/index.html
8 of 73 tests passed with 0 warnings in 969 seconds
```

**Environment (please complete the following information):**
 - OS: fedora29
 - Toolchain (e.g Zephyr SDK, ...): zephyr
 - Commit SHA or Version used: a907e69

wentongwu, 2019-03-19:
run the case(e.g. kernel.common) manually, it works well. It should be someone's commit cause this issue. I'm checking this issue.

wentongwu, 2019-03-21:
can reproduce the issue(tests/kernel/workq/work_queue_api case) when CONFIG_COVERAGE=y. And if CONFIG_COVERAGE=n, the issue won't happen. and the output is that:
qemu-system-arm: warning: nic lan9118.0 has no peer
***** MPU FAULT *****
   Data Access Violation
   MMFAR Address: 0x20018014
***** Hardware exception *****
Current thread ID = 0x20014c58
Faulting instruction address = 0x20fe0
Fatal fault in thread 0x20014c58! Aborting.

check the zephyr.lst file, and address 0x20018014 is in sys_work_q_stack and 0x20014c58 is the thread ID of k_sys_work_q.thread. 
And also check arm mpu code, the GCOV section is configured as a mpu region with K_MEM_PARTITION_P_RW_U_RW(I think it's read and write) attribute. It seems there will be MPU fault if add the coverage code(-ftest-coverage -fprofile-arcs).  
@andrewboie  could you please give some ideas ? Thanks

andrewboie, 2019-03-21:
@wentongwu I don't understand well what the compiler is doing when profiling is enabled. I saw your email earlier this morning but I haven't had a chance to analyze it yet.

What code is at address 0x20fe0 which is accessing this data?
Nominally, the thread is supposed to be able to access its own stack buffer, so it's not clear to me why we would be faulting in this case.

wentongwu, 2019-03-21:
from below, it seems 0x20fe4 - 0x20ff0 is some code added by compiler for gcov. And it seems 0x20fe0 is doing stack pushing operation. I'm not sure if there is stack overflow.  @andrewboie 
00020fd8 <z_remove_thread_from_ready_q>:
{
   20fd8:       e92d 4ff0       stmdb   sp!, {r4, r5, r6, r7, r8, r9, sl, fp, lr}
   20fdc:       b0ad            sub     sp, #180        ; 0xb4
   20fde:       af00            add     r7, sp, #0
   20fe0:       f8c7 0084       str.w   r0, [r7, #132]  ; 0x84
   20fe4:       4b4c            ldr     r3, [pc, #304]  ; (21118 <z_remove_thread_from_ready_q+0x140>)
   20fe6:       cb18            ldmia   r3, {r3, r4}
   20fe8:       1c59            adds    r1, r3, #1
   20fea:       f144 0200       adc.w   r2, r4, #0
   20fee:       4b4a            ldr     r3, [pc, #296]  ; (21118 <z_remove_thread_from_ready_q+0x140>)
   20ff0:       e9c3 1200       strd    r1, r2, [r3]
        LOCKED(&sched_spinlock) {
   20ff4:       2300            movs    r3, #0
   20ff6:       f8c7 3098       str.w   r3, [r7, #152]  ; 0x98
   20ffa:       4b48            ldr     r3, [pc, #288]  ; (2111c <z_remove_thread_from_ready_q+0x144>)
   20ffc:       f8c7 30ac       str.w   r3, [r7, #172]  ; 0xac
        __asm__ volatile(

andrewboie, 2019-03-21:
can you set a breakpoint in the compiler and backtrace to when it crashes?
Try reproducing on qemu_x86 it prints a backtrace report on exception, I think there is a similar issue opened with that arch.

andrewboie, 2019-03-21:
Can you backtrace to where in the source code this is coming from?
Try qemu_x86, it prints a backtrace and also has memory protection enabled.

wentongwu, 2019-03-21:
@andrewboie  I will do that. And I increased system workq's stack size, system workq thread will not be mpu fault, another thread trigger's mpu fault, I'm trying to increase this thread's stack size.

wentongwu, 2019-03-21:
do the same test on qemu_x86, but it's Stack Check Fail as #14499 instead of MPU fault. And for mps2_an385, enlarge the stack of thread, some not passed tests can pass, and the others still MPU fault.

wentongwu, 2019-03-21:
go back to old version 9072d34f306078acdc9d950c9beae5cc0738ea19, with 0.9.5 SDK the test can pass, but with 0.10.0 SDK, still MPU fault. both tests are set -DCONFIG_COVERAGE=y .

wentongwu, 2019-03-21:
with latest code version and SDK 0.9.5, the test can also pass with DCONFIG_COVERAGE=y.

before the test, I apply below patch:
diff --git a/cmake/toolchain/zephyr/host-tools.cmake b/cmake/toolchain/zephyr/host-tools.cmake
index 21417fb858..e9d8a71ecb 100644
--- a/cmake/toolchain/zephyr/host-tools.cmake
+++ b/cmake/toolchain/zephyr/host-tools.cmake
@@ -15,7 +15,7 @@ if(NOT ZEPHYR_SDK_INSTALL_DIR)
   return()
 endif()
 
-set(REQUIRED_SDK_VER 0.10.0)
+set(REQUIRED_SDK_VER 0.9.5)
 set(TOOLCHAIN_VENDOR zephyr)
 set(TOOLCHAIN_ARCH x86_64)

cinlyooi-intel, 2019-03-21:
@wentongwu Confirmed if switch to 0.9.5, sanitycheck --coverage works. If not, I have a lot of handler crashs and timeouts.

andrewboie, 2019-03-21:
Even if the errors are different between the two targets, I have a suspicion that this is a stack overflow in both cases. The new GCC may be using a lot more stack space when profiling is enabled in the 0.10 SDK.

Some things to investigate:

- What is the offset from the beginning of sys_work_q_stack where the MPU fault occurred? If it is within the first 32 bytes, on ARM the first 32 bytes of the stack buffer are reserved for a small MPU region which detects stack overflows in supervisor mode.
- What are the characteristics of the MPU region that is causing this access to fault?

Stack overflows are detected via MPU faults on ARM. The code is supposed to do some checks to see if an MPU fault is due to a stack overflow instead of some other reason, and report  Stack Check Fail, but it might not be working correctly.

wentongwu, 2019-03-22:
thanks @andrewboie.
MMFAR Address: 0x20018014 and sys_work_q_stack: 0x0000000020018000, so the address of the location that generated MPU fault located in the first 32bytes of this thread.  And the first 32 bytes is an exception frame of ARM which will be used for the first Swap. I'm reviewing zephyr's mpu code for arm, try to finding the attribute of this MPU region.

by the way, as below the new compiler will consume more stack space when doing function call.   
  new compiler:
00020fd8 <z_remove_thread_from_ready_q>:
{
   20fd8:       e92d 4ff0       stmdb   sp!, {r4, r5, r6, r7, r8, r9, sl, fp, lr}
   20fdc:       b0ad            sub     sp, #180        ; 0xb4
   20fde:       af00            add     r7, sp, #0
   20fe0:       f8c7 0084       str.w   r0, [r7, #132]  ; 0x84

old compiler:
0001c350 <z_remove_thread_from_ready_q>:
{
   1c350:       b590            push    {r4, r7, lr}
   1c352:       b08d            sub     sp, #52 ; 0x34
   1c354:       af00            add     r7, sp, #0
   1c356:       6078            str     r0, [r7, #4]

wentongwu, 2019-03-25:
> 
> Some things to investigate:
> 
>     * What is the offset from the beginning of sys_work_q_stack where the MPU fault occurred? If it is within the first 32 bytes, on ARM the first 32 bytes of the stack buffer are reserved for a small MPU region which detects stack overflows in supervisor mode.
  ----> yes, this 32 bytes is allocated with the real stack and it's called MPU Guard region. And the layout is like below:
  ---------------------------- <---- Stack allocation address + stack size +
  |                                 |            MPU_GUARD_ALIGN_AND_SIZE
  |  Some thread data  | <---- Defined when thread is created
  |        ...                      |
  |---------------------       | <---- Actual initial stack ptr
  |  Initial Stack Ptr        |       aligned to STACK_ALIGN_SIZE
  |        ...                      |
  |        ...                      |
  |        ...                      |
  |  Stack Ends             |
  |-------------------------- <---- Stack Buffer Ptr from API
  |  MPU Guard,           |
  |     if present             |
  --------------------------- <---- Stack Allocation address

> 
>     * What are the characteristics of the MPU region that is causing this access to fault?
 ----> this MPU region(MPU_GUARD_ALIGN_AND_SIZE = 32) is configured as Read only and none executable(_K_MEM_PARTITION_P_RO_U_NA). 
And doing "20fe0: f8c7 0084 str.w r0, [r7, #132] ; 0x84" means writing some data in this thread's stack, so it's the read-only attribute that casue MPU fault.

andrewboie, 2019-03-25:
Yeah, that's a stack overflow then. We'll need to fix the ARM code that reports overflows, @ioannisg 
Try increasing CONFIG_TEST_EXTRA_STACKSIZE that should cover most stacks

wentongwu, 2019-03-25:
yes, as previous comment, increasing stack could make some case pass. But there is new error happen. Checking the code, it seems stack issue again, from this line(" Stacking error (context area might be not valid"), it should be fault happen during exception stacking and then do same check and abort this thread. Maybe we should increase exception's stack. Let me review the code of failed case and see if there is exception and irq.

Running test suite workqueue_api
===================================================================
starting test - test_workq_start_before_submit
PASS - test_workq_start_before_submit
===================================================================
starting test - test_user_workq_start_before_submit
***** MPU FAULT *****
  Stacking error (context area might be not valid)
  Data Access Violation
  MMFAR Address: 0x2002880c
***** Stack Check Fail! *****
Current thread ID = 0x2001450c
Faulting instruction address = 0x0
Fatal fault in thread 0x2001450c! Aborting.

wentongwu, 2019-03-25:
0x2002880c is located in .priv_stacks.noinit.

andrewboie, 2019-03-25:
mis-click

andrewboie, 2019-03-25:
> Maybe we should increase exception's stack

Indeed, this is controlled by CONFIG_ISR_STACK_SIZE

wentongwu, 2019-03-26:
> > Maybe we should increase exception's stack
> 
> Indeed, this is controlled by CONFIG_ISR_STACK_SIZE

@andrewboie  did the test, it seems not work for increasing exception stack. And failing code try to access priv_stack, I didn't understand this stack well for now...

wentongwu, 2019-03-26:
it works when increasing CONFIG_PRIVILEGED_STACK_SIZE. I will test more.

wentongwu, 2019-03-26:
with all kinds of stack size's adjustment, it seems the timeout issue on platform mps2_an385 has been fixed. Overnight test is running and also I'm cooking the patch. Thanks

cinlyooi-intel, 2019-03-29:
@wentongwu : How do I increase CONFIG_PRIVILEGED_STACK_SIZE? and to what value? Many thanks

wentongwu, 2019-03-29:
> @wentongwu : How do I increase CONFIG_PRIVILEGED_STACK_SIZE? and to what value? Many thanks

Today I will submit a PR for this issue, but it will take some time for mps2_an385. I have already submit a PR for qemu_x86. Thanks.

nashif, 2019-04-03:
@cinlyooi-intel:

Making this bug the master for all coverage related bugs, please add issues here instead of opening new bugs for each test/sample that might fail.

Here is a list of closed issues related to coverage:

#15149 #15146 #15150 #15151 #15152  #15153

nashif, 2019-04-03:
one more #15148

PR #15005 (2019-04-04): tests: adjust stack size for mps2_an385's coverage test
for SDK 0.10.0, it consumes more stack size when coverage enabled
on mps2_an385 platform, adjust stack size for most of the test cases,
otherwise there will be stack overflow.

Fixes: #14500
commit b991962a (2019-04-04): tests: adjust stack size for qemu_x86 and mps2_an385's coverage test

for SDK 0.10.0, it consumes more stack size when coverage enabled
on qemu_x86 and mps2_an385 platform, adjust stack size for most of
the test cases, otherwise there will be stack overflow.

Fixes: #14500.
closed 2019-04-04 by nashif