zephyrproject-rtos/zephyr #14501: crash in qemu_x86_64:tests/kernel/fifo/fifo_usage/kernel.fifo.usage

andrewboie, 2019-03-14:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Screenshots or console output**
```
c[?7l[2J[0mSeaBIOS (version rel-1.12.0-0-ga698c8995f-prebuilt.qemu.org)
Booting from ROM..Running test suite test_fifo_usage
===================================================================
starting test - test_single_fifo_play
PASS - test_single_fifo_play
===================================================================
starting test - test_dual_fifo_play

    Assertion failed at /home/buildslave/src/github.com/zephyrproject-rtos/zephyr/tests/kernel/fifo/fifo_usage/src/main.c:198: test_dual_fifo_play: rx_data not equal to (void *)&data1[i]

FAIL - test_dual_fifo_play
===================================================================
starting test - test_isr_fifo_play
ASSERTION FAIL [thread->base.pended_on] @ /home/buildslave/src/github.com/zephyrproject-rtos/zephyr/kernel/sched.c:388
*** FATAL ERROR vector 5 code 1016
***  RIP 8:0x1014dd RSP 16:0x107668 RFLAGS 0x2
***  RAX 0x105cad RCX 0xa RDX 0x3f8 RSI 0x184 RDI 0x105fff
***  R8 0xffffffff R9 0x106c28 R10 0x106c28 R11 0x0
ASSERTION FAIL [z_spin_lock_valid(l)] @ ../../../../../../../include/spinlock.h:77
	Recursive spinlock
*** FATAL ERROR vector 5 code 1016
***  RIP 8:0x1014dd RSP 16:0x107528 RFLAGS 0x6
***  RAX 0x105d1c RCX 0xa RDX 0x3f8 RSI 0x4d RDI 0x105ccb
***  R8 0xffffffff R9 0x105fff R10 0x1 R11 0x0
ASSERTION FAIL [z_spin_lock_valid(l)] @ ../../../../../../../include/spinlock.h:77
	Recursive spinlock
*** FATAL ERROR vector 5 code 1016
***  RIP 8:0x1014dd RSP 16:0x1073e8 RFLAGS 0x6
***  RAX 0x105d1c RCX 0xa RDX 0x3f8 RSI 0x4d RDI 0x105ccb
*ASSERTION FAIL [z_spin_lock_valid(l)] @ ../../../../../../../include/spinlock.h:77
	Recursive spinlock
*** FATAL ERROR vector 5 code 1016
***  RIP 8:0x1014dd RSP 16:0x107128 RFLAGS 0x6
***  RAX 0x105d1c RCX 0xa RDX 0x3f8 RSI 0x4d RDI 0x105ccb
***  R8 0xffffffff R9 0x105ccb R10 0x1 R11 0x0
ASSERTION FAIL [z_spin_lock_valid(l)] @ ../../../../../../../include/spinlock.h:77
	Recursive spinlock
*** FATAL ERROR vector 5 code 1016
***  RIP 8:0x1014dd RSP 16:0x106fe8 RFLAGS 0x6
***  RAX 0x105d1c RCX 0xa RDX 0x3f8 RSI 0x4d RDI 0x105ccb
***  R8 0xffffffff R9 0x105ccb R10 0x1 R11 0x0
ASSERTION FAIL [z_spin_lock_valid(l)] @ ../../../../../../../include/spinlock.h:77
	Recursive spinlock
*** FATAL ERROR vector 5 code 1016
***  RIP 8:0x1014dd RSP 16:0x106ea8 RFLAGS 0x2
***  RAX 0x105d1c RCX 0xa RDX 0x3f8 RSI 0x4d RDI 0x105ccb
***  R8 0xffffffff R9 0x105ccb R10 0x1 R11 0x0
ASSERTION FAIL [z_spin_lock_valid(l)] @ ../../../../../../../include/spinlock.h:77
	Recursive spinlock
*** FATAL ERROR vector 5 code 1016
***  RIP 8:0x1014dd RSP 16:0x106d68 RFLAGS 0x2
***  RAX 0x105d1c RCX 0xa RDX 0x3f8 RSI 0x4d RDI 0x105ccb
***  R8 0xffffffff R9 0x105ccb R10 0x1 R11 0x0
ASSERTION FAIL [z_spin_lock_valid(l)] @ ../../../../../../../include/spinlock.h:77
	Recursive spinlock
*** FATAL ERROR vector 5 code 1016
***  RIP 8:0x1014dd RSP 16:0x106c28 RFLAGS 0x6
***  RAX 0x105d1c RCX 0xa RDX 0x3f8 RSI 0x4d RDI 0x105ccb
***  R8 0xffffffff R9 0x105ccb R10 0x1 R11 0x0
ASSERTION FAIL [z_spin_lock_valid(l)] @ ../../../../../../../include/spinlock.h:77
	Recursive spinlock
*** FATAL ERROR vector 5 code 1016
***  RIP 8:0x1014dd RSP 16:0x106ae8 RFLAGS 0x6
***  RAX 0x105d1c RCX 0xa RDX 0x3f8 RSI 0x4d RDI 0x105ccb
***  R8 0xffffffff R9 0x105ccb R10 0x1 R11 0x0
ASSERTION FAIL [z_spin_lock_valid(l)] @ ../../../../../../../include/spinlock.h:77
	Recursive spinlock
*** FATAL ERROR vector 5 code 1016
***  RIP 8:0x1014dd RSP 16:0x1069a8 RFLAGS 0x2
***  RAX 0x105d1c RCX 0xa RDX 0x3f8 RSI 0x4d RDI 0x105ccb
***  R8 0xffffffff R9 0x105ccb R10 0x1 R11 0x0
```

andyross, 2019-03-15:
Spurious SMP failure.  This one doesn't fail that much more than, say, mps2_an385 does due to timing issues, so it got sorta hidden in the noise.  My "timing-reliable parallel qemu test" rig is up to 595 iterations of this now with zero failures.

andyross, 2019-03-15:
(Just remembered to shut the test down as it crossed 4400 successful tests without a failure)

PR #14582 (2019-03-15): tests/kernel/fifo/fifo_usage: Disable SMP
This test isn't SMP-safe and won't pass reliably on x86_64 by default
(though it does pass often enough to get CI passes on most things, it
fails spuriously in ways that aren't timing related).  Turn off the
second CPU.  Fixes #14501
commit 5a56ee56 (2019-03-15): tests/kernel/fifo/fifo_usage: Disable SMP

This test isn't SMP-safe and won't pass reliably on x86_64 by default
(though it does pass often enough to get CI passes on most things, it
fails spuriously in ways that aren't timing related).  Turn off the
second CPU.  Fixes #14501
closed 2019-03-15 by nashif