zephyrproject-rtos/zephyr #14504: mempool can return success if no memory was available

pdunaj, 2019-03-14:
**Describe the bug**
We replicated a bug where assertion on data failed. Problem replicates randomly and rarely (I would say once per hour of operation). Quick investigation shown that application gets `0x4` from `k_malloc`. Quick look at the code shown that this is possible if `_sys_mem_pool_block_alloc` returns success but `block.data` is NULL.
Looking at `_sys_mem_pool_block_alloc` proved that this is possible as the loop that is allocating a block gives no guarantee of actually allocating (i.e. `data` can be NULL) but it is always returning zero.

As a bonus there is no lock taken when `level_empty` is called in the first loop so return from it can be invalid (it can be reported as nonempty when in fact it is). I am not 100% sure that giving a lock here would serve any purpose as in the end there is absolutely no guarantee what will be available when second loop of `_sys_mem_pool_block_alloc` is called.

**To Reproduce**
Steps to reproduce the behavior:
Allocate from various context close to no memory condition.

**Expected behavior**
`'k_malloc' should return NULL if no memory is available.

**Impact**
Showstopper

**Screenshots or console output**
N/A

**Environment (please complete the following information):**
ncs zephyr `745326266d9c32ba3aa67d5af1f727059d9a88e3` (from `d9876be` upstream)

**Additional context**
N/A

pdunaj, 2019-03-14:
I reviewed the code and it seems the problem can happen even if memory is available. See https://github.com/zephyrproject-rtos/zephyr/issues/14508

pdunaj, 2019-03-14:
FYI @andyross

pdunaj, 2019-03-14:
Fix pushed

andyross, 2019-03-14:
There was an almost identical bug about a year ago, and I thought this was all fixed.  May have been reintroduced?  I'll take it.

andyross, 2019-03-14:
Sorry, reassign back.  Didn't notice you had a patch up.

PR #14514 (2019-03-19): lib: mempool: Fix issues in mempool allocator
This commits fixes three issues:
Fixes: #14504
1. Allocator returning success even if allocation failed

Fixes: #14508
2. Allocator fails to allocate while memory is available
but was hidden by other context calling free.
3. Allocator fails to allocate while memory is available
but was moved to another level.
commit 2c7d6800 (2019-03-19): lib: mempool: Return error if no block found

Return -ENOMEM if no block is available on any level.

Fixes: #14504
commit 41e90630 (2019-03-19): lib: mempool: Synchronize level checks

Do not perform early level usage check. This can lead to situation
where block is seen as available on level when it was taken from
the other context.

Fixes: #14504
closed 2019-03-19 by galak