zephyrproject-rtos/zephyr #14523: echo-client doesn't close socket if echo-server is offline

mtuxpe, 2019-03-14:
**Describe the bug**
The echo-client app calls socket() and then connect() 
If echo-server is offline then conf.ipv4.sock and  conf.ipv6.sock =0 
but a net connection was opened.

void stop_tcp(void)
{
	if (IS_ENABLED(CONFIG_NET_IPV6)) {
		if (conf.ipv6.tcp.sock > 0) {
			(void)close(conf.ipv6.tcp.sock);
		}
	}

	if (IS_ENABLED(CONFIG_NET_IPV4)) {
		if (conf.ipv4.tcp.sock > 0) {
			(void)close(conf.ipv4.tcp.sock);
		}
	}
}

What have you tried to diagnose or workaround this issue?

Close socket even conf.ipv4.tcp.sock = 0  or  conf.ipv6.tcp.sock= 0 

void stop_tcp(void)
{
	if (IS_ENABLED(CONFIG_NET_IPV6)) {
		if (conf.ipv6.tcp.sock >= 0) {
			(void)close(conf.ipv6.tcp.sock);
		}
	}

	if (IS_ENABLED(CONFIG_NET_IPV4)) {
		if (conf.ipv4.tcp.sock >= 0) {
			(void)close(conf.ipv4.tcp.sock);
		}
	}
}

**To Reproduce**
Steps to reproduce the behavior:
1. Build echo-client application with net shell CONFIG_NET_SHELL=y
2. Turn off your echo server application
3. Open net shell console execute "net conn"

**Expected behavior**
bsd socket must close close all pending connections

**Impact**

**Screenshots or console output**
Using net shell to show open connections
uart:~$ net conn
     Context    Iface         Flags Local               Remote
[ 1] 0x20001504 0x20015000    6ST         [::]:65042    [fd11:1111:1122:0:4034:92fa:a904:9c4c]:4242

TCP        Context   Src port Dst port   Send-Seq   Send-Ack  MSS    State
0x200029a0 0x20001504   65042     4242 1082166264    

**Environment (please complete the following information):**
 - Ubuntu 16
 - Zephyr SDK 0.10
 - Commit b87920bf3c789ad4eeebb749d9a06ae72f15d589

**Additional context**
Add any other context about the problem here.

jukkar, 2019-03-18:
> ```
> void stop_tcp(void)
> {
> 	if (IS_ENABLED(CONFIG_NET_IPV6)) {
> 		if (conf.ipv6.tcp.sock >= 0) {
> 			(void)close(conf.ipv6.tcp.sock);
> 		}
> 	}
> 	if (IS_ENABLED(CONFIG_NET_IPV4)) {
> 		if (conf.ipv4.tcp.sock >= 0) {
> 			(void)close(conf.ipv4.tcp.sock);
> 		}
> 	}
> }
> ```

So you tried this and did it work as expected?

mtuxpe, 2019-03-18:
Hi Jukka,
Yes work as expected.No pending connections.I used socket echo-client
example. Probably other socket applications has the same bug;
Best regards,
Marcio Montenegro

jukkar, 2019-03-18:
> Yes work as expected.No pending connections.

Great, could you send a PR that fixes the issue? You fix looks just fine.

mtuxpe, 2019-03-18:
Hi Jukka, PR sent:
https://github.com/zephyrproject-rtos/zephyr/pull/14614

rveerama1, 2019-03-26:
> Hi Jukka, PR sent: #14614

PR merged. So closing this issue. Reopen if issue still exists.

closed 2019-03-26 by rveerama1