zephyrproject-rtos/zephyr #14527: [wip] Generic support for out-of-tree drivers

thomasstenersen, 2019-03-14:
An out-of-source subsystem requires control over a specific peripheral and needs  a generic way of forcing its own driver. An example is a proprietary Bluetooth controller that must control the NVMC (flash driver) in order to meet real-time requirements. 

**Requirements**
- support for de-selecting an in-tree driver (prevent user from selecting the in-tree driver)
- do not pollute upstream Zephyr with configuration options not related to Zephyr itself

A work-around for this problem has been made in https://github.com/zephyrproject-rtos/zephyr/pull/14509.

The solution to this use-case may boil down to solving https://github.com/zephyrproject-rtos/zephyr/issues/8181.

ulfalizer, 2019-03-14:
IMO, unless something absolutely has to be out-of-tree for legal reasons (should double-check and not assume), make it in-tree. No risk of having the interface drift out of sync, potentially free improvements to the code from people working on the main repo, and simpler code.

carlescufi, 2019-03-14:
@ulfalizer sometimes in-tree is simply not an option. Sometimes you don't own the IP, you have acquired it, so you are not allowed to push it upstream. In other cases you  own it but you depend on other infrastructure that makes it infeasible to have it in-tree. The question is how to support those use cases effectively since we are stuck with them.

ulfalizer, 2019-03-14:
Simplest option might be to just have custom patches on top of Zephyr. That keeps the core clean, and might not be any more of a maintenance burden in practice (in the Nordic case at least).

thomasstenersen, 2019-03-15:
@ulfalizer As @carlescufi says, there is a plethora of reasons for not upstreaming stuff, and often multiple reasons at the same time. Having custom patches on top of Zephyr is not very easy, only the last couple of weeks I've fixed several merge conflicts for my patches (https://github.com/zephyrproject-rtos/zephyr/pull/14509/commits/54d99bcc3d1ea47b9fa4eb63995954fb9ca241e6) -- and these patches have been pretty simple. Also, if we were to start adopting a solution of patching upstream to meet our needs, IMO that's a slippery slope that's hard to get out of. We'd rather contribute more to upstream and keep our fork as vanilla as possible.

gmarull, 2022-09-26:
We already support oot drivers as _first-class_ citizens. An application will need to de-select an in-tree driver if it provides an alternative driver for the same compatible, though.

closed 2022-09-26 by gmarull