zephyrproject-rtos/zephyr #14528: dts: nrf52840_pca10059: No way to use "fstab-debugger.dts"

MarekPieta, 2019-03-14:
DTS for nrf52840_pca10059 has no way to use "fstab-debugger.dts" (flash partition table without Nordic nRF5 bootloader). We also cannot add new flash configuration later in overlay file for application (duplicate_label ERROR). Currently we may only modify dts for the board in Zephyr (changing included file), but it isn't a good solution.

lemrey, 2019-03-15:
You can use the `fstab-debugger.dts` by including it in the board DTS file; that will work as documented, so the issue title is incorrect. This is not really a problem with this specific board, and it is a consequence of Zephyr's model. I agree to the bottomline here, but I think that should be discussed in a broader context.

carlescufi, 2019-03-26:
Changing to enhancement since this is expected although can be improved.

jcorde, 2019-05-10:
Using `fstab-debugger.dts` is not enough as `BOARD_HAS_NRF5_BOOTLOADER` still selected by `Kconfig`. So `Kconfig.defconfig` sets `FLASH_LOAD_OFFSET` to 0x1000.

jcorde, 2019-05-10:
If we follow [Programming Option 3 - External debug probe](https://docs.zephyrproject.org/latest/boards/arm/nrf52840_pca10059/doc/index.html#option-3-using-an-external-debug-probe), the offset will not be set to 0x000.

crespum, 2020-06-09:
> Using `fstab-debugger.dts` is not enough as `BOARD_HAS_NRF5_BOOTLOADER` is still selected by `Kconfig`. So `Kconfig.defconfig` sets `FLASH_LOAD_OFFSET` to 0x1000.

This is correct but it's only slightly related to the original issue. We could just improve the documentation IMHO.

To use an external debugging probe:

1. Edit `nrf52840dongle_nrf52840.dts` and replace `fstab-stock.dts` by `fstab-debugger.dts`.
2. Run `make menuconfig`
    1. Disable `BOARD_HAS_NRF5_BOOTLOADER`
    2. Set `FLASH_LOAD_OFFSET` to 0

In any case, shall I open a new issue or even a new PR?

carlescufi, 2021-12-20:
This is currently being covered by https://github.com/zephyrproject-rtos/zephyr/pull/40669, but I will send a PR for the doc

closed 2021-12-20 by carlescufi