zephyrproject-rtos/zephyr #14540: kernel: message queue MACRO not compatible with C++

rodrigopex, 2019-03-14:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?

*It is not possible to use the Zephyr defined macro for creating a message queue in C++.*

**To Reproduce**
Steps to reproduce the behavior:

*Just add a message queue macro to init a message queue and compile it.*

``` c++
K_MSGQ_DEFINE(lora_input_queue, sizeof(u8_t), 256, 1);
...
```

**Expected behavior**
A clear and concise description of what you expected to happen.

*The code would run in C++ as well, but is not compiling.*

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

*The code does not compile. The developer must use the non-macro way to init a message queue which is not so easy as the macro.*

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

*The compilation error is:*
``` console
In file included from ../zephyr/include/device.h:11:0,
                 from ../src/tc_lora_uart_task.cpp:6:
../zephyr/include/kernel.h:3184:2: sorry, unimplemented: non-trivial designated initializers not supported
  }
  ^
../zephyr/include/kernel.h:3229:9: note: in expansion of macro ‘_K_MSGQ_INITIALIZER’
         _K_MSGQ_INITIALIZER(q_name, _k_fifo_buf_##q_name,     \
         ^~~~~~~~~~~~~~~~~~~
../src/tc_lora_uart_task.cpp:19:1: note: in expansion of macro ‘K_MSGQ_DEFINE’
 K_MSGQ_DEFINE(lora_input_queue, sizeof(u8_t), 256, 1);
 ^~~~~~~~~~~~~
../zephyr/include/kernel.h:3184:2: sorry, unimplemented: non-trivial designated initializers not supported
  }
  ^
../zephyr/include/kernel.h:3229:9: note: in expansion of macro ‘_K_MSGQ_INITIALIZER’
         _K_MSGQ_INITIALIZER(q_name, _k_fifo_buf_##q_name,     \
         ^~~~~~~~~~~~~~~~~~~
../src/tc_lora_uart_task.cpp:19:1: note: in expansion of macro ‘K_MSGQ_DEFINE’
 K_MSGQ_DEFINE(lora_input_queue, sizeof(u8_t), 256, 1);
 ^~~~~~~~~~~~~
../zephyr/include/kernel.h:3184:2: sorry, unimplemented: non-trivial designated initializers not supported
  }
  ^
../zephyr/include/kernel.h:3229:9: note: in expansion of macro ‘_K_MSGQ_INITIALIZER’
         _K_MSGQ_INITIALIZER(q_name, _k_fifo_buf_##q_name,     \
         ^~~~~~~~~~~~~~~~~~~
../src/tc_lora_uart_task.cpp:19:1: note: in expansion of macro ‘K_MSGQ_DEFINE’
 K_MSGQ_DEFINE(lora_input_queue, sizeof(u8_t), 256, 1);
 ^~~~~~~~~~~~~
../zephyr/include/kernel.h:3184:2: sorry, unimplemented: non-trivial designated initializers not supported
  }
  ^
../zephyr/include/kernel.h:3229:9: note: in expansion of macro ‘_K_MSGQ_INITIALIZER’
         _K_MSGQ_INITIALIZER(q_name, _k_fifo_buf_##q_name,     \
         ^~~~~~~~~~~~~~~~~~~
../src/tc_lora_uart_task.cpp:19:1: note: in expansion of macro ‘K_MSGQ_DEFINE’
 K_MSGQ_DEFINE(lora_input_queue, sizeof(u8_t), 256, 1);
 ^~~~~~~~~~~~~
../zephyr/include/kernel.h:3184:2: sorry, unimplemented: non-trivial designated initializers not supported
  }
  ^
../zephyr/include/kernel.h:3229:9: note: in expansion of macro ‘_K_MSGQ_INITIALIZER’
         _K_MSGQ_INITIALIZER(q_name, _k_fifo_buf_##q_name,     \
         ^~~~~~~~~~~~~~~~~~~
../src/tc_lora_uart_task.cpp:19:1: note: in expansion of macro ‘K_MSGQ_DEFINE’
 K_MSGQ_DEFINE(lora_input_queue, sizeof(u8_t), 256, 1);
 ^~~~~~~~~~~~~
../zephyr/include/kernel.h:3184:2: sorry, unimplemented: non-trivial designated initializers not supported
  }
  ^
../zephyr/include/kernel.h:3229:9: note: in expansion of macro ‘_K_MSGQ_INITIALIZER’
         _K_MSGQ_INITIALIZER(q_name, _k_fifo_buf_##q_name,     \
         ^~~~~~~~~~~~~~~~~~~
../src/tc_lora_uart_task.cpp:19:1: note: in expansion of macro ‘K_MSGQ_DEFINE’
 K_MSGQ_DEFINE(lora_input_queue, sizeof(u8_t), 256, 1);
 ^~~~~~~~~~~~~
../zephyr/include/kernel.h:3184:2: sorry, unimplemented: non-trivial designated initializers not supported
  }
  ^
```
**Environment (please complete the following information):**
 - OS: Ubuntu 18.04.1
 - Toolchain Zephyr SDK 0.9.5
 - Commit SHA: bbe6fa04b52644191da96aa5a289ae00b026641d

**Additional context**
Add any other context about the problem here.

*I have changed the code of the macro and it is running on my fork. The code before the changes:*
``` c
#define _K_MSGQ_INITIALIZER(obj, q_buffer, q_msg_size, q_max_msgs) \
	{ \
	.wait_q = Z_WAIT_Q_INIT(&obj.wait_q), \
	.max_msgs = q_max_msgs, \
	.msg_size = q_msg_size, \
	.buffer_start = q_buffer, \
	.buffer_end = q_buffer + (q_max_msgs * q_msg_size), \
	.read_ptr = q_buffer, \
	.write_ptr = q_buffer, \
	.used_msgs = 0, \
	_OBJECT_TRACING_INIT \
	}
```
*The code after chaging. To solve the problema I had to change the order of some designated initializers and add a missing one: *.lock*. The lock attribut can be a empty struct, thefore I had to create different initializers. *
``` c
#if defined(CONFIG_SMP) || defined(SPIN_VALIDATE)
#define SPIN_LOCK_NULL 0
#else
#define SPIN_LOCK_NULL
#endif

#define _K_MSGQ_INITIALIZER(obj, q_buffer, q_msg_size, q_max_msgs) \
	{ \
	.wait_q = Z_WAIT_Q_INIT(&obj.wait_q), \
	.lock = {SPIN_LOCK_NULL},\
	.msg_size = q_msg_size, \
	.max_msgs = q_max_msgs, \
	.buffer_start = q_buffer, \
	.buffer_end = q_buffer + (q_max_msgs * q_msg_size), \
	.read_ptr = q_buffer, \
	.write_ptr = q_buffer, \
	.used_msgs = 0, \
	_OBJECT_TRACING_INIT \
	}
```

andrewboie, 2019-03-14:
I would move SPIN_LOCK_NULL definition to include/spinlock.h and call it K_SPINLOCK_INITIALIZER.

However, I also notice that you didn't need to initialize 'flags' in your change.
Maybe you can just move the lock member to the end of the struct, and sort the remaining members that are initialized?

rodrigopex, 2019-03-15:
Great, moving it to the related file will be better.

That would be a good solution as well (better than I've suggested).

galak, 2019-03-18:
Do you mind submitting a PR for this?

gnuless, 2019-03-18:
@galak @rodrigopex I'm working on a patch for this now.

rodrigopex, 2019-03-18:
@gnuless I guess it will be necessary to reorder the lock position at the struct as well. It must be declared after the used field at the initializer. The order matters here.

gnuless, 2019-03-18:
@rodrigopex I don't think reordering the struct is necessary.  My understanding is that C++ requires that the designated initializers be in order, but not that every initializer be present.  Try the patch and if you still have problems, I'd be interested in which compiler you are using.  G++ 8.3 out of the 0.10.0 SDK finds the patch acceptable.

rodrigopex, 2019-03-18:
@gnuless I will try that.

PR #14645 (2019-03-19): kernel/msg_q: reorder _K_MSGQ_INITIALIZER() initializers
This is a trivial change to satisfy C++, which requires that designated
initializers appear in the same order as the members they initialize.

Fixes: #14540
commit 6d01f67f (2019-03-19): kernel/msg_q: reorder _K_MSGQ_INITIALIZER() initializers

This is a trivial change to satisfy C++, which requires that designated
initializers appear in the same order as the members they initialize.

Fixes: #14540
closed 2019-03-19 by andrewboie