zephyrproject-rtos/zephyr #14547: Kconfig shell prompt configuration

rodrigopex, 2019-03-15:
**Is your enhancement proposal related to a problem? Please describe.**
*A clear and concise description of what the problem is.*

There is no straightforward (CONFIG) way to change the prompt value for a shell backend.

**Describe the solution you'd like**
*A clear and concise description of what you want to happen.*

I suggest creating a Kconfig option like SHELL_PROMPT where the user can change that easily through the *prj.conf*.

**Describe alternatives you've considered**
*A clear and concise description of any alternative solutions or features you've considered.*

I would be good a way to remote the prompt at least. 

**Additional context**
*Add any other context or graphics (drag-and-drop an image) about the feature request here.*

If you would create a AT like interface it would be easy if we can remove the prompt and other features like echo, wildcards, etc.

aescolar, 2019-03-15:
@rodrigopex : Please do not leave the template text ("A clear and concise ..") into the issue itself. That is just meant to be replaced with your own content.

jarz-nordic, 2019-03-15:
Hi
First part of a request is clear. I will take care.

@rodrigopex
May you please elaborate a bit more about „AT interface like” to remove echo, meta keys... i don’t get it. Thease features can be already disabled via kconfig. In addition echo can be disabled in runtime.

rodrigopex, 2019-03-15:
> @rodrigopex : Please do not leave the template text ("A clear and concise ..") into the issue itself. That is just meant to be replaced with your own content.

Ok, I will do that now on.

rodrigopex, 2019-03-15:
> Hi
> First part of a request is clear. I will take care.
> 
> @rodrigopex
> May you please elaborate a bit more about „AT interface like” to remove echo, meta keys... i don’t get it. Thease features can be already disabled via kconfig. In addition echo can be disabled in runtime.

We are developing a system that must respond to commands like an AT command interface. In this case, removing the prompt is very important to enable us to use Zephyr shell as the controller of the command interface with no additional work. 

In short, for doing the "command interface" we just removed the prompt manually, disable all the other features of the shell and the interface is done. Commands like "b s 0 2" (just an example) will be easy to implement with shell subcommands and args management.

The prompt will change the message flow in a bad way.

PR #14843 (2019-04-30): shell: prompt configuration during project generation
Adding possibility to modify shell prompt in Kconfig and in prj.config file.
Fixes #14547.
commit 0e83ad65 (2019-04-30): shell: prompt configuration during project generation

Adding possibility to modify shell prompt in Kconfig and in prj.config
file.
Fixes #14547.
closed 2019-04-30 by nashif