zephyrproject-rtos/zephyr #14554: UP2 console no output after commit fb4f5e727b.

fulong82, 2019-03-15:
**Describe the bug**
After this commit ```fb4f5e727b logging: Use k_uptime_get_32 for high frequency system clock```
and build Zephyr for up squared board with the default configs + CONFIG_SHELL:
1.) https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/x86/up_squared/up_squared_defconfig#L4
2.) https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/timer/Kconfig#L19
3.) https://github.com/zephyrproject-rtos/zephyr/blob/master/kernel/Kconfig#L754
There is no console output from UART port 0.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=up_squared ../
3. make menuconfig & select CONFIG_SHELL
4. make
5. Boot up and wait for console output

**Expected behavior**
Console shell is able to show up on UART port 0 and user able to interact with Zephyr through the shell terminal.

**Impact**
User unable to interact with Zephyr shell terminal.

**Screenshots or console output**
[image]

**Environment (please complete the following information):**
 - OS: Ubuntu 14.04
 - Toolchain: Zephyr SDK 0.9.5
 - Commit: 96235c5027461534b75059a65f8833231d293312

**Additional context**
To workaround this issue, we could compile with CONFIG_TICKLESS_KERNEL=n

dcpleung, 2019-03-19:
I tried commit 2320973fea31086deedbbe1610b5e848e4ea9e79 using `samples/subsys/shell/shell_module` on the UP Squared board and it worked fine without any issue.

Did you update the BIOS? If so, the MMIO addresses might have changed. Do you have `CONFIG_PCI` and `CONFIG_PCI_ENUMERATION` turned on? Zephyr will probe for address and IRQ if they are both turned on.

fulong82, 2019-03-20:
Hi,

Here i attached my config file for your debugging reference:-
Not working config: [up2_no_console_config.txt](https://github.com/zephyrproject-rtos/zephyr/files/2986319/up2_no_console_config.txt)
Workaround config that able to see console: [up2_wa_console_output_config.txt](https://github.com/zephyrproject-rtos/zephyr/files/2986341/up2_wa_console_output_config.txt)
w/  the workaround below is the console output screenshot:
[image]

Thanks & Regards,
Wilson

fulong82, 2019-03-20:
Hi,

I do not know which BIOS version you are referring, below screenshot is the BIOS i am using:
[image]

Thanks & Regards,
Wilson

dcpleung, 2019-03-20:
The BIOS version is the "R4.0" in "UP-APL01 R4.0". Since you have PCI and PCI enumeration enabled, BIOS version should not matter.

Just to confirm, `samples/hello_world` with `CONFIG_SHELL=y` is not working for you, right?

Also, could you try `samples/subsys/shell/shell_module` without any modifications?

dcpleung, 2019-03-20:
`samples/hello_world` works fine for me with commit 3f34fead6053a6d29d6a252025d5ba4b59761051. I only added `CONFIG_SHELL=y` to the `prj.conf` and it worked.

```
WARNING: no console will be available to OS.strip  659.16KiB  100%  859.06B/s ]
error: no suitable video mode found.
***** Booting Zephyr OS v1.14.0-rc1-1334-g3f34fead6053 *****
Hello World! up_squared

uart:~$ help
Please press the <Tab> button to see all available commands.
You can also use the <Tab> button to prompt or auto-complete all commands or its subcommands.
You can try to call commands with <-h> or <--help> parameter for more information.
Shell supports following meta-keys:
Ctrl+a, Ctrl+b, Ctrl+c, Ctrl+d, Ctrl+e, Ctrl+f, Ctrl+k, Ctrl+l, Ctrl+u, Ctrl+w
Alt+b, Alt+f.
Please refer to shell documentation for more details.
uart:~$ 
  clear    help     history  resize   shell
uart:~$ 
```

Here is the config file: 
[working_config.txt](https://github.com/zephyrproject-rtos/zephyr/files/2989942/working_config.txt)

dcpleung, 2019-03-20:
I tried cherry-picking config options from the your uploaded files. The issue is actually an chicken-or-egg problem. When `CONFIG_PCI_ENUMERATION=y` and `CONFIG_PCI_LOG_LEVEL_DBG=y` are both enabled, the NS16550 UART driver is not initialized by default. However, during device scans, the PCI driver is trying to output through UART. And it caused an exception. To workaround this, find the actual MMIO address for the UARTs, and set `CONFIG_UART_NS16550_PCI=n`.

Not an actual bug on UP2. So close this one and opening another one on logging...

closed 2019-03-20 by dcpleung