zephyrproject-rtos/zephyr #14566: mcuboot doesn't link into code-partition

nvlsianpu, 2019-03-15:
mcuboot doesn't link into code-partition despite it is chosen.
```
	chosen {
		zephyr,code-partition = &boot_partition;
	};
```

**To Reproduce**
build mcuboot against nrf52840_pca10059 zephyr target or any else target which has shifted code-partition.

**Expected behavior**
linking into code partition.

**Impact**
mcuboot will not work on XIP target which has mcuboot partition started at offset different than Zero.

**Environment**
 - SHA 2153c014ea2eaf27d9f3442ae00af44386abe879 (master line)

**mcuboot issue**
https://github.com/JuulLabs-OSS/mcuboot/issues/444

nvlsianpu, 2019-03-15:
I would introduce Kconfig property:
```
config USE_CODE_PARTITION
	bool "Link application into code partition"
```
And use it for conditional assignment to `FLASH_LOAD_OFFSET` and `FLASH_LOAD_SIZE` the value of
`$(dt_hex_val,DT_CODE_PARTITION_OFFSET)` (as it is now done with `BOOTLOADER_MCUBOOT`).
Then `BOOTLOADER_MCUBOOT` should just select `USE_CODE_PARTITION`.

nvlsianpu, 2019-03-15:
cc @SebastianBoe

lemrey, 2019-03-15:
Can you confirm that this affects other boards?

nvlsianpu, 2019-03-15:
I confirm that this is issue for other boards as well.

mike-scott, 2019-03-15:
I mentioned this to @carlescufi a while back on Slack.  I was having a problems getting binaries built from Zephyr to have the correct flash offset settings.

I eventually got it work by manually specifying how I wanted mcuboot built:
```
west build -d build/mcuboot -b nrf52840_pca10059 -s mcuboot/boot/zephyr -- -DCONFIG_FLASH_LOAD_OFFSET=0x1000 -DCONFIG_FLASH_LOAD_SIZE=0xf000
```

Using this method, I was able to take the .hex file directly and place it on the board using the nRF-Connect GUI tool for Linux.

nvlsianpu, 2019-03-16:
@mike-scott Thanks for post this workaround.

SebastianBoe, 2019-03-18:
@nvlsianpu : Could you do a PR with your proposed solution?

I'm not familiar enough with these symbols to review unfortunately.

lemrey, 2019-03-18:
https://github.com/JuulLabs-OSS/mcuboot/pull/445

nvlsianpu, 2019-03-18:
@SebastianBoe 
https://github.com/zephyrproject-rtos/zephyr/pull/14620 + https://github.com/JuulLabs-OSS/mcuboot/pull/447

PR #14620 (2019-03-19): zephyr: fix mcuboot can't link into boot-partition
Objective of the path is to allow to select by the application
to link into the chosen code-partition.
Introduced USE_CODE_PARTITION will be set by bootloader project.

fixes #14566
commit 2b1227f1 (2019-03-19): zephyr: fix mcuboot can't link into boot-partition

Objective of the path is to allow to select by the application
to link into the chosen code-partition.
Introduced USE_CODE_PARTITION will be set by bootloader project.

fixes #14566
closed 2019-03-19 by carlescufi