zephyrproject-rtos/zephyr #14568: I2C stm32 LL driver V2 will hang when trying again after an error occurs

ranbochen, 2019-03-15:
**Describe the bug**
After I2C slave returns an NAK, stm32 ll driver v2 will hang when try transmit again.

Because on the second attempt, the driver did not start the transfer.
It's a logic bug.

See:
https://github.com/zephyrproject-rtos/zephyr/blob/83aa7cecd400c121a8eb1ead10669cbaa1d0e416/drivers/i2c/i2c_ll_stm32_v2.c#L34-L36

Hangs here:
https://github.com/zephyrproject-rtos/zephyr/blob/83aa7cecd400c121a8eb1ead10669cbaa1d0e416/drivers/i2c/i2c_ll_stm32_v2.c#L538-L541

gon1332, 2019-03-15:
@nikooiko

erwango, 2019-03-18:
@ydamigos, @ldts,  would you have time to look ta this one?

ydamigos, 2019-03-18:
@ranbochen Does the following patch solve the issue?
```
diff --git a/drivers/i2c/i2c_ll_stm32_v2.c b/drivers/i2c/i2c_ll_stm32_v2.c
index 67a938172b..1768461a4d 100644
--- a/drivers/i2c/i2c_ll_stm32_v2.c
+++ b/drivers/i2c/i2c_ll_stm32_v2.c
@@ -471,32 +471,42 @@ static inline int check_errors(struct device *dev, const char *funcname)
 {
        const struct i2c_stm32_config *cfg = DEV_CFG(dev);
        I2C_TypeDef *i2c = cfg->i2c;
+       bool error_occured = false;
 
        if (LL_I2C_IsActiveFlag_NACK(i2c)) {
                LL_I2C_ClearFlag_NACK(i2c);
                LOG_DBG("%s: NACK", funcname);
-               return -EIO;
+               error_occured = true;
+               goto error;
        }
 
        if (LL_I2C_IsActiveFlag_ARLO(i2c)) {
                LL_I2C_ClearFlag_ARLO(i2c);
                LOG_DBG("%s: ARLO", funcname);
-               return -EIO;
+               error_occured = true;
+               goto error;
        }
 
        if (LL_I2C_IsActiveFlag_OVR(i2c)) {
                LL_I2C_ClearFlag_OVR(i2c);
                LOG_DBG("%s: OVR", funcname);
-               return -EIO;
+               error_occured = true;
+               goto error;
        }
 
        if (LL_I2C_IsActiveFlag_BERR(i2c)) {
                LL_I2C_ClearFlag_BERR(i2c);
                LOG_DBG("%s: BERR", funcname);
-               return -EIO;
+               error_occured = true;
+               goto error;
        }
 
        return 0;
+error:
+       if (LL_I2C_IsEnabledReloadMode(i2c)) {
+               LL_I2C_DisableReloadMode(i2c);
+       }
+       return -EIO
 }
 
 static inline int msg_done(struct device *dev, unsigned int current_msg_flags)
```

ranbochen, 2019-03-22:
test ok for me.

PR #14849 (2019-03-28): i2c_ll_stm32_v2: Disable ReloadMode, if needed, when error occurs
Disable ReloadMode, if needed, when error occurs in polling mode.

Fixes #14568
commit 5f83759c (2019-03-28): i2c_ll_stm32_v2: Disable ReloadMode, if needed, when error occurs

Disable ReloadMode, if needed, when error occurs in polling mode.

Fixes #14568
closed 2019-03-28 by galak