zephyrproject-rtos/zephyr #14571: TCP: sending lots of data deadlocks with slow peer

dgloeck, 2019-03-15:
**Describe the bug**
When sending data on a TCP socket `net_pkt_alloc_with_buffer` might stall while holding `context->lock`, which is locked in `net_context_send_new`. This prevents `tcp_established` from processing ACK packets. But when ACK packets can't be processed, `net_pkt_alloc_with_buffer` will timeout, leading to an ENOMEM error.

**To Reproduce**
Steps to reproduce the behavior:
1. Send lots of data to a TCP peer, f.ex. with `send`
2. Wait until the peer becomes distracted and does not immediately send an ACK packet
3. Zephyr will use this opportunity to send data packets until its tx pool is exhausted
4. The peer will catch up and send an ACK packet that acknowledges all data packets
5. The rx_workq is now stuck in k_mutex_lock and can't process the ACK packet
6. The TCP retransmit timer will expire and Zephyr repeats an old packet although it was acknowledged by the unprocessed ACK packet
7. The peer will send more ACK packets that remain unprocessed
8. net_pkt_alloc_with_buffer will timeout, send fails, and the application will most likely drop the connection

**Expected behavior**
Received ACK packets should be processed while the TX path is waiting for free packets/buffers.

**Impact**
We aim for sending thousands of packets per second, so this is a showstopper.

**Screenshots or console output**
```
376499  33.187535 192.168.2.45 → 192.168.2.166 TCP mqtt(1883) → 63307 [ACK] Seq=1 Ack=3531217 Win=29200 Len=0
376500  33.189126 192.168.2.166 → 192.168.2.45 MQTT Publish Message
376501  33.189129 192.168.2.166 → 192.168.2.45 TCP [TCP segment of a reassembled PDU]
376502  33.189131 192.168.2.166 → 192.168.2.45 MQTT Publish Message
376503  33.189131 192.168.2.166 → 192.168.2.45 TCP [TCP segment of a reassembled PDU]
376504  33.189132 192.168.2.166 → 192.168.2.45 MQTT Publish Message
376505  33.189134 192.168.2.166 → 192.168.2.45 TCP [TCP segment of a reassembled PDU]
376506  33.189135 192.168.2.45 → 192.168.2.166 TCP mqtt(1883) → 63307 [ACK] Seq=1 Ack=3531269 Win=29200 Len=0
376507  33.189135 192.168.2.166 → 192.168.2.45 MQTT Publish Message
376508  33.189136 192.168.2.166 → 192.168.2.45 TCP [TCP segment of a reassembled PDU]
376509  33.189137 192.168.2.166 → 192.168.2.45 MQTT Publish Message
376510  33.189138 192.168.2.166 → 192.168.2.45 TCP [TCP segment of a reassembled PDU]
376511  33.189139 192.168.2.166 → 192.168.2.45 MQTT Publish Message
376512  33.189140 192.168.2.166 → 192.168.2.45 TCP [TCP segment of a reassembled PDU]
376513  33.189141 192.168.2.166 → 192.168.2.45 MQTT Publish Message
376514  33.189142 192.168.2.166 → 192.168.2.45 TCP [TCP segment of a reassembled PDU]
376515  33.189143 192.168.2.45 → 192.168.2.166 TCP mqtt(1883) → 63307 [ACK] Seq=1 Ack=3531388 Win=29200 Len=0
376516  33.231554 192.168.2.45 → 192.168.2.166 TCP mqtt(1883) → 63307 [ACK] Seq=1 Ack=3531399 Win=29200 Len=0
376517  33.389372 192.168.2.166 → 192.168.2.45 TCP [TCP Spurious Retransmission] 63307 → mqtt(1883) [PSH, ACK] Seq=3531217 Ack=1 Win=1280 Len=15[Reassembly error, protocol TCP: New fragment overlaps old data (retransmission?)]
376518  33.389388 192.168.2.45 → 192.168.2.166 TCP [TCP Dup ACK 376516#1] mqtt(1883) → 63307 [ACK] Seq=1 Ack=3531399 Win=29200 Len=0
376519  33.799360 192.168.2.166 → 192.168.2.45 TCP [TCP Spurious Retransmission] 63307 → mqtt(1883) [PSH, ACK] Seq=3531217 Ack=1 Win=1280 Len=15[Reassembly error, protocol TCP: New fragment overlaps old data (retransmission?)]
376520  33.799371 192.168.2.45 → 192.168.2.166 TCP [TCP Dup ACK 376516#2] mqtt(1883) → 63307 [ACK] Seq=1 Ack=3531399 Win=29200 Len=0
376521  34.609373 192.168.2.166 → 192.168.2.45 TCP [TCP Spurious Retransmission] 63307 → mqtt(1883) [PSH, ACK] Seq=3531217 Ack=1 Win=1280 Len=15[Reassembly error, protocol TCP: New fragment overlaps old data (retransmission?)]
376522  34.609385 192.168.2.45 → 192.168.2.166 TCP [TCP Dup ACK 376516#3] mqtt(1883) → 63307 [ACK] Seq=1 Ack=3531399 Win=29200 Len=0
376523 123.893927 192.168.2.45 → 192.168.2.166 TCP mqtt(1883) → 63307 [FIN, ACK] Seq=1 Ack=3531399 Win=29200 Len=0
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit d910aa6029e78d485f773f00c5be0ee544fe12eb

aescolar, 2019-03-16:
CC: @jukkar

tbursztyka, 2019-03-18:
Could you detail your hardware configuration, could you also paste a grep NET zephyr/.config  (from your build directory) ?

There is indeed an issue with lock, however there are also things that can be tweaked to help. Let's see.

dgloeck, 2019-03-18:
Well, the hardware is a SAM E70 Xplained board on the same switch as my PC.

Here is the NET part from my .config. I have omitted the LOG_LEVEL messages to shorten it a litte:
```
CONFIG_NET_L2_ETHERNET=y
# CONFIG_NET_L2_IEEE802154 is not set
# CONFIG_SOC_SERIES_KINETIS_K6X is not set
# CONFIG_SOC_SERIES_KINETIS_KL2X is not set
# CONFIG_SOC_SERIES_KINETIS_KWX is not set
# CONFIG_TELNET_CONSOLE is not set
# CONFIG_NET_LOOPBACK is not set
# CONFIG_NEURAL_NET_ACCEL is not set
# CONFIG_LOG_BACKEND_NET is not set
CONFIG_NET_BUF=y
CONFIG_NET_BUF_USER_DATA_SIZE=4
# CONFIG_NET_BUF_LOG is not set
# CONFIG_NET_BUF_POOL_USAGE is not set
CONFIG_NETWORKING=y
# CONFIG_NET_HOSTNAME_ENABLE is not set
# CONFIG_NET_L2_DUMMY is not set
# CONFIG_NET_L2_BT_SHELL is not set
# CONFIG_NET_L2_ETHERNET_MGMT is not set
# CONFIG_NET_VLAN is not set
CONFIG_NET_ARP=y
CONFIG_NET_ARP_TABLE_SIZE=2
CONFIG_NET_ARP_GRATUITOUS=y
# CONFIG_NET_GPTP is not set
# CONFIG_NET_LLDP is not set
CONFIG_NET_LLDP_CHASSIS_ID="CHASSIS_ID_PLACEHOLDER"
CONFIG_NET_LLDP_PORT_ID="PORT_ID_PLACEHOLDER"
# CONFIG_NET_L2_WIFI_MGMT is not set
# CONFIG_NET_L2_WIFI_SHELL is not set
# CONFIG_NET_L2_CANBUS is not set
CONFIG_NET_INIT_PRIO=90
# CONFIG_NET_IPV6 is not set
CONFIG_NET_IPV4=y
CONFIG_NET_INITIAL_TTL=64
CONFIG_NET_IF_MAX_IPV4_COUNT=1
CONFIG_NET_IF_UNICAST_IPV4_ADDR_COUNT=1
CONFIG_NET_IF_MCAST_IPV4_ADDR_COUNT=1
# CONFIG_NET_ICMPV4_ACCEPT_BROADCAST is not set
# CONFIG_NET_DHCPV4 is not set
# CONFIG_NET_IPV4_AUTO is not set
# CONFIG_NET_SHELL is not set
CONFIG_NET_TC_TX_COUNT=1
CONFIG_NET_TC_RX_COUNT=1
CONFIG_NET_TC_MAPPING_STRICT=y
CONFIG_NET_TX_DEFAULT_PRIORITY=1
CONFIG_NET_IP_ADDR_CHECK=y
CONFIG_NET_MAX_ROUTERS=1
CONFIG_NET_TCP=y
CONFIG_NET_TCP_CHECKSUM=y
CONFIG_NET_TCP_BACKLOG_SIZE=1
CONFIG_NET_TCP_TIME_WAIT_DELAY=250
CONFIG_NET_TCP_ACK_TIMEOUT=1000
CONFIG_NET_TCP_INIT_RETRANSMISSION_TIMEOUT=200
CONFIG_NET_TCP_RETRY_COUNT=9
CONFIG_NET_UDP=y
CONFIG_NET_UDP_CHECKSUM=y
CONFIG_NET_MAX_CONN=4
# CONFIG_NET_CONN_CACHE is not set
CONFIG_NET_MAX_CONTEXTS=6
# CONFIG_NET_CONTEXT_NET_PKT_POOL is not set
CONFIG_NET_CONTEXT_SYNC_RECV=y
CONFIG_NET_CONTEXT_CHECK=y
# CONFIG_NET_CONTEXT_PRIORITY is not set
# CONFIG_NET_CONTEXT_TIMESTAMP is not set
# CONFIG_NET_TEST is not set
# CONFIG_NET_TRICKLE is not set
CONFIG_NET_PKT_RX_COUNT=14
CONFIG_NET_PKT_TX_COUNT=14
CONFIG_NET_BUF_RX_COUNT=36
CONFIG_NET_BUF_TX_COUNT=36
CONFIG_NET_BUF_FIXED_DATA_SIZE=y
# CONFIG_NET_BUF_VARIABLE_DATA_SIZE is not set
CONFIG_NET_BUF_DATA_SIZE=256
CONFIG_NET_DEFAULT_IF_FIRST=y
# CONFIG_NET_DEFAULT_IF_ETHERNET is not set
# CONFIG_NET_PKT_TIMESTAMP is not set
# CONFIG_NET_PROMISCUOUS_MODE is not set
CONFIG_NET_TX_STACK_SIZE=1200
CONFIG_NET_RX_STACK_SIZE=1500
CONFIG_NET_MGMT=y
CONFIG_NET_MGMT_EVENT=y
CONFIG_NET_MGMT_EVENT_STACK_SIZE=512
CONFIG_NET_MGMT_EVENT_THREAD_PRIO=7
CONFIG_NET_MGMT_EVENT_QUEUE_SIZE=2
# CONFIG_NET_MGMT_EVENT_INFO is not set
# CONFIG_NET_DEBUG_MGMT_EVENT_STACK is not set
# CONFIG_NET_STATISTICS is not set
# CONFIG_NET_OFFLOAD is not set
CONFIG_NET_LOG=y
# CONFIG_NET_DEBUG_NET_PKT_ALLOC is not set
CONFIG_NET_DEBUG_NET_PKT_EXTERNALS=0
# CONFIG_NET_DEBUG_NET_PKT_NON_FRAGILE_ACCESS is not set
CONFIG_NET_CONFIG_AUTO_INIT=y
CONFIG_NET_CONFIG_INIT_PRIO=95
CONFIG_NET_CONFIG_INIT_TIMEOUT=30
# CONFIG_NET_CONFIG_NEED_IPV6 is not set
# CONFIG_NET_CONFIG_NEED_IPV4 is not set
CONFIG_NET_CONFIG_SETTINGS=y
CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.168.2.166"
CONFIG_NET_CONFIG_MY_IPV4_NETMASK="255.255.255.0"
CONFIG_NET_CONFIG_MY_IPV4_GW=""
CONFIG_NET_CONFIG_PEER_IPV4_ADDR="192.168.2.45"
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_POSIX_NAMES=y
CONFIG_NET_SOCKETS_POLL_MAX=3
# CONFIG_NET_SOCKETS_SOCKOPT_TLS is not set
# CONFIG_NET_SOCKETS_OFFLOAD is not set
# CONFIG_NET_SOCKETS_PACKET is not set
# CONFIG_NET_SOCKETS_CAN is not set
```

I have thought about raising CONFIG_NET_PKT_TX_COUNT and CONFIG_NET_BUF_TX_COUNT, but that will not eliminate the problem. It will just be less likely. Temporarily detaching the PC from the switch should still cause the error.

What I did try was to unlock the mutex while calling `net_pkt_alloc_with_buffer` (both in `context_alloc_pkt` and in `prepare_segment`). Some other places also need locking added or temporarily removed for this to work. The application ran the whole weekend constantly sending >8000 TCP packets per second. I know the approach is ugly, but it does appear to work. Of course it has to be thoroughly checked if there are any side effects when the state of the socket does change during `net_pkt_alloc_with_buffer`.

rljordan, 2019-03-19:
@tbursztyka Please review priority.

tbursztyka, 2019-03-19:
@dgloeck what about unlocking only in context_alloc_pkt()? That could be sufficient. Indeed playing with unlock in tcp side is potentially creating issues (it's already quite unstable...)

dgloeck, 2019-03-19:
@tbursztyka I tried that at first, but then it got stuck in `prepare_segment`.

tbursztyka, 2019-03-19:
Ok, well let me send a "fix" based on you tests

pfalcon, 2019-03-19:
Of related issue, we have functions which take timeout parameter, but call `k_mutex_lock(&context->lock, K_FOREVER);` in them. I.e. even despite timeout parameter, they could deadlock forever (of course, one would hope that deadlock can never happen in them, but issues like this teach otherwise).

dgloeck, 2019-03-19:
@tbursztyka, at first I was surprised how short your patch is. But then I saw that you did not unlock the mutex in `prepare_segment`. I have uploaded my patch to dgloeck/zephyr@e61634cf72abfdb824c96b8924e07f552c210a07. I believe I have covered all places that call into `prepare_segment`. Most of this is untested, though.

tbursztyka, 2019-03-20:
prepare_segment should not unlock the lock: the problem, currently, is that caller to prepare_segment might not have locked context. It's an error to unlock a non-locked mutex. (try your patch with CONFIG_ASSERT=y, or directly tests/net/tcp/).

dgloeck, 2019-03-20:
@tbursztyka Did you try my patch with CONFIG_ASSERT=y? I added locking to all functions that might eventually call `prepare_segment`:
 - `prepare_segment` is called by `net_tcp_prepare_segment` and `net_tcp_prepare_reset`
 - `net_tcp_prepare_segment` is called by `net_tcp_prepare_ack`, `net_tcp_queue_data`, `queue_fin`, and `send_syn_segment`
 - `net_tcp_prepare_ack` is only called by `send_ack`, which is only called by `tcp_established`, which already has locking
 - `net_tcp_queue_data` is called by `sendto` and `context_sendto_new`
 - `sendto` is only called by `net_context_send`, which already has locking
 - `context_sendto_new` is only called by `net_context_sendto_new`, which already has locking
 - `queue_fin` is only called by `net_tcp_put`, which is only called by `net_context_put`, which already has locking
 - `send_syn_segment` is called by `send_syn` and `send_syn_ack`
 - `send_syn` is only called by `net_tcp_connect`, which is only called by `net_context_connect`, which already has locking
 - `send_syn_ack` is only called by `tcp_syn_rcvd`, where I added locking
 - `net_tcp_prepare_reset` is only called by `send_reset`
 - `send_reset` is called by `backlog_ack_timeout`, `tcp_synack_received`, and `tcp_syn_rcvd` and I added locking to all three functions
 - `since tcp_synack_received` has to be able to run in parallel to `net_tcp_connect`, I temporarily unlock the mutex there

tbursztyka, 2019-03-21:
@dgloeck I am fine with your approach, it's just going much further than first thought. It's basically fixing also commit 93e5181fbdf

@jukkar as you introduced the context lock, what do you think about @dgloeck's patch?

jukkar, 2019-03-21:
> @jukkar as you introduced the context lock, what do you think about @dgloeck's patch?

Looks quite ok to me. Needs a bit more testing. @dgloeck, could you prepare a proper PR for that?

tbursztyka, 2019-03-27:
@dgloeck ping

ioannisg, 2019-08-27:
@tbursztyka is this issue still present?

jukkar, 2019-08-29:
There has been fixes in TCP side so it is difficult to know if this is really fixed or not. There has not been reports about issues like this lately.

rljordan, 2019-08-29:
This was not seen in any of our tests.  Can this be closed?

rljordan, 2019-08-29:
Lowering the priority as it has not been seen in some time.

pfalcon, 2019-08-29:
> There has not been reports about issues like this lately.

There was no reports, because everything stays as it was, the TCP implementation is broken.

jukkar, 2020-03-11:
This is an old issue and it talks about problems when sending data so just fyi atm. I fixed several issues when receiving lot of incoming data in #23334 and it might show improvements in the sending side too.

closed 2020-10-04 by github-actions[bot]