zephyrproject-rtos/zephyr #14581: Network interfaces should be able to declare if they work with IPv4 and IPv6 dynamically

mike-scott, 2019-03-15:
**Is your enhancement proposal related to a problem? Please describe.**
Recently, I've been working with several LTE Cat M1 / NB-IoT modems.  Depending on the carrier / APN used, support for IPv6 has been spotty.  In most cases, the modem wants an IPv4 connection and it's hard to get away from.  Also, in the case of FRDM K64F, the ethernet port might or might not get IPv6 information based on the network it's connected to.

Most of the networking samples in Zephyr enable both IPv6 and IPv4 by default.  In most of the code, IPv6 will be the default in order of execution if they are both enabled.  I find myself needing to manually disable the IPv6 configs a lot depending on the circumstances.

**Describe the solution you'd like**
It seems like the network interface should be able to declare if it works with IPv4/IPv6 **AFTER** being initialized.

In the case of the modem, if the carrier / APN used only supports an IPv4 connection then the modem's network interface should be able to say so.  Then all of the Zephyr code can use that interface flag (combined with the CONFIG_NET_IPV4/IPV6 checks) to determine which peer settings to use.  In the case of ethernet, it could be that the port never receives a global IPv6 address, etc.

We have place in the code where we are using #ifdef checks such as this:
https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/ip/net_shell.c#L445

These ad-hoc #ifdef checks could go away and only look at the interface flag (BT 6lowpan and 802.15.4 interfaces would always disable IPv4 support regardless of whether CONFIG_NET_IPV4 was enabled).

mike-scott, 2019-03-15:
Adding @pfalcon @jukkar for discussion.  Feel free to add anyone else.

jukkar, 2019-03-18:
The net_if struct already has dynamic (runtime) support for IPv6 / IPv4 enable/disable feature. See `net_if->config.ip.{ipv6|ipv4}`, it is just that we are missing some way to only enable/disable the support when needed. Those pointers are manipulated by `net_if_config_ipv{4|6}_get()` and `net_if_config_ipv{4|6}_put()`

laperie, 2019-04-05:
@jukkar  -- please close /resolve the issue as you see fit

jukkar, 2019-04-14:
@mike-scott I think we have all the information already in network interface for what you want. As you have some specific use case on mind, could you propose an API how the information in network interface could be used for that?

commit 897698bc (2021-03-29): net: if: Allow app to disable IPv4 or IPv6 for the interface

Application can disable IPv4 or IPv6 later if those are not
needed nor used for a given network interface.

Fixes #14581
closed 2021-03-29 by nashif