zephyrproject-rtos/zephyr #14591: Infineon Tricore architecture support

ilanbiala, 2019-03-15:
Are there plans to support [Infineon's Tricore architecture](https://www.infineon.com/cms/en/product/microcontroller/32-bit-tricore-microcontroller/)? The MCU's ASIL rating goes up to ASIL-D and Infineon is a very popular choice for automotive MCUs, so it seems like a great fit considering Zephyr is targeting ASIL-D.

ilanbiala, 2019-04-26:
Bump, any update on this?

nashif, 2019-05-22:
@ilanbiala no plans currently however, we would consider any contributions supporting this architecture.

parthitce, 2019-10-07:
@ilanbiala I have a received xmc4000 based development board today. I will start working on this for Zephyr support. I will be able to start for Tricore after this.

carlescufi, 2020-03-17:
Please submit a Pull Request

ilanbiala, 2020-08-13:
@parthitce just wanted to see if you ever got a chance to work on Tricore support?

parthitce, 2020-08-13:
@ilanbiala Yes I did ported the arch parts of the Zephyr with the help of Aurix 2x ISA. But the board which I have from Infineon is based on Aurix 3x and it's not straight forward with the ISA for 3x series. I had to go via various NDA process which is almost done and waiting for Infineon to release the ISA in their collaborative platform.

IFX HAL is BSD licensed in general, but am still not sure about the open sourcing the IFX based in 3x. But HAL for 2x is possible. So the SoC porting is yet to be done and waiting for the ISA and license clearance from Infineon.

I used the Toolchain from hightech (close sourced), but there is also an attempt to port the GCC backend for tricore from my end. GCC backend work is posted here: https://github.com/parthitce/gcc, but it's long way to go.

I will push the WIP tree for tricore Zephyr port once I get the license clearance from Infineon.

ilanbiala, 2020-08-20:
Awesome @parthitce, glad to hear that things are at least progressing! Any easy way to follow what's going on?

mgiaco, 2021-02-09:
@parthitce  Do you share the ported parts?

parthitce, 2021-03-29:
@mgiaco 
https://github.com/linumiz/zephyr/tree/tricore Here is the work in progress tree.
Please note that this tree includes only Architecture changes (not yet compilable) and SoC port for Aurix 3x is done but I can't release it to open source until I get a green signal for the iLLD HAL (it's Boost Software License - Version 1.0) & SoC ports from Infineon. Although NDA is done with Infineon, this will take some more time before getting into mainline Zephyr.

But in short, Linumiz and I myself handling this.

go2sh, 2022-05-29:
Hey, whats the status on the tricore port? I have quite some experience with the tc3xx devices and a few boards to hand. I just saw one commit in the branche there.

UVV-gh, 2022-11-29:
Is there any update on this?

Jeepgoing, 2023-03-24:
Is there any update on this?