zephyrproject-rtos/zephyr #14599: Can you add a ADC sample?

hexiangdong2017, 2019-03-16:
I'm using a nRF52832 broad to develope a project and I use Zephyr on it. I need to use NXP MPXV7002 pressure sensor and it's a analog sensor. But I can't find the samples of ADC.
Follows are my codes, but the value is always 2526300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
```
#include <zephyr.h>
#include <misc/printk.h>
#include <adc.h>
#include <gpio.h>

#define INTERVAL 2
#define SLEEPTICKS MSEC(INTERVAL)
#define ADC_BUFFER_SIZE 100

static uint8_t seq_buffer[ADC_BUFFER_SIZE];//ADC数据缓冲区

void adc_callback(void){
	int i=0;
	for(i=0;i<100;i++){
		printk("%d",seq_buffer[i]);
	}
	printk("\n");
}

void main(void)
{
	//struct device *P0;
	//P0=device_get_binding("GPIO_0");
	//gpio_pin_configure(P0,2,GPIO_DIR_IN);
	struct device *adc0;
	adc0=device_get_binding("ADC_0");//获取ADC0的设备
	struct adc_channel_cfg ch_cfg;//ADC Channel设置
	ch_cfg.gain=ADC_GAIN_1;//增益为1
	ch_cfg.reference=ADC_REF_INTERNAL;//参考值为VDD
	ch_cfg.acquisition_time=ADC_ACQ_TIME(ADC_ACQ_TIME_MICROSECONDS,40);//获取时间为40MS
	ch_cfg.channel_id=0;
	ch_cfg.differential=0;
	struct adc_sequence_options adc0_sq_opt;
	adc0_sq_opt.interval_us=0;
	adc0_sq_opt.callback=adc_callback;
	adc0_sq_opt.extra_samplings=0;
	struct adc_sequence adc0_sq;
	adc0_sq.options=&adc0_sq_opt;
	adc0_sq.channels=1;
	adc0_sq.buffer=seq_buffer;
	adc0_sq.buffer_size=100;
	adc0_sq.resolution=14;
	adc0_sq.oversampling=0;
	if(adc_channel_setup(adc0,&ch_cfg)==0){
		printk("Open ADC0 Successfully!\n");
		while(1){
			int x=0;
			x=adc_read(adc0,&adc0_sq);
			printk("%d\n",x);
		}
	}
}
```

galak, 2019-03-26:
Are you wanting a sample for this sensor?

andrewboie, 2019-04-01:
@hexiangdong2017 Have you looked at tests/drivers/adc/adc_api?

ioannisg, 2019-07-31:
@hexiangdong2017 please, re-open if needed, to continue discussion

closed 2019-07-31 by ioannisg