zephyrproject-rtos/zephyr #14601: Failed to get model for elem_idx 3 mod_idx 0

lucaspeixotot, 2019-03-16:
Hello, I have a hello world code to learn about bluetooth mesh. When I implemented it zephyr was in 1.13.99 version and everything was running good. Yesterday I tried to running it again and I receive this errors:

```
[00:00:00.017,669] <err> bt_mesh_access: Invalid element index 3
[00:00:00.017,669] <err> bt_mesh_settings: Failed to get model for elem_idx 3 mod_idx 0
[00:00:00.017,669] <err> settings: set-value failure. key: �- error(-2)
[00:00:00.017,944] <err> bt_mesh_access: Invalid element index 3
[00:00:00.017,944] <err> bt_mesh_settings: Failed to get model for elem_idx 3 mod_idx 0
[00:00:00.017,974] <err> settings: set-value failure. key: �- error(-2)
[00:00:00.018,524] <err> bt_mesh_access: Invalid element index 3
[00:00:00.018,524] <err> bt_mesh_settings: Failed to get model for elem_idx 3 mod_idx 0
[00:00:00.018,524] <err> settings: set-value failure. key: �- error(-2)
[00:00:00.019,042] <err> settings: set-value failure. key: �- error(-22)
[00:00:00.019,165] <err> settings: set-value failure. key: �- error(-22)
```

For me this problem is very strange because my component have just two elements, and this error says as I had more. 

For understanding the code follow some prints that I think that are necessary.

[image]
[image]
[image]

This code is meant to create a generic on/off server and a generic level server. 

The client side that will consume the services is on another code, that will run on another board. 
PS: I am trying to run this code on a NRF 52

Follow up is the prj.conf that I got from a example bluetooth code at the zephyr page.

```
#CONFIG_INIT_STACKS=y      # if set, lots of noise from mesh/adv.c: STACK_ANALYZE
CONFIG_MAIN_STACK_SIZE=512
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048

CONFIG_BT_SETTINGS=y
CONFIG_FLASH=y
CONFIG_FLASH_PAGE_LAYOUT=y
CONFIG_FLASH_MAP=y
CONFIG_FCB=y
CONFIG_SETTINGS=y
CONFIG_SETTINGS_FCB=y
CONFIG_BT_MESH_RPL_STORE_TIMEOUT=600

CONFIG_BOOT_BANNER=y

CONFIG_BT_OBSERVER=y
CONFIG_BT_BROADCASTER=y
CONFIG_BT_CTLR_DUP_FILTER_LEN=0
CONFIG_BT_CTLR_LE_ENC=n
CONFIG_BT_CTLR_LE_PING=n
CONFIG_BT_DATA_LEN_UPDATE=n
CONFIG_BT_PHY_UPDATE=n
CONFIG_BT_CTLR_CHAN_SEL_2=n
CONFIG_BT_CTLR_MIN_USED_CHAN=n
CONFIG_BT_CTLR_ADV_EXT=n
CONFIG_BT_CTLR_PRIVACY=n

CONFIG_BT_PERIPHERAL=y

CONFIG_BT=y
CONFIG_BT_TINYCRYPT_ECC=y
CONFIG_BT_RX_BUF_COUNT=30
CONFIG_BT_L2CAP_RX_MTU=69
CONFIG_BT_L2CAP_TX_MTU=69
CONFIG_BT_L2CAP_TX_BUF_COUNT=8

CONFIG_BT_MESH=y
CONFIG_BT_MESH_RELAY=y
CONFIG_BT_MESH_LOW_POWER=n
CONFIG_BT_MESH_FRIEND=n
CONFIG_BT_MESH_FRIEND_QUEUE_SIZE=16
CONFIG_BT_MESH_ADV_BUF_COUNT=40
CONFIG_BT_MESH_TX_SEG_MAX=12

CONFIG_BT_MESH_PB_GATT=y
CONFIG_BT_MESH_PB_ADV=y
CONFIG_BT_MESH_GATT_PROXY=y
CONFIG_BT_MESH_CFG_CLI=y

#CONFIG_BT_MESH_LPN_SCAN_LATENCY=30
#CONFIG_BT_MESH_LPN_RECV_DELAY=40
#CONFIG_BT_MESH_LPN_POLL_TIMEOUT=300

CONFIG_BT_MESH_SUBNET_COUNT=2
CONFIG_BT_MESH_APP_KEY_COUNT=2
CONFIG_BT_MESH_MODEL_KEY_COUNT=2
CONFIG_BT_MESH_MODEL_GROUP_COUNT=2
CONFIG_BT_MESH_LABEL_COUNT=3

#CONFIG_BT_MESH_IV_UPDATE_TEST=y

CONFIG_UART_CONSOLE=y

# this outputs btmon formatted data to the serial port
#CONFIG_BT_DEBUG_MONITOR=y

CONFIG_BT_DEBUG_LOG=y

CONFIG_BT_MESH_DEBUG=y
#CONFIG_BT_MESH_DEBUG_PROV=y
#CONFIG_BT_MESH_DEBUG_PROXY=y
#CONFIG_BT_MESH_DEBUG_BEACON=y
#CONFIG_BT_MESH_DEBUG_NET=y
#CONFIG_BT_MESH_DEBUG_TRANS=y
#CONFIG_BT_MESH_DEBUG_LOW_POWER=y
#CONFIG_BT_MESH_DEBUG_FRIEND=y
#CONFIG_BT_MESH_DEBUG_MODEL=y
#CONFIG_BT_MESH_DEBUG_ACCESS=y
#CONFIG_BT_MESH_DEBUG_CRYPTO=y
#CONFIG_BT_MESH_DEBUG_ADV=y
#CONFIG_BT_MESH_SELF_TEST=y

#CONFIG_BT_HCI_VS_EXT=n

#CONFIG_STACK_USAGE=y

CONFIG_BT_RX_STACK_SIZE=4096

#CONFIG_BT_DEBUG_HCI_DRIVER=y
#CONFIG_BT_DEBUG_HCI_CORE=y
#CONFIG_BT_DEBUG_GATT=y
#CONFIG_BT_DEBUG_CONN=y

CONFIG_BT_MAX_CONN=1
CONFIG_BT_CTLR_RX_BUFFERS=6
CONFIG_BT_CTLR_TX_BUFFERS=4
CONFIG_BT_HCI_CMD_COUNT=4

CONFIG_BT_ATT_PREPARE_COUNT=2

#CONFIG_NET_BUF_SIMPLE_LOG=y
#CONFIG_NET_BUF_LOG=y
#CONFIG_BT_CTLR_DEBUG_PINS=y

CONFIG_PWM=y
```
Please, does someone know how to solve this? 

I need to implement a light lightness server and I was basing on this code and on samples/bluetooth/mesh_demo that have the same logic.

WilliamGFish, 2019-03-16:
Interesting, I would have a look at the samples under "board" / nrf5 as there are some examples that have implemented light server etc. 

Billy..

jhedberg, 2019-03-16:
It looks like you've at some point experimented with code that had at least 4 elements and the stack had store information about them in flash (hence the "invalid element index 3" from the settings subsystem). I'd suggest erasing the storage partition where this kind of info is stored. In general this is expected behavior if you go ahead and change the composition for your application - if you do that you should also erase any previously stored info in flash since it will most likely be invalid.

lucaspeixotot, 2019-03-16:
> It looks like you've at some point experimented with code that had at least 4 elements and the stack had stored information about them in flash (hence the "invalid element index 3" from the settings subsystem). I'd suggest erasing the storage partition where this kind of info is stored. In general this is expected behavior if you go ahead and change the composition for your node - if you do that you should also erase any previously stored info in flash since it will most likely be invalid.

This is interesting, I really used this code before with more than 2 elements, maybe this make sense. But, how can I erase the storage partition? I never work with it.

jhedberg, 2019-03-16:
> how can I erase the storage partition? I never work with it.

@lucaspeixotot that depends on the board that you're using. E.g. with nrfjprog you'd do it using `nrfjprog -e` (IIRC). If you're using pyocd-tool it has some similar thing for it. Generally I'd just recommend erasing all flash and reflashing the Zephyr image onto it - that way you don't need to figure out the exact offset and size of the storage partition.

jhedberg, 2019-03-16:
@lucaspeixotot btw, there might be some cleaner way to force a full erase using west that works with all boards, but I'm not so familiar yet with this tool. Maybe @carlescufi or @mbolivar have an idea?

jhedberg, 2019-03-16:
Even better, it'd be nice if west had a way of requesting it to erase only the storage partition.. I realize this is starting to turn into a feature request :)

lucaspeixotot, 2019-03-16:
> It looks like you've at some point experimented with code that had at least 4 elements and the stack had stored information about them in flash (hence the "invalid element index 3" from the settings [...]

I think that this was exactly my problem, thank you. I erased the storage partition and so far all my bluetooth codes are working as expected.

> @lucaspeixotot that depends on the board that you're using. E.g. with nrfjprog you'd do it using `nrfjprog -e` (IIRC). If you're using pyocd-tool it has some similar thing for it. Generally I'd just [...]

I'm using both nrf52_pca10040 and nrf52840_pca10056. Now i'm flashing my code with this routine:

```
#!/bin/bash
nrfjprog --eraseall -f nrf52
nrfjprog --program zephyr/zephyr.hex -f nrf52
nrfjprog --reset -f nrf52
```

> Even better, it'd be nice if west had a way of requesting it to erase only the storage partition.. I realize this is starting to turn into a feature request :)

I would like to trust in west again to flash my code, this would be good.

lucaspeixotot, 2019-03-21:
The problem/question was solved, so I will close this issue. Thank you @jhedberg

closed 2019-03-21 by lucaspeixotot